T-Troll/alienfx-tools #1: Alienware LightFX DLLs

crim2som, 2020-07-25:
The pre-requirement to have Alienware LightFX DLLs installed on your computer is invalid when AlienFX v1.0 and/or AlienFX v2.0 DLLs are not installed by the Alienware Command Center.  When these DLLs are not installed the alienfx-tools should provide them.

T-Troll, 2020-07-26:
Looks strange for me, at both of my rigs (API 2.0 and API 1.0) it's installed into /windows/system32.
Anyway, good point, i'll add it into archive.
Also thank you to reveal a fact i forget to request enable 3rd-party access into AWCC.

T-Troll, 2020-07-26:
I include DLL into 0.4.4, but looks strange - let me check deep how it possible to have AWCC without it. Maybe some old versions doesn't have it (i see this a year ago...).

closed 2020-07-26 by T-Troll