T-Troll/alienfx-tools #2: VCRUNTIME140_1.dll

crim2som, 2020-07-25:
alienfx-tools require VCRUNTIME140_1.dll to be pre-installed. Please provide the link to get the latest supported Visual C++ from https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads which will then install the VCRUNTIME140_1.dll.

T-Troll, 2020-07-26:
Aha... VCC redestributable...
It should not to be involved, seems like i need to change some project settings.
I'll fix it or include package in next release.

T-Troll, 2020-07-26:
Check latest release, it should not depend of VC/MSVC anymore.

crim2som, 2020-07-26:
Hello @T-Troll thank you for updating your AlienFX applications. We used alienfx-tools v0.4.4 on our Alienware 17 R5 Laptop whose AWCC FX tab refuses to open. 
There was no apparent response when trying to run alienfx-cli.exe with local admin privileges. The alienfx-haptics.exe did respond this time and opened a Universal AlienFX haptics window. We opened the Parameters tag but the drop down menu could not find our device. 
We are unsure how alienfx-tools will get our preinstalled AlienFX v2.0 LED hardware to work.  
[image]
[image]

T-Troll, 2020-07-26:
Can you provide me "alienfx-cli status" output?
All this apps doesn't require elevation!

crim2som, 2020-07-26:
Hi @T-Troll thank you for asking about "alienfx-cli status" output. Not sure where to find status output. Where / how is this to be found?

T-Troll, 2020-07-26:
Just run "alienfx-cli.exe status" command and share output for me.
Also, can you do the same without DLL i provide as well?

PS: I hope you Windows is 64-bit?

crim2som, 2020-07-26:
Please excuse my ignorance, just run "alienfx-cli.exe status" command. Is this via PowerShell? 
OS is Windows 10 Professional x64 complete with all Windows Updates.

crim2som, 2020-07-26:
OK, have consulted internet and its Windows + R, and "alienfx-cli.exe status" command says Windows cannot find 'alienfx-cli.exe'.

crim2som, 2020-07-26:
Guess that alienfx-cli.exe is not installed, running or active within Win10 x64.

crim2som, 2020-07-26:
Troubleshooting report says Incompatible Program. Fix program alienfx-cli was Not fixed. 
[image]

crim2som, 2020-07-26:
Command and PowerShell say there are problems. 
[image]
[image]

T-Troll, 2020-07-27:
alienfx-cli is a part of the package, it's into the same archive.
Now i release updated alienfx-cli with more diagnostic for you.
Also, check - do you have LightFX.dll into /windows/system32 or/end /windows/sysWOW64?
if you don't have, it means you AWCC installed incorrectly or outdated (they have some buggy versions without AlienFX support about a year ago) - reinstall it in this case (just update is OK).
If you have blank device list into alienfx-haptics it's mean no DLL or no devices found (i'll add some message in this case later on).

crim2som, 2020-07-27:
Hi @T-Troll thank you for creation of alienfx-tools v0.4.5. However, there was no response from the updated alienfx-cli.exe or the alienfx-haptics.exe applications. 

Our Alienware 17 R5 Laptop has:- 
(1) The LightFX.dll file installed in both Windows > System32 and SysWOW64. 
(2) Program Files > Alienware > Alienware Command Center has 67 items and these include AlienFX.dll and Alien64.dll files, but not AlienFX Tester, etc. 
(3) Program Files (x86) > Alienware > Media Plugin > x64 with the LightFX.dll file installed. 

We have the legacy AWCC Suite v5.1.18.0 installed and clicking on the FX tab does not work. Not sure when Alienfx v2.0 was required for 17 R5. Dell support service tag says the current AWCC Suite is v5.2.89.0, but there is no FX tab when this is installed. 

Our Alienware laptop out of the transit box, was dark and continues to be dark. 
It would be great if we could turn the keyboard lights on.

T-Troll, 2020-07-27:
Ok, o'm open new issue (this is about MFC, not LightFX) - https://github.com/T-Troll/alienfx-tools/issues/3
I''ll answer you there.

closed 2020-07-27 by T-Troll