T-Troll/alienfx-tools #3: Can't init AlienFX at some rig (f.e. Alienware 17 R5).

T-Troll, 2020-07-27:
Hi @T-Troll thank you for creation of alienfx-tools v0.4.5. However, there was no response from the updated alienfx-cli.exe or the alienfx-haptics.exe applications.

Our Alienware 17 R5 Laptop has:-
(1) The LightFX.dll file installed in both Windows > System32 and SysWOW64.
(2) Program Files > Alienware > Alienware Command Center has 67 items and these include AlienFX.dll and Alien64.dll files, but not AlienFX Tester, etc.
(3) Program Files (x86) > Alienware > Media Plugin > x64 with the LightFX.dll file installed.

We have the legacy AWCC Suite v5.1.18.0 installed and clicking on the FX tab does not work. Not sure when Alienfx v2.0 was required for 17 R5. Dell support service tag says the current AWCC Suite is v5.2.89.0, but there is no FX tab when this is installed.

Our Alienware laptop out of the transit box, was dark and continues to be dark, despite having the pre-installed Alienfx hardware. It would be great if we could turn the keyboard lights on.

T-Troll, 2020-07-27:
Let's start from the beginning.
What do you mean saying "no response"? it didn't start? Start, but hungs? No output?
Just to be sure, what result you got if start cmd.exe then issue this commands:
`cd c:/<where you unpack my archive>
alienfx-cli.exe status`?

Uff.... So you can't switch FX using AWCC? In this case check Device Manager - i see the situation FX device (in fact, is marked as "HID-compliant Vendor-defined device" (but in fact, just USB HID)) was not start correctly. You can find it across others checking hardware ID's - VID should be VID_187C, PID depends of you gear - f.e. PID_0550 for m15 (it's FX device ID, in fact). So check for lost/wrong device and try to reinstall/restart it. In other case, i think, it should be a hardware problem.

> AWCC Suite v5.1.18.0
It should be OK, alienfx-tools works at even older version, i also check with 4.5.19.1 from 2017 at my old 13R2.
I have 5.2.89.0, but seems like the disable M17 support from latest versions - only Intel 7, 8 and 9-gen now supported.
I see 17R5 in AWCC profile, but for i9-8950HK only (yes, Dell is strange sometimes).

crim2som, 2020-07-27:
Administrator CMD: cd C:\Users . . . alienfx-cli.exe status = "The system cannot find the path specified." (and yes, the path was double checked and it was correct). But the alienfx-cli.exe path does not exist as reported by PowerShell and posted in Issue 2 (guess that somehow the alienfx-cli.exe file is not recognised by Windows 10 Professional (x64) Version 2004 (build 19041.388). 

Device Manager, System devices, AWCC Driver, Properties, General tab = "This device is working properly." There are no Device Manager problems anywhere. All the Dell SupportAssist hardware tests = pass. So it is not a hardware problem. 

Our 17 R5 has 8th Generation Intel Core i7-8750H CPU. Dell Service tag gives list of AWCC installers to use with this laptop. Many Dell Alienware Community reports say **incorrect AWCC installer code** is preventing installation of Alienfx applications.

T-Troll, 2020-07-28:
1. If path have spaces, put it into double quota.
2. If this not help, try to make folder at the root of C: and unpack archive there to avoid virtualization, check alienfx-cli.exe in place.
But looks crazy, like antivirus block it.

3. If you sure HUD device are ok, you can try other check:
- Write down device PID.
- Uninstall AWCC _completely_ - google for how to do it, there are good manual.
- Download and unpack direct AlienFX tester from attach
- Run it and follow on-screen instruction (you will be asked for PID from above).
It work at driver level, not above AWCC. It may not work for new systems, but worth a try.

I'm quite interested about the result.

[alienfx-led-tester.zip](https://github.com/T-Troll/alienfx-tools/files/4985667/alienfx-led-tester.zip)

crim2som, 2020-07-28:
Results
1. There are no spaces, Path is C:\alienfx-tools\alienfx-cli.exe 

2. Unpack archive at C:\alienfx-tools
cmd line = cd C:\alienfx-tools\alienfx-cli.exe status
cmd reponse = The system cannot find the path specified. 
There was no reported antivirus blocking.

3. Thank you for alienfx-led-tester application, which was unpacked in a root folder. 
Eventually found how to get USB Vendor ID and Product ID from Device Manager. Alienware 17 R5 = USB\VID_0781&PID_5591. 
There was no alienfx-led-tester ReadMe file so do not know what this application wants, e.g. PID_5591 or 5591 or other? We used PID_5591 and then 5591 when asked to insert the product id of the device, both were declared "Invalid product id". 
As alienfx-led-tester works at driver level, do not know why "Uninstall AWCC completely" is required. We have Revo Uninstaller, so can use this to completely remove AWCC.

T-Troll, 2020-07-29:
2. it's TWO commands, not one:
cd C:\alienfx-tools\ [Enter]
alienfx-cli.exe status [Enter]
You can also enter:
dir c:\alienfx-tools\ and check all 4 files in place.

3. Ohhh.... I don't use it for long time, forget to mention, sorry. You need to enter 0x[PID], 0x5591 in you case. AWCC uninstall is optional, you need to do it in case led tester can open device, but can't switch lights. BTW, it's open source too, so if it can control you gear - i can made alienfx-cli using this method, not SDK-based like now.

Yes, i see community posts. "files" are not missing. ID's inside database are. You can check inside Profiles.zip into /[Awcc folder]/OcControlService/ - just for fun (but it's about fan control and overclocking, not fx).

PS: Expect delayed answers from me - my main m15 going to battery replace. After 9 months after order.

crim2som, 2020-07-29:
**cmd Results for 2.** 
cd C:\alienfx-tools\alienfx-cli.exe
The directory name is invalid

C:\alienfx-tools\alienfx-cli.exe status
API version 5.2.0.0 detected
Devices found: 0

dir C:\alienfx-tools\alienfx-cli.exe
 Volume in drive C has no label.
 Volume Serial Number is 8CCF-E2C8

Directory of C:\alienfx-tools
 282,112 alienfx-cli.exe
 1 File(s) 282,112 bytes

**cmd Results for 3.** 
Do you want to test a custom device? (y/n)
y

Insert the product id of the device
0x5591

Is it a new alienfx device? (y/n)
 (everything after a Area-51 is considered new)
n
AlienFX device not found. Program terminated

Is it a new alienfx device? (y/n)
 (everything after a Area-51 is considered new)
y
AlienFX device not found. Program terminated

Please note that last entry of y or n gets same response. 

**Other**
AWCC has no FX tab, and Alienware - AWCC folder status, AlienFX 2.0 applications/files are not loaded/visible. 
Do not understand clarification that Alienfx 2.0 applications/files are not missing. 

Thank you for your endeavours. Kind regards.

T-Troll, 2020-07-29:
> C:\alienfx-tools\alienfx-cli.exe status
API version 5.2.0.0 detected
Devices found: 0

Great. Alienfx-cli found SDK, but no devices into the system.

> Is it a new alienfx device? (y/n)
(everything after a Area-51 is considered new)
y
AlienFX device not found. Program terminated

NOT good. It means problem into USB device driver/BIOS, not in AWCC!

It should be (Even WITHOUT AWCC at all):

> Insert the product id of the device
0x550
Is it a new alienfx device? (y/n)
 (everything after a Area-51 is considered new)
y
Please enter the name of the Alienware Pc/Laptop you are using
m15
Please make shure that if you have a laptop your touchpad is enabled
Press any key to continue . . .
Disabeling all leds...

In you case, it can't access device trough driver (driver is NOT part of AWCC, but Windows). Is PID correct?
For dirty try, i suggest to play with BIOS - upgrade/downgrade to other version. For more deep - let me think.

crim2som, 2020-07-29:
**Question** is PID correct?
**Answer** PID = 5591. This is how it was obtained:- 
Internet example of How to get the USB Vendor ID and USB Product ID from Windows Device Manager
[image]
We put USB into our Alienware 17 R5 Laptop to reveal USB Mass Storage Device properties and USB PID = 5591 
[image]

T-Troll, 2020-07-30:
No, it's not correct. Is't NOT mass storage, and VID should be 187C (Alienware).
Look here:
[image]

crim2som, 2020-07-30:
Hi @T-Troll thank you for the clarification complete with image. 
Seemingly our Alienware 17 R5 Laptop has no HID-compliant vendor-defined device. Not sure if this is to be expected. 
Is there another way to get PID?
[image]

T-Troll, 2020-07-30:
No HIDs? Hmm, if it's really like this, THAT'S a source of the issue. And, as i say, it's BIOS/Microsoft related.
To be sure - check "Consumer control device from you pic above.
Look for VID_187C.
Also check under "Unknown devices".
I check with my old M13R2 - it have the same structure, so seems like they don't change it between models.

crim2som, 2020-07-30:
1. Our 17 R5 image of HID drivers is correct. Scan for hardware changes, does not update anything. HID-compliant consumer control device, etc. driver properties do not have VID_187C.
2. There are no "Unknown Devices" or warning triangles in Device Manager. 
3. Windows Update is up to date. 
4. Please note that our 17 R5 Windows Update has installed Windows 10 Professional (x64) Version 2004 (build 19041.388). Not sure what Microsoft has changed. 
5. Is there another way to get PID?
6. If incomplete Device Manager drivers is the source of this issue, guess should install "HID-compliant vendor-defined device" driver. But internet search did not give this driver to install.

T-Troll, 2020-08-01:
Ok. In general - yes, AWCC can't find fx device because it not present in Windows at all.
3. As a first measure, try to DOWNGRADE BIOS to the OLDEST available (ok, as a first step - just reset BIOS settings to default).
4. If it not helps, try to DOWNGRADE Windows to any 18XX release (you can get ISO from MS website).
6. AWCC doesn't have device drivers for HID. It just use windows ones. So you can install or uninstall it.... doesn't matter.
If 3-4 not helps - you have hardware fault, and need to go to service.

crim2som, 2020-08-01:
Hi @T-Troll thanks for the update. 
3. 17 R5 out of the transit box had BIOS 1.8.0 with default settings. Recently installed BIOS 1.8.1 with default settings. For both BIOS versions, preinstalled AlienFX devices were not detected by AWCC. The BIOS version makes no difference. 
4. Many Dell Alienware Community users/customers report AWCC does not detect AlienFX devices and attribute this failure to incorrect Dell code, with proviso wait for next release to get AlienFX working again. Our licenced windows applications are dependant on not trying different versions, so will stick with the Win10 Pro x64 that came preinstalled on the 17 R5. 
6. Dell diagnostics say that 17 R5 system board hardware test is pass. There has been speculation that AlienFX 2.0 application is just locked into dark mode (by incorrect Dell code). The AWCC installer does not provide AlienFX Tester, can this application be provided from your Alienware m15?

T-Troll, 2020-08-02:
3. Sad. Easy way to fix.
4. Yes, they report. Why do you think you have the same case?
5. On my gear, diag did not check FX at all, only main (Intel) motherboard functionality. So question is what is you gear motherboard check includes? Is lights changed/blinks during the test?

crim2som, 2020-08-02:
Hi @T-Troll, Dell do not provide the hardware AlienFX Test installer (v1.0 and/or v2.0) which supposedly comes with the AWCC Suite installer, but not for our 17 R5. We do not know if failure to install any of the AlienFX applications is a hardware and/or software (code) fault. Speculate the 'easy' next step is to install the AlienFX Test application, which is why we ask for help.

T-Troll, 2020-08-02:
You can try other ways as well, but, at my point of view, we discover it's a hardware/Microsoft driver problem.
alienfx-leds-tester working on low level, just like awcc, so if no device - both fails.
I don't know what else i can do.

crim2som, 2020-08-02:
OK @T-Troll thanks for your assistance. Kind regards.

T-Troll, 2020-08-03:
You are welcome. Please keep me informed if you have some updates.

closed 2020-08-03 by T-Troll