T-Troll/alienfx-tools #4: Question

PhSMu, 2020-08-24:
Excuse me, but I have a Dell G5 (PID is 550, and it has 4 Keyboard Zones but no other lights and I want this app to work standalone. Which files does this need to work without keeping AWCC (os is this impossible?), and should I simply copy them to the program folder or do they need to be elsewhere?

[image]

T-Troll, 2020-08-25:
Funny.... You have the same PID as mine m15R1, and Dell SDK working as well.
For keep this app working, for now you need lightfx.dll in windows/system or the same folder.
But it's based on AWCC service (maybe other service name for Dell), and can't work if it's stopped.

I have a good news for you anyway. I'm modify AlienFX_SDK (low-level access library) to new gear, including ours, and will port all apps to it soon. It will not require anything, except windows driver ^_^.

By the way, can you please check - does it works on you Dell? Just download sample probe app from [here](https://github.com/T-Troll/AlienFX-SDK/releases), then run and tell me:
1. It's output (device version, id, ready state)
2. Does it change you lights correctly? (Press Enter to change next light, color is always green, i probe 16 lights)

T.I.A.

PhSMu, 2020-08-25:
[image]
0 is Keyboard Left, 1 is Keyboard center, 2 is keyboard right, 3 is numpad, the rest do nothing, so far as I know. And yes, I've found I need to install the AWCC package from dell, but can delete the UWP app afterwards and the command line keeps working.

T-Troll, 2020-08-25:
Great!

I know about AWCC bug with CLI (see readme), you need app working to keep colors (or AWCC reset it to default),
I see you interested in CLI mostly? By the way, do you need only change light colors, or different light modes as well?

An issue for low-level access is it's not possible to define how many lights you have and which one is which - Dell SDK uses config files for it, not driver access. And IDs in Dell SDK and driver call are different. Now i think about how to solve it.

Anyway, wait for the new release - i will make CLI keeping colors with AWCC active, as well as low-level support.

crim2som, 2020-08-25:
Our Alienware 17 R5 laptop says:- 
[image]

PhSMu, 2020-08-25:
Well, I was mostly thinking of something I could use to change colors, yes, I admit I'm very salty at dell that I needed to install a 700mb software to get the functionality most of their notebooks do with FN+F10 and have said program stay resident in memory afterwards.

I remember using effects and the like back in my old M11X-R2, but I don't really use them these days, mostly because the old AlienFX made the transitions smooth, and the new one plays them and then loop back to the start, weirdly enough.

Now another question: If its low level hooks, we're basically bypassing the API/drivers. Does that mean this would finally work on other OSs (I mean, if you share these findings or maybe through Wine-like emulators for Linux and such?

T-Troll, 2020-08-25:
@crim2som please open new issue, this one for other. BTW, did you try my probe app mentioned above?

Aye, Dell/AW become a bit weird last times. They still don't fix a lot of bugs in their SDK, that's why i start digging low-level, but make some changes making CLI useless (and, seems like, any app need to stay in memory or colors reset to defaults). 
Having AWCC is not a big issue for me, i use most of it's functionality (profiles, fusion, overclock).

Yes, low-level API will work in any OS, only finding device ID and init is different. Anyone can use my source code.
I still not integrate it (have a fight with GitHub limits today), just make a possibility to use cli for user color changes. But it will require to stay in memory if AWCC installed as well.

PhSMu, 2020-08-25:
> Our Alienware 17 R5 laptop says:-
> [image]

This is weird. On my G5, it works flawlessly. Maybe there's a big difference between old WinAPI and the new UWP(cr)API?

[image]

T-Troll, 2020-08-25:
He is use PowerShell, not CMD.
For power shell right command is `./alienfx-cli.exe status`

PhSMu, 2020-08-25:
Ah, powershell. Don't use it, I'm oldschool like that. :P Been messing with PCs since Windows 95 (And have used them before Windows itself) despite not being all that old.

One advantage of the CLI being a service is that you could, theoretically, use it to bypass the limitations of current LightFX, like making loops smoother with for loops with increasing or decreasing values, and even more than two colors. The thing though, is that would use only the "SET" function of CLI, and it would be a service as well. Don't know if this would be good or bad though.

In one hand, one more resident to SysMem, on the other it is lighter than the previous occupant of that seat. On the third mutant tentacle, it is open source so we can audit the code!

And yes, G-Series only get RGB Keyboard, no light strips or anything. It's probably that.

T-Troll, 2020-08-25:
I'm a bit older, i start from Atari & ZX Spectrum.

Current cli utilize common Dell SDK, so it have even more limitations then AWCC (f.e. you can't change the color of the power button). And there are some bugs with rapid changes - that's why i dig low-level.

You idea is noted, i think about effects app - but for really cool effects it should stay in RAM as well. Current low-level can do more, then SDK, but not much more (no other effects, but you can control speed). As i say, i want it for speed and less bugs i need to fight.

My target for this project was return back some features i enjoy at my old 13R2 - like audio/video lights. But i think about more.

PhSMu, 2020-08-25:
Oh, don't worry about me. Set a static color and change brightness is more than enough for me!

crim2som, 2020-08-26:
Hi T-Troll, happy to open new issue. 
Thank you for your continuing willingness to help. 
Kind regards. 

From: T-Troll
Sent: 25 August 2020 11:28
To: T-Troll/alienfx-tools
Cc: crim2som; Mention
Subject: Re: [T-Troll/alienfx-tools] Question (#4)

@crim2som please open new issue, this one for other. BTW, did you try my probe app mentioned above?
Aye, Dell/AW become a bit weird last times. They still don't fix a lot of bugs in their SDK, that's why i start digging low-level, but make some changes making CLI useless (and, seems like, any app need to stay in memory or colors reset to defaults).
Having AWCC is not a big issue for me, i use most of it's functionality (profiles, fusion, overclock).
Yes, low-level API will work in any OS, only finding device ID and init is different. Anyone can use my source code.
I still not integrate it (have a fight with GitHub limits today), just make a possibility to use cli for user color changes. But it will require to stay in memory if AWCC installed as well.

T-Troll, 2020-08-26:
> Oh, don't worry about me. Set a static color and change brightness is more than enough for me!

First, thank you for feedback and testing.

It's my pet project, and hardware is not so popular, so i luck of feedback.
Feel free to share any idea and suggestion, i'll try to help.

PhSMu, 2020-08-26:
well, another question then, how do I compile this for testing something. I plan on trying to make an auto-it interface for the CLI when I have the time, and having a simpler 'service' to run would help immensely in that respect.

PhSMu, 2020-08-26:
So, even a batch driven set of commands like this:

---------- // ---------- // ----------
:start
%~dp0%~dp0alienfx-cli set-all=64,128,255,255
%~dp0alienfx-cli set-all=64,128,255,240
%~dp0alienfx-cli set-all=64,128,255,224
%~dp0alienfx-cli set-all=64,128,255,208
%~dp0alienfx-cli set-all=64,128,255,192
%~dp0alienfx-cli set-all=64,128,255,176
%~dp0alienfx-cli set-all=64,128,255,160
%~dp0alienfx-cli set-all=64,128,255,144
%~dp0alienfx-cli set-all=64,128,255,128
%~dp0alienfx-cli set-all=64,128,255,112
%~dp0alienfx-cli set-all=64,128,255,96
%~dp0alienfx-cli set-all=64,128,255,80
%~dp0alienfx-cli set-all=64,128,255,64
%~dp0alienfx-cli set-all=64,128,255,48
%~dp0alienfx-cli set-all=64,128,255,32
%~dp0alienfx-cli set-all=64,128,255,16
%~dp0alienfx-cli set-all=64,128,255,0
%~dp0alienfx-cli set-all=64,128,255,16
%~dp0alienfx-cli set-all=64,128,255,32
%~dp0alienfx-cli set-all=64,128,255,48
%~dp0alienfx-cli set-all=64,128,255,64
%~dp0alienfx-cli set-all=64,128,255,80
%~dp0alienfx-cli set-all=64,128,255,96
%~dp0alienfx-cli set-all=64,128,255,112
%~dp0alienfx-cli set-all=64,128,255,128
%~dp0alienfx-cli set-all=64,128,255,144
%~dp0alienfx-cli set-all=64,128,255,160
%~dp0alienfx-cli set-all=64,128,255,176
%~dp0alienfx-cli set-all=64,128,255,192
%~dp0alienfx-cli set-all=64,128,255,208
%~dp0alienfx-cli set-all=64,128,255,224
%~dp0alienfx-cli set-all=64,128,255,240
goto :start
---------- // ---------- // ----------
gives me some measure of breathing effect (choppy, but I can't use less than a second wait in batch like AutoIt does) instead of increase and turn off suddenly like AWCC does.

I do have to check how to make an autoit tray app, but I know it is possible (AutoHotkey is/was made that way), and then the minimum delay becomes that of your app executing.

Again, I'm not complaining, just trying to make a parallel version of the AWCC FX without the UWP baggage, but would it be faster and possible to have / compile a lighter version of the AlienFX-Cli with only the 'set-all' command (possibly with a "SetFX R G B BR" syntax)?

T-Troll, 2020-08-27:
Ok, first things first.
To build you own app, you need to do:
1. Install Microsoft Visual Studio (free "Community edition" version is OK), at install select "C++ Desktop development" and "GitHub support".
2. Start Visual Studio, select "Git->Clone Repository", and enter my repository URL (see in "Code" tab here, green "Code" button).
3. Build project in Visual studio.
If you want to play with low-level SDK, you need also clone my another repo, and install Microsoft WDK (Windows Driver Kit) - i use it for low-level access.

alienfx-cli interface is defferent. Don't need massive calls, just list all commands you need into one call - `alienfx-cli <command1><command2> <command3> .... <commandN>`. After testing, add "loop" command to the end of this string - it will keep all commands running, so AWCC can't switch colors back.
It's not possible to send CLI application to tray, you need application window for this, so it's a choice - cli or tray.

T-Troll, 2020-08-27:
I publish the new release.

Especially for you, new set-tempo command added to alienfx-cli. By SDK description, it should change the tempo of the next effect, setted by set-action, set-zone-action, but, as usual, it's buggy at Dell SDK.

Also, i solve an issue about light names in low-level SDK (see alienfx-probe), so next alienfx-cli release will support low-level. Then i use this command as a pause between others (and, also, add action time in action commands).

PhSMu, 2020-08-27:
Sorry for the amateurish questions and thanks for the quick answers! I'm halfway done with a quick GUI (in autoit) to change colors without effects, and will add the code here for you if you want to put it or play with it.

And yes, the idea of a low-level service was to create a small resident app (AWCC resident services sum almost 80mb on my machine, I was thinking of a 100-250kb app) and use that to create and manipulate the light effects.

T-Troll, 2020-08-27:
Uff.... Design UI sketch for me, it's not a big deal to implement it. You can use `alienfx-ambient` as an example.

Footprint will be larger - about 1-2M - thanks to VCRedist.

As i say before, i have different reason (i need AWCC running any case) - SDK updates are sooooo slooooow, so it's hard to implement some things.

PhSMu, 2020-08-27:
[LiteFX.zip](https://github.com/T-Troll/alienfx-tools/files/5136599/LiteFX.zip)
Here, this is a small app (icon was taken from the internet), which does a 'set-all' in the chosen color. Its a very barebones thing, but what I managed to make today. I'll probably play more with this on the weekend.

T-Troll, 2020-08-28:
I see. Simple GUI, i can do the same (with keeping colors and minimize to tray) if you wish. What about breathing and other effects?

BTW, i release 0.6.0 with low-level API support. Well, in fact alienfx-cli and alienfx-probe supports BOTH api.
So you can try and remove AWCC if works at low-level for you (should, we test it before).

PS: Read readme before use - it's a LOT of changes and limitations for now!

PhSMu, 2020-08-28:
no need for you to waste time writing an UI. Like I've said, that was kind of a small project of mine and I intend to make it work with everything your program can do... eventually.

The thing is that if you do a minimize to tray function, you should split the service and the UI itself, so that the service doesn't eat too much memory.

T-Troll, 2020-08-28:
Ohhh... Spanish? Funny.

Thanks, i fix it tomorrow (Friday night here right now...), i know there it come from.

In Windows, minimize to tray is not possible for console apps, so we need GUI app to doing this. So, if AWCC installed, i need GUI to keep colors, but if you plan not to use it - low-level will keep it until reset or game.

PhSMu, 2020-08-28:
> Ohhh... Spanish? Funny.

Brazilian Portuguese, actually, and sorry for that. It roughly translates as "Alienfx-cli.exe - Application Error: The 0x0000000000000000 instruction referenced the 0x0000000000000000 memory block. The memory couldn't be written. Click OK to close the program. Click Cancel to depure the program."

T-Troll, 2020-08-29:
Don't need to sorry, English is not my native as well.

Check [0.6.2](https://github.com/T-Troll/alienfx-tools/releases), exception should gone away.

PhSMu, 2020-08-29:
> Check [0.6.2](https://github.com/T-Troll/alienfx-tools/releases), exception should gone away.

I don't think that is the case. Maybe there is some conflicting code between the high level and low level apps. I'm not use if its possible, but split the app into AlienFX-cli and AlienFX-LL(or some other name, I'm not the best with those) with the later not checking for said DLLs?

T-Troll, 2020-08-29:
No, it's my fault - i'm not isolate high and low in set-all correctly. Fixed.
See 0.6.3 (will be in some minutes) - it should fix it.

About brightness - did you read readme? Yes, low-level doesn't support it (high-level too, in fact, just modify color by it), so it's ignored.

PhSMu, 2020-08-29:
> About brightness - did you read readme? Yes, low-level doesn't support it (high-level too, in fact, just modify color by it), so it's ignored.

ah, my derp then. Must have overlooked that. Yep, now it works!

T-Troll, 2020-08-30:
Glad to hear. Let me think about GUI app.

PhSMu, 2020-08-30:
I don't think you need to worry about that. To me you can consider this issue solved.

PhSMu, 2020-08-30:
Ok, Three things I thought off, playing with the software today:

1: I did notice something odd: If using AlienFX-Probe to set the names of my Keyboard Zones, if there is no AWCC installed, they'll not be set.

[image]
[image]

2: Another useful info, On G5, at least, the FN+F10 Key combo, used to change keyboard brightness, is tied to AWCC, but if it is uninstalled, it reads as F18 (00_69) as an input code, not sure if that is useful, but seems like it would be.

3: Is there a way to read which color is currently set?

T-Troll, 2020-08-31:
1. As far as i see they set, but dropping by last space. Looks like a bug, let me check.
2. I have Fn+F12 instead, and it works like on-off switch. I think this feature will be a part of alienfx-gui app, seems like i need to make a kind of light-weighted AWCC. ^_^
3. I din't find it yet, and, as i see, AWCC does not read light colors as well (and it doesn't work in high-level API as well - see readme).  So i can't find right command for it without full specs from Dell ^-(
But for GUI app i can remember light colors inside app and restore it upon restart.

PS: if you want to play with this at low-level - install USBPcap and Wireshark (to view logs) - you can intercept any AWCC request to the USB device and maybe found something interesting. F.e. i din't catch how to set macro for now, as well as how to control effect length (but i know there it stored).

T-Troll, 2020-08-31:
1. Strange... Can't reproduce this bug:
`C:\work\DevStudio\alienfx-tools\x64\Debug>alienfx-probe.exe
Probing low-level access... PID: 550                                                                          
Probing Dell SDK... 
Lights found for Alienware m15: 
Light #0 - KB Left
Light #1 - KB Center-Left
Light #2 - KB Center-Right
Light #3 - KB Right
Light #4 - Alienhead
For each light please enter LightFX SDK light ID or light name if ID is not available
Tested light become green, and change color to blue after testing.
Just press Enter if no visible light at this ID to skip it.                                                  
Testing light #0: Back logo                                                                                   
Final name is Back logo, Testing light #1: Power Button                                                       
Final name is Power Button, Testing light #2: 0                                                               
Final name is KB Left, Testing light #3: 1                                                                    
Final name is KB Center-Left, Testing light #4: 2                                                             
Final name is KB Center-Right, Testing light #5: 3                                                            
Final name is KB Right, Testing light #6:                                                                     
[....]                                                                                  
Skipped. Testing light #f:                                                                                    
Skipped.
C:\work\DevStudio\alienfx-tools\x64\Debug>alienfx-cli.exe low-level status                                    
Current device ID: 550                                                                                        
Device ID#550, Light ID#0, Name: Back logo
Device ID#550, Light ID#1, Name: Power Button                                                                 
Device ID#550, Light ID#2, Name: KB Left                                                                      
Device ID#550, Light ID#3, Name: KB Center-Left                                                               
Device ID#550, Light ID#4, Name: KB Center-Right                                                              
Device ID#550, Light ID#5, Name: KB Right`
But fix another one.

PhSMu, 2020-08-31:
> * As far as i see they set, but dropping by last space. Looks like a bug, let me check.

Ah,  sorry, I did not clarify. Those were the names I have assigned before.

> * I have Fn+F12 instead, and it works like on-off switch. I think this feature will be a part of alienfx-gui app

That's the thing, at first I thought it went FN+F1 = F13, FN+F2= F14 and so on, but it seems that is not the case, those are arbitrarily decided. So the actual 'base key' doesn't quite matter for the keycode of the FN+Fx combo. How I wish the old M11x-R2 was operational right now so I could test this...

> * I din't find it yet, and, as i see, AWCC does not read light colors as well (and it doesn't work in high-level API as well - see readme).  So i can't find right command for it without full specs from Dell ^-(

Ah, that is how it is then. My bad, I had no idea.

> * seems like i need to make a kind of light-weighted AWCC. ^_^

Again, don't mind me, I'm just throwing some of my findings your way as I try to make enough sense of C++ to get what you're doing and how it works!

To be fair, if you did, and somehow the resulting app compiled and ran on linux, you'd be miles ahead of any competition, as the alternatives there seem to hang on anything UWP based. I tried the handful that exist, and they didn't work, some even crashing the system. Other than that, my G5 runs buttery smooth fine on Linux Mint 20.

T-Troll, 2020-09-01:
Let me clarify some things.

It's not for you, you just test and bring some ideas. If i find it useful - i'll do for all.

alienfx-cli, as well as low-level SDK, can be ported to Linux easily, you just need to change some windows-specific functions for USB device access and discovery. Unfortunately, i can't do it myself - i only have Linux in WSL, and don't plan to migrate to standalone install - it's enough for my needs. But i can help if anyone will decided to port SDK to Linux.

PhSMu, 2020-09-01:
It makes me a lot more comfortable if you're taking my bystander suggestions as they are. ^_^

T-Troll, 2020-09-06:
I made set-action working without AWCC as well. But not 100% - their sdk more limited, then hardware. Command format changed (now it's possible to define not only tempo, but also phase time. In fact, 2 of them, but i didn't push it in UI now).

PhSMu, 2020-09-06:
not yet, there have been issues on my end, completely unrelated to PC (with my tablet, in fact), which I am fixing.

T-Troll, 2020-09-22:
I release `alienfx-gui` (alpha now, functionality is limited) - it shows what our hardware really can do. You can play now and share you opinion. Next will be names setup (like probe), events (power, hdd, ram, cpu, gpu, notifications, ???), tray operations (on/off/dim, etc)... maybe more.
I plan to stop use AWCC after i finish it.

PhSMu, 2020-09-22:
You really did it! You madman!

Yes, it is working! ...sorta.

When I first open it, it assigns random colors to my 4 keyboard zones, and when I set effects it plays them so fast the system kind of mixes the two colors together? I'm not sure what is happening here, and will have to try more later today.

I'm not good at programming, just passable at the logic behind it, but I can help develop an UI if you want to.

Also, my personal recommendation is to have a text file .ini or .cfg with the settings for the program stored so they aren't lost, and have the  actual 'program' (i.e. the part that interfaces with the hardware, or the cli) and the UI to be different things. Modularization, so you can find errors and troubleshoot it more easily.

T-Troll, 2020-09-23:
It's an ALPHA, just demo, in fact. I write it because AWCC have a bug slowing down my gear if i use hardware control, so i want to uninstall it.

Let me explain.

First, about colors and effects. There are 2 phases of each light (hardware), you can set any TYPE for any of them. Now it's set the same (just UI). Second, play with speed and length values - one for phase 1, second is for phase 2. For slow blink/morph increase speed, for longer blink time/morph time increase length. For blink, it first blink first color to black (defined speed and length), then second (with second settings). For morph, first it's morph from first color to second (defined speed), then from second to first (defined as well).
Good news it's all are in hardware, so it will works until reboot even if app not active (and AWCC not reset it).

I have settings save/load support (in registry), but it's disabled now for the simple reason - app in active development, so stored data format not stable, i change it from time to time. Next release will do it.

Also, i'm already have minimize to tray and tray menu now, it will be available next release as well.

T-Troll, 2020-09-24:
New release - now possible to set 2 different actions at one light circle (play with it!), tray functions and config save/load.
About GUI - share you vision, if you want (check events and setup pages as well, although it doesn't work for now), i have some issues and appreciate you help.

PhSMu, 2020-09-24:
Man you work fast!

On a quick test (I'll probably only have enough time to thoroughly test it on the upcoming weekend) it seems to work fine, except pulse which speed seems to be inconsistent with mine (it turns out to be blinking furiously instead of fading in and out).

On Gui, I'd have a device selector at the starting page not only on the settings (although for me it'd do nothing as I only have one. XD), and RGB values instead of windows color picker, with maybe (if not too much work) a quick menu of predefined colors.

Another useful thing to implement would be a hotkey (which probs have to be 'configurable' as dell doesn't make it a standard :/, either to pull it from tray, or to cycle through settings (on-off or maybe three levels like: Bright>>Dim>>Off>>Bright...)

PhSMu, 2020-09-26:
[AlienIcons.zip](https://github.com/T-Troll/alienfx-tools/files/5287056/AlienIcons.zip)

Here's a small, one morning work of mine, for you: Some icons for the project, although I'll likely pick this later and pop some proper .ico files and use better re-scaling. I made them into Corel, so if there's any other color you want or if you want to mess with the effects and colors by yourself, I can send you the .cdr file.

T-Troll, 2020-09-27:
> pulse which speed seems to be inconsistent with mine
Use morph to black if you want breathing - pulse is not about it. Also, try to play with length - it can provide some morph emulation.

> I'd have a device selector at the starting page not only on the settings 
I'm based on old AlienFX SDK (from Aurora project) to keep compatibility with older devices. One of the issues of it - it's only possible to control one device in a time (Singleton class.... Guys was strange), so it's just useless for now. What's why i apply current UI design.

> and RGB values instead of windows color picker
I have it in Haptics (BTW, IP Address control fits well for RGBBr), but you know what? I can't use it after i have color picker - it's much faster all the time. In picker, you can enter RGB values as well, if you need it (rare, in fact). What i plan to implement is a custom colors for picker - so you can save you favorite color set for quick settings (yes, as you said). It's already implemented, but not stored yet.

> Also, a 'set-all' option
Nice idea... I can do it (and a little more) trough implementing multiply lights selection and set for all of them. Not so hard, in fact.

> Another useful thing to implement would be a hotkeys
Already done but not UI part.

>  Some icons for the project
Ohh... I not use CorelDraw from the middle 90s... ^_^
Thanks, looks good. The issue here i'm made not the same icons for a reason - f*cking copyright. I don't want to mess with Dell in this. So it should NOT looks exactly like their, but share the same idea. BTW,,, Alien... What do you think about using UFO for an icon? ^_^

I finish the setting page (you can use it instead of Probe app) and power control for now, working with performance meters (my gear doesn't have HDD light so i want to use one of other light for it). I plan to release new version as soon as i finish this, then i can concentrate for UI and bugfix.

T-Troll, 2020-09-28:
I release 0.8.2 - with settings and events. Now i can stop AWCC at my gear and switch to it.
Maybe also i need profiles and autoswitch/disable for some apps.... Let's see.
Some features still not finished (see issues in release notes), but i think i do it this week.

PhSMu, 2020-09-28:
[OvniIcons.zip](https://github.com/T-Troll/alienfx-tools/files/5293526/OvniIcons.zip)
And as you asked, here are the Ovni Icons!

[image]
[image]
[image]
[image]
[image]
[image]

T-Troll, 2020-09-29:
Wow! Good idea, i like this light effects. But looks a bit like hat, not UFO. BTW, max icon size is 256x256px.
PS: I publish 0.8.3, but check issues (i meet hardware limitation). The rest is working (i even made default color set, so you don't need to run gui app if you don't use events), now i want to polish GUI, as we discuss before. Will be glad to hear other ideas from you as well.

PhSMu, 2020-09-29:
Technically speaking, what's the difference between length and speed? I mean, it'd probably simplify the usage if you have one slider for the duration of the effect, instead of both. Also, maybe that popup that show a number on hover or a number field, because I was trying to sync all my lights and that involved a lot of > keyboard presses and counting. XD

> But looks a bit like hat, not UFO

Well, UFOs _are_ hat shaped, but any ideas on how to fix that impression?

> I publish 0.8.3, but check issues (i meet hardware limitation). The rest is working (i even made default color set, so you don't need to run gui app if you don't use events), now i want to polish GUI, as we discuss before. Will be glad to hear other ideas from you as well.

I still haven't figured how the performance monitors work, but it might be that my windows is a bit too tweaked for it to run on my PC.

And it occurred to me that if you can set a third color for morph effect, you could emulate that rainbow effect of AWCC, by morphing to A > B > C

Other odd thing I noticed is that the app doesn't scale well on monitors with 125% DPI.

T-Troll, 2020-09-29:
> Technically speaking, what's the difference between length and speed?

It's a big difference. Speed is a blink speed (or color change slope) for pulse and morph. Length is a time change phase are stay in play. So one is about speed inside phase, other is about length of phase.

> Well, UFOs are hat shaped

Not exaclly, a bit more complex (well... i see one before `^-^`). Maybe change bottom side to less opened or adding land gear or adding some windows can help?

> I still haven't figured how the performance monitors work

Simple, as usual. You have one color (first one from Color) and second one (from this page). Loading measured from 0% to 100%. then used to blend them (0% - first one, 100% second one). So, try to put, f.e. green and red for 0 and 100% cpu then run a game. For Indicators, all above 0% simple switch to second color. I use it for HDD (NB: Network and GPU still not working), reflects nice.

> And it occurred to me that if you can set a third color for morph effect
 
3rd color not supported by hardware.  Then i switch to "3td" - i switch light manually, in fact. 5 fps rate, if you are interested.
So rainbow is possble, if you configure out how to implement it in UI (only if app running, instead of hardware).

> Other odd thing I noticed is that the app doesn't scale well on monitors with 125% DPI.

Looks strange. I have 2 monitors, one is 4k with 250% scale and check for it. But it's only for 0.8.3 (see release notes), 0.8.2- have this issue, right.

PhSMu, 2020-09-29:
> Not exaclly, a bit more complex (well... i see one before `^-^`). Maybe change bottom side to less opened or adding land gear or adding some windows can help?
> 
> >

will do!

Also, I remember my old M11X-R2 Keyboard did have an UFO Key to open command center:
[image]

> Looks strange. I have 2 monitors, one is 4k with 250% scale and check for it. But it's only for 0.8.3 (see release notes), 0.8.2- have this issue, right.

[image]

> Simple, as usual. You have one color (first one from Color) and second one (from this page). Loading measured from 0% to 100%. then used to blend them (0% - first one, 100% second one). So, try to [...]

Oh, so maybe it wasn't working because I did not set a base color. Stupid me! *bangs head on desk* no more staying up too late playing gatcha games! Ok, I'll test and report back. Theoretically (once these issues are solved) it'd be possible to do a CPU-Mem-GPU-HDD with my four zone keyboard then?

T-Troll, 2020-09-29:
> Also, I remember my old M11X-R2

Fancy, i have M14x as my first one. Yes, something like it (but not same).

> Stupid me! bangs head on desk

No, it means i didn't make simple enought UI for this. In fact, under the hood, all modes have 2 colors in set, but i only use 2nd for events to avoid set the first 2nd time. Need to clarify here, let me think.
More generally, i don't sure about logic at this point - you can use multiply modes/events at once, but how they should work in this case? Still thinking about it as well.

> Theoretically (once these issues are solved) it'd be possible to do a CPU-Mem-GPU-HDD with my four zone keyboard then?

Yes, i use CPU indicator right now. But only CPU, RAM and HDD supported in current release - GPU and Network require summary and normalization accross multiply devices i didn't implement yet. Ohh... And temperatures! I need them - m15 quite hot under loading.

PS: Hmmm... high DPI looks well for me, let me check deeper tomorrow (it's a deep night here).
PPS: Check readme `^_^`

PhSMu, 2020-09-29:
[image]

is this one any better?

T-Troll, 2020-09-30:
Top one are ok, can you please render 256x256 png transparent with 3 different colors (for GUI, Ambient and Haptics)? I change current one to it!
Botton looks not like a common hat, but like Sombrero this time `^-^`. Thanks anyway, but let us stay at first one.

I release 0.8.4, now all counter works (CPU, RAM, GPU, LAN, HDD), as well as local and global keyboard shortcuts - see release notes. BTW, feel free to request global/local hotkey for a function you need, i'll add it.

PhSMu, 2020-09-30:
> Top one are ok, can you please render 256x256 png transparent with 3 different colors (for GUI, Ambient and Haptics)? I change current one to it!
> Botton looks not like a common hat, but like Sombrero this time `^-^`. Thanks anyway, but let us stay at first one.

lol Mexican aliens? well, someone has to have taught the guys their math!

Jokes aside, here it is:

[UFO-Icons.zip](https://github.com/T-Troll/alienfx-tools/files/5304968/UFO-Icons.zip)

> BTW, feel free to request global/local hotkey for a function you need, i'll add it.

In my machine the Illumination On/Off Hotkey is 'F18' (FN+F10, you can check it with a nifty piece of software called sharpkeys). Is it possible to make it a Bright-Dim-Off cycle on F18?

And now to play with the software... after breakfast. (Timezones...)

T-Troll, 2020-09-30:
> Jokes aside, here it is:

Thanks, i'll add it after dinner. `^_^`

> In my machine the Illumination On/Off Hotkey is 'F18' (FN+F10, you can check it with a nifty piece of software called sharpkeys). Is it possible to make it a Bright-Dim-Off cycle on F18?

Check attach.
[alienfx-gui-update.zip](https://github.com/T-Troll/alienfx-tools/files/5305084/alienfx-gui-update.zip)

T-Troll, 2020-10-01:
I release 0.8.5 (fix some bugs and add temperature monitoring). It's not like i send you yesterday, please update.
Also, i check you icon... I think it needs a bigger (and, maybe, more simple) symbol and more contrast, not looks good now:
[image]

PhSMu, 2020-10-01:
Will do!

PhSMu, 2020-10-01:
[UFO-PNG.zip](https://github.com/T-Troll/alienfx-tools/files/5311614/UFO-PNG.zip)
[UFO-ICO.zip](https://github.com/T-Troll/alienfx-tools/files/5311634/UFO-ICO.zip)

I reworkerd the transparency and some shape nodes which got out of place (sorta). Is it any better?

[image]

PhSMu, 2020-10-01:
On the UI, it occurs to me to ask why cant 'settings' be under the menu, and 'events' a type of effect. Something like this:

[image]

Where Device Selection and Mode are comboboxes, and instead of sliders, 'Effect Speed', I think should be renamed 'Transition Speed' and 'Effect Length' or 'Effect Duration' and could use Number Edit controls (that type of thing which has a edit field but aso up and down arrows on the side?)

I'd also move Events there as a 'mode', and you could built your sounds and video modules in here as well or just add shortcuts?

One interesting thing which can be done is add 'Breathing Mode' like this: Morph Colors would go 1 > 2 > 3 > 1 > 2 > 3 While Breathing would go like a pendulum 1 > 2 > 3 > 2 > 1 > 2. Of course, the difference would be moot with 2 colors.

Another thing to keep in mind, maybe dim the controls that have no effect for the current mode?

Lastly but not least important, Keep in mind I'm not a programmer, so I have no idea how easy or hard are these things to make/change! So it it's too hard or bothersome, please ignore my suggestion/rambling!

[74 more replies]

closed 2021-05-16 by T-Troll