T-Troll/alienfx-tools #6: Can't init AlienFX v2.0 at some rig (e.g Alienware 17 R5)

crim2som, 2020-08-26:
AlienFX v2.0 code is used to enable the LED lights on the current versions of Dell or Alienware computers. The Alienware Command Center (AWCC) install application's code fails to handshake with the EEPROM code on the system board. Consequently, the AWCC install application refuses to install the AWCC AlienFX v2.0 applications and there is no AWCC FX tab. The Dell or Alienware LED Lights remain dark and inaccessible. However, alienfx-tools could install code to handshake with the EEPROM code on the system board and get the AlienFX LED Lights to work. It would be great if the keyboard lights could be switched on and off.

crim2som, 2020-08-26:
Our Alienware 17 R5 Laptop Operating System is Windows 10 Professional (x64) Version 2004 (build 19041.450). All Windows Updates have been installed. BIOS is the most recent Alienware 17 R5 v1.8.1. Alienware Command Center (AWCC) application v5.2.91.0 is the most recent install application. All Alienware updates have been installed. SupportAssist says System Board hardware test is pass.

crim2som, 2020-08-26:
Alienware 17 R5 says
[image]

crim2som, 2020-08-26:
Hi @T-Troll is there any prospect of alienfx-tools code being updated to handshake with AlienFX v2.0 EEPROM code on Alienware 17 R5 System Board?

T-Troll, 2020-08-27:
Check [This probe tool](https://github.com/T-Troll/AlienFX-SDK/releases), it uses low-level access and try to locate device. It doesn't need Dell SDK,
If it doesn't works - you have hardware/BIOS fault, and need system repair.

crim2som, 2020-08-27:
Hi @T-Troll  run AlienFX Sample App.exe gives no response. Open cmd and AlienFX Sample App.exe status says 'AlienFX' is not recognized as an internal or external command, noperable program or batch file.
[image]
Is this the "If it doesn't works - you have hardware/BIOS fault, and need system repair."?

T-Troll, 2020-08-27:
App name have have spaces - make it all in quotes. And it doesn't require parameters.

crim2som, 2020-08-27:
Hi @T-Troll thank you for clarification. Have renamed "AlienFX Sample App.exe" to AlienFXProbe.exe. Open cmd and repeat with new name. C:\AlienFXProbe>AlienFXProbe.exe status says PID: ffffffff. Is this good or bad?
[image]

T-Troll, 2020-08-27:
Bad. You have no AlienFX device in system. I think you need repair it.

T-Troll, 2020-08-28:
So, as i say before, try to downgrade bios, if it does not help - go to repair.
I can't help you with this case, sorry.

crim2som, 2020-08-30:
Hi @T-Troll, @RealDeuce used your AlienFX-Probe application yesterday, and both of us do not understand the updated message that says "please check device manager". What are we looking to find or not find in device manager?
[image]

T-Troll, 2020-08-31:
Probe app try to locate any Alienware-manufactured device into the system.
If no device found, it returns ffffff and tells about it.

So, being simple, it's what i'm talking about - you have no Alienware device recognized by Windows.

AWCC using the same low-level interface, so results for Probe and AWCC will be the same.

crim2som, 2020-08-31:
Hi @T-Troll thank you for the AlienFX-Probe's "please check device manager" clarification. RealDeuce and I both have Alienware 17 R5 Laptops, we both have Alienware hardware, but our AlienFX LED Lights stopped working after an AWCC update. 

When the AWCC application is installed, so is the AWCC Driver in Device Manager. From what you have imparted, the AlienFX hardware makes use of USB drivers in Device Manager, but does not explicitly say which USB driver it is using. 

Hence the AlienFX-Probe question: What are we looking to find or not find in Device Manager?

(The AWCC Home and FX tabs are "missing" and AlienFX v2.0 applications are not installed. Postulate that the AWCC code does not handshake with the AlienFX EEPROM code on the System Board. Rollback of the AWCC version does not re-enable AlienFX Lights.)

T-Troll, 2020-08-31:
We are running circles a bit.
Both AWCC and my probe looking for USB HID with VID=187c (Alienware) and any PID.
If no device found = light system not present into Windows and lights not working.

NB: AWCC also provides firmware update for FX device (check C:\Program Files\AWCC\Firmware). Seems like they soft-brick you device into the one of updates, so it's no longer accessible. It's not possible to rollback device firmware if device not present into the system....
MAYBE, just MAYBE, BIOS update can also alter it, fixing broken firmware.
Another MAYBE - try to force-clear all EEPROM (open case, remove main battery, remove BIOS battery if present, push Power button for 30 sec, assembly back).

If it doesn't help - no one can help you, but Alienware repair specialists.

closed 2020-09-22 by T-Troll