T-Troll/alienfx-tools #7: Ambient lighting wont work

bagorocks, 2021-01-18:
I have an Alienware 15 R4 laptop. I can install and run all the software and stuff. I can probe and test each light. When I  open the GUI I don't have a problem setting colors, but when I run the Ambient, they dont change, I select each section as described, but my keyboard remains the color I left it as tested in the GUI.

T-Troll, 2021-01-18:
Thank you to point me this issue.

Most probably, screen capture not initialized correctly. I'll add diagnostics at this point and drop a test version for you.

Meanwhile, can i see what do you have in registry (HKCU/Software/Alienfx_SDK and HKCU/Software/Alienfxambient) to make sure this is not setup issue?

bagorocks, 2021-01-18:
In HKCU/Software/Alienfx_SDK it has the names for each of the 12 lights I gave in the probe, then it also has 12 files called "flag 1360" all of which have data=0x000000000
in HKCU/Software/Alienfxambient it only has a folder called "mappings"

bagorocks, 2021-01-18:
Also, after playing with AWCC and deleting and reinstalling alienfx, the GUI no longer changes the color of any of the lights for more than a second.

T-Troll, 2021-01-18:
> in HKCU/Software/Alienfxambient it only has a folder called "mappings"
Aha! So it's a configuration issue. Ok, let me check...

> Also, after playing with AWCC and deleting and reinstalling alienfx, the GUI no longer changes the color of any of the lights for more than a second.
It's known issue, AWCC trying to revert color back. You can both stop AWCCservice (as i recommend, gui app try to do it automatically at start if "Disable AWCC" on in "Settings", or enable "Autorefresh" option in Settings - it will refresh lights faster, then AWCC trying to change it ^_^

Also, please check you have DirectX installed - Ambient use it to grab the screen. I'll add this to requirements.

bagorocks, 2021-01-18:
I think I just broke it more when I installed DirectX I cant open the probe now

bagorocks, 2021-01-18:
I think the DirectX was my problem, It works now! Thank you so much

closed 2021-01-18 by bagorocks

T-Troll, 2021-01-18:
You are welcome!
I add this requirement to read.me