T-Troll/alienfx-tools #8: Per-key lights keyboard support

T-Troll, 2021-01-24:
Discussing issue related to per-key devices (m17r3 as an example).

T-Troll, 2021-01-24:
O-o....

So it's virtual device or not recognized by API!
Can you, please, check device manager (Human Interface Devices part), and find any device here with VID_187C at Details->Hardware IDs?

GensaGames, 2021-01-24:
Yes I can find one of them. 
[image]

T-Troll, 2021-01-25:
Great.

You find the same chip as all m15/m17 series and Dell G5.
But not other devices you have.

So, seems like Dell changes VID (Vendor ID) for light device OR change interface inside light device (or both - you mouse can have different VID, but keyboard have other interface). We need to scan correct devices and interfaces in this case.

Anyway, i need you help to check exactly (and add to my API), and this time it will be not so easy:
1. Install [USBPCap](https://desowin.org/usbpcap/) tool (not in program files, but in some folder opened for writing). Reboot after it install it's filter driver.
2. Open 2 console windows, change folder to USBPCap for one, and for my tools for other. 
3. Run USBPcamCMD.exe from console. You will see a long list of USB devices, grouped under some numbers.
4. Copy list for me.
5. Select the group number which have "USB vendor-defined device" (it can be more then one, mouse can be different group then other lights).
6. Type a file name (f.e. `per-keyafx.pcap`)
7. Keep it running, switch to other (my tools) console.
8. Issue the command: `alienfx-cli high-level set-one=0,0,255,0,255` (you mouse should be violet)
9. Issue the command: `alienfx-cli high-level set-one=1,24,0,255,0 set-one=1,28,0,0,255` (you "4" button should be green, "8" button should be blue).
10. Switch back to USBPCap console, press Ctrl+C.
11. If you doing all things right, you will see a log file (with the name from p.6) in this folder. Attach log from p.4 and this file to this issue.
NB: AWCC service should running for pp. 8 & 9, so better keep it running all time. But you can keep AWCC application closed.

T.I.A.

GensaGames, 2021-01-25:
- Running `USBPcapCMD`

```
Following filter control devices are available:
1 \\.\USBPcap1
  \??\USB#ROOT_HUB30#5&17abd317&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
2 \\.\USBPcap2
  \??\USB#ROOT_HUB30#4&1df88fde&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 4] USB Input Device
      HID-compliant vendor-defined device
    [Port 7] USB Composite Device
      Integrated Webcam
    [Port 8] USB Composite Device
      USB Input Device
        HID-compliant mouse
      USB Input Device
        HID Keyboard Device
        HID-compliant consumer control device
      USB Input Device
        Alienware 610M Wired/Wireless Gaming Mouse Device
          Alienware Command Center Win32 Component Device
          Alienware Control Center SmartInstaller Component Device
    [Port 10] USB Input Device
      HID-compliant vendor-defined device
      HID-compliant vendor-defined device
      HID Keyboard Device
      HID-compliant consumer control device
    [Port 14] Intel(R) Wireless Bluetooth(R)
      Bluetooth Device (RFCOMM Protocol TDI)
      Microsoft Bluetooth Enumerator
        MDR-XB950N1 Stereo
        MDR-XB950N1 Avrcp Transport
        MDR-XB950N1 Avrcp Transport
        MDR-XB950N1 Hands-Free AG
          MDR-XB950N1 Hands-Free AG Audio
        CSR GAIA™
        Phone Call Audio Device
        Phone Input Device v2
        MDR-XB950N1
      Bluetooth Device (Personal Area Network)
      Microsoft Bluetooth LE Enumerator
    [Port 18] Realtek USB 3.0 Card Reader
3 \\.\USBPcap3
  \??\USB#ROOT_HUB30#7&314bd266&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
Select filter to monitor (q to quit):
```

GensaGames, 2021-01-25:
Now logs in the attachments below. Everything worked, I saw color changes for keyboard and mouse. Note. As you can see above, I had all the "vendor-defined device" under the same group - number 2. So I performed this operation just once. 

[Google Drive. Logs.](https://drive.google.com/file/d/1Od7hdFghxkZGR2CTTJ-mwcX7YLduQzwP/view?usp=sharing)

T-Troll, 2021-01-25:
> USB Input Device
>         Alienware 610M Wired/Wireless Gaming Mouse Device
>           Alienware Command Center Win32 Component Device
>           Alienware Control Center SmartInstaller Component Device

Quite interesting!

Ok, let me check, i'll provide feedback after.

T-Troll, 2021-01-25:
I take a look... Uff...

1. I didn't see the mouse light control commands. Can you, please, make another log, but for mouse only? (use `set-one=0,0,R,G,B` command with different colors (tell me what color you set)?
2. Seems like they change protocol completely. But i see it, and can try to support.
3. I need VID and PID for all HID "Vendor-defined devices" to understand which one is keyboard (it's bus number 2:6:0, if it helps, other lights at 2:2:0 - PID 0x550)

GensaGames, 2021-01-25:
1. For the Mouse. I did the testing again: 
``` 
C:\Program Files\Alienfx-Tools>alienfx-cli high-level set-one=0,0,255,255,255
alienfx-cli v1.0.0
Done.

C:\Program Files\Alienfx-Tools>
C:\Program Files\Alienfx-Tools>alienfx-cli high-level set-one=0,0,255,255,0
alienfx-cli v1.0.0
Done.

C:\Program Files\Alienfx-Tools>alienfx-cli high-level set-one=0,0,255,0,0
alienfx-cli v1.0.0
Done.

C:\Program Files\Alienfx-Tools>
```

[Logs available here](https://drive.google.com/file/d/1k29koQnor8NMXkxPV5RRN0RPYfznPwcN/view?usp=sharing). PS. I don't use colors for the Mouse, because of it's battery usage.

Note. I have very wired behavior from time to time. I think it's because of AlienwareFX application where I have some themes. For ex. last time I lunched it, and command `alienfx-cli` doesn't work because of error "Cannot init Dell library". And sometimes it's mapping wrong. For ex. I was trying to set colors for mouse it was changing colors for keyboard. Once I restarted laptop everything were fixed and I could make my tests above.

GensaGames, 2021-01-25:
And how I can get "VID and PID" for "vendor-defined devices"? Is it form Device properties?

GensaGames, 2021-01-25:
Below images of all "HID-compliant vendor-defined device". 
[image]
[image]
[image]
[image]

T-Troll, 2021-01-25:
Thanks.
Let me think (maybe i will ask you for more logs to better support effects), then i make probe/test app for you.
If it works, i'll add v4 API support to all apps.

T-Troll, 2021-01-26:
Uff... Still didn't see any mouse light activity, neither define keyboard PID exactly.
So, i'll ask you for a bit more actions (let's delay mouse and concentrated by keyboard):

1. Download, unpack and run USBTreeView tool from [here](https://www.uwe-sieber.de/usbtreeview_e.html).
2. Select "File"-"Save text report" from the top menu.
3. Send it to me (NB: you can not use GDrive, but attach file in this thread directly - see description at the bottom of edit box here - just click this line to select a file and attach).

Just FYI - we are looking for device with bus ID 2-6-0. The rest is understandable.

GensaGames, 2021-01-26:
Below the logs files. (I tried to send above logs in .pcap extension and it was not supported here)

[UsbTreeView_report_GENSAGAMES-AW.txt](https://github.com/T-Troll/alienfx-tools/files/5871537/UsbTreeView_report_GENSAGAMES-AW.txt)

T-Troll, 2021-01-26:
Ok, got it.

Bad news: This world is build in China more and more. Now Dell uses Darfon OEM keyboard instead of original one.
Good news: It's quite popular, so maybe i find SDK for it. Anyway, USB API looks a bit wierd, but understandable.

Give me a couple of days, then i drop test app for check it works at you gear.

T-Troll, 2021-01-28:
Ok.
Can you please unpack and run this:
[AlienFXSample.zip](https://github.com/T-Troll/alienfx-tools/files/5885176/AlienFXSample.zip)
Then show me output and results (most probably it will not work this time, it's ok)?

GensaGames, 2021-01-28:
Result 
```
PID: 1a1c
API v4
Now try to set colors....
Set color result 0
Update result 0
```

T-Troll, 2021-01-28:
Good.
Check this app please - [AlienFXSample.zip](https://github.com/T-Troll/alienfx-tools/files/5885456/AlienFXSample.zip)
What this version shows?

NB: After it stop - check the light (it should change color at button "1"), if it is - press Enter - it change more lights. Press Enter again to exit it.

GensaGames, 2021-01-28:
The same, no effect. (BTW there are two button "1". Another one is on the numpad)

```
PID: 1a1c
API v4
Now try to set colors....
Set color result 0
Update result 0

Ok, how about multi colors?
```

T-Troll, 2021-01-28:
This time is not so good.
It means device didn't accept command (result 0, should be 1).
Ok, let me think and check more.
And yes, i mean "1" at the top row. Other one "Numpad 1".

T-Troll, 2021-07-01:
New test application, let's check, but i found somethings.
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/6746704/AlienFX.Sample.App.exe.zip)

T-Troll, 2021-07-02:
Previous test app was wrong, try this one:
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/6751487/AlienFX.Sample.App.exe.zip)

GensaGames, 2021-07-02:
Hey. Cool. I wil check out a bit later.

GensaGames, 2021-07-02:
I forget what I need to do. The same as above?

T-Troll, 2021-07-02:
Yes, just run and share log for me. It will not change color now (i fix it already), but i need to be sure it work with you device.

PS: You are not alone now, some guy from Reddit help us too, he have other keyboard device.

T-Troll, 2021-07-02:
We are continue to test.
Another testing app:
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/6752943/AlienFX.Sample.App.exe.zip)

T-Troll, 2021-07-02:
Another series of tests: 
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/6753235/AlienFX.Sample.App.exe.zip)

T-Troll, 2021-07-03:
I found another possible cause of this issue, please check:
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/6757741/AlienFX.Sample.App.exe.zip)
Cross you fingers...

GensaGames, 2021-07-08:
Oh man, sorry let me do it right now.

GensaGames, 2021-07-08:
Output from run

```
PID: 1a1c
API v4
Trying to reset...
IO result is Unknown (1)
Reset result 0
Now try to set colors....
Set color result 0
Update result 0

```

T-Troll, 2021-07-08:
We are solve this already, now testing quite close to final stage.
I'll come back here then we done with lights to check at you hardware (other guy have m15r4, different PID).
Thanks anyway.

T-Troll, 2021-07-08:
Please check this:
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/6783067/AlienFX.Sample.App.exe.zip)
It should change 4 buttons to red, one to yellow.
This works at R4, did it work for you?

GensaGames, 2021-07-09:
Doesn't work for me. That's wired if it works on R4

```
Scan: VID: 0xd62, PID: 0x1a1c, Version: 13063, Length: 12
Scan: VID: 0x187c, PID: 0x550, Version: 512, Length: 12
2 device(s) detected.
Output Report Length 0, Input Report Length 0, Feature Report Length 64
Usage ID cc, Usage Page ff89
Output caps 0, Index 0
Attributes - length: 12, version: 13063
Path: \\?\hid#vid_0d62&pid_1a1c&col02#6&23696941&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Init: VID: 0xd62, PID: 0x1a1c, Version: 13063, Length: 64
VID: 0xd62, PID: 0x1a1c, API v4
Status result 8c
Trying to reset... Reset status... Status result 8c
Now try to set colors... Status result cc
Status result cc
Status result cc
Status result cc
Set color result 1
Update result 1
Status result cc
Final status... Status result 8c
```

T-Troll, 2021-07-09:
Thanks.

Well, according to the log - it works well.
But the issue it have limited on-off control support now.
So if you lights disabled into AWCC - it will not enable it back for now.
Can you please check again, but first set all keyboard light to some color? (not black, f.e. green one)

GensaGames, 2021-07-09:
You are right! It works with static, before I had a "Breathing Rainbow". Thanks for update.

T-Troll, 2021-07-09:
Splendid!

I fix initial state (so doesn't need to do it in AWCC) then release.

T-Troll, 2021-07-11:
Check [Latest release](https://github.com/T-Troll/alienfx-tools/releases/tag/3.0.0)

GensaGames, 2021-07-12:
Will try soon. Thank you for the update! Would be nice to add donation page to the readme.

T-Troll, 2021-07-13:
Thank you, but i have enough from my main projects.
You just can make me good PR/promo if you want to make me a favor. Maybe i add a button for it. ^_^

T-Troll, 2021-07-18:
Status update - we want to try to support hardware effects for v5, then i assume this issue finished and close it.

T-Troll, 2021-07-24:
Good news: I found effect commands.
Bad news: it's keyboard-wide, not key-wide. So i can use it in special cases only.

So - that's all i can do for RGB, some small features will be added, but main part of functions works.
I close this issue.

closed 2021-07-24 by T-Troll