T-Troll/alienfx-tools #9: Alienware m15 R1 alienfx-cli not doing anything

trlaing, 2021-02-08:
I am unable to set my Alienware lights using "alienfx-cli.exe". alienfx-gui.exe does work, but it does not save the settings and I need to change them each time I boot up.
alienfx-gui:
[image]
I can add all the lights and set them, but after a reboot all lights are not listed and I see the window above - even after saving the profile.

However, I would rather use alienfx-cli as I like the command line.
```
D:\Downloads\AlienFX\alienfx-tools>alienfx-cli.exe status
alienfx-cli v1.0.0
Device ID#1360
Done.
```
I'm not sure if this behaviour is correct or not. When I enter the following, nothing happens  to the lights at all:
 
```
D:\Downloads\AlienFX\alienfx-tools>alienfx-cli.exe set-all=0,0,255
alienfx-cli v1.0.0
Done.
```

T-Troll, 2021-02-08:
First issue looks like a bug.
Let me check, i have the same gear on hands.
Meanwhile, you can use `alienfx-probe` to define the names.

Second - some CLI commands don't works with low-level API if names not defined. I mention it in read.me, check please.
But you can use `alienfx-cli high-level set-all=.....` (the only issue it will work until console is active, then AWCC switch it back).

trlaing, 2021-02-08:
Yes, alienfx-cli.exe works after running alienfx-probe. Sorry for not reading the readme thoroughly. Thank you!

```
D:\Downloads\AlienFX\alienfx-tools>alienfx-probe
alienfx-probe v1.0.0
For each light please enter LightFX SDK light ID or light name if ID is not available
Tested light become green, and turned off after testing.
Just press Enter if no visible light at this ID to skip it.
Probing low-level access... Found 1 device(s)
Probing Dell SDK... Can't init Dell library!
No LightFX SDK detected, you should provide names yourself!
Probing device PID 0x...550 Connected.
Enter device name or id:
Final name is
Testing light #0(enter name or ID, ENTER for skip): Cover
Final name is Cover, Testing light #1(enter name or ID, ENTER for skip): Power
Final name is Power, Testing light #2(enter name or ID, ENTER for skip): KB 1
Final name is KB 1, Testing light #3(enter name or ID, ENTER for skip): KB 2
Final name is KB 2, Testing light #4(enter name or ID, ENTER for skip): KB 3
Final name is KB 3, Testing light #5(enter name or ID, ENTER for skip): KB 4
Final name is KB 4, Testing light #6(enter name or ID, ENTER for skip):
Skipped. Testing light #7(enter name or ID, ENTER for skip):
Skipped. Testing light #8(enter name or ID, ENTER for skip):
Skipped. Testing light #9(enter name or ID, ENTER for skip):
Skipped. Testing light #a(enter name or ID, ENTER for skip):
Skipped. Testing light #b(enter name or ID, ENTER for skip):
Skipped. Testing light #c(enter name or ID, ENTER for skip):
Skipped. Testing light #d(enter name or ID, ENTER for skip):
Skipped. Testing light #e(enter name or ID, ENTER for skip):
Skipped. Testing light #f(enter name or ID, ENTER for skip):
Skipped.
D:\Downloads\AlienFX\alienfx-tools>alienfx-cli.exe set-all=0,0,255
alienfx-cli v1.0.0
Done.

D:\Downloads\AlienFX\alienfx-tools>alienfx-cli.exe set-all=128,128,255
alienfx-cli v1.0.0
Done.
```

UPDATE: For some reason, the 3rd keyboard light is flashing between the old colour I set days ago (violet) and the new colour I have set using the above commands. If I open alienfx-gui the colours are all violet and if I change them all to cyan the 3rd keyboard flashes cyan/violet and the rest stay cyan. Then, when I click "save" the alienfx-gui window just disappears and is not minimized (is no longer running and hasn't saved the settings).

T-Troll, 2021-02-08:
No, `alienfx-cli` works without names, just some commands require lights defined (set-all and some other). You can use set-one without names.
I can't reproduce `alienfx-gui` bug, press "save" button or just close app works and save the lights. Let me guess. more here.

About 3rd light it looks like hardware bug. Do you have AWCC running (i see you don't according to probe output)?
Try to stop AWCCService in service control and then call `alienfx-cli set-one=0,4,128,128,255` and let me know.

T-Troll, 2021-02-08:
Update - it also possible if you set it as a "hardware power button" in gui. In this case you need to set REAL hardware power button in the app. BTW, i use it for HDD led also ^_^

trlaing, 2021-02-08:
It isn't set up as a power button - none are right now. I disabled AWCC yesterday so it is not running. However, after running the command you recommended (alienfx-cli set-one=0,4,128,128,255) nothing happened, but I opened alienfx-gui, didn't change anything and saved (without crash) and closed, then ran "alienfx-cli.exe set-all=128,128,255" and the keyboard lights are now all cyan, but the case alien head is green (probably cyan as it never looks the right colour!) and the power button is violet. However, I am happy with this :-)
By the way, the reason I am doing this is because I get the "no supported AlienFX devices" error and just gave up with Alienware Command Center after wasting hours trying to sort it.
Can you manipulate the 4 shortcut keys at the top right of the keyboard?

T-Troll, 2021-02-08:
Hmm... You have something wrong with you system. Ok, let me provide you debug gui version tomorrow (it's an evening here) and check.
No, i don't manipulate macro keys (and it's weird even in AWCC), as well as fans and power profiles (do you know, what setting fan (NOT power profile) to "performance" in AWCC gives you TDP boost?), but looking about fans (old DELL libs don't work in 15R1).

T-Troll, 2021-02-08:
Ah, and about GUI crash... can you dump me registry?
Dragons can live there...
I need the HKCU/SOFTWARE/Alienfx_SDK and HKCU/SOFTWARE/Alienfxgui
First keeps all device names and mappings, second keeps all GUI settings.

trlaing, 2021-02-08:
Alienfx_SDK
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\SOFTWARE\Alienfx_SDK]
"Dev#1360"="Ø"
"1360-0"="Cover"
"Flags1360-0"=dword:00000000
"1360-1"="Power"
"Flags1360-1"=dword:00000001
"1360-2"="KB 1"
"Flags1360-2"=dword:00000000
"1360-3"="KB 2"
"Flags1360-3"=dword:00000000
"1360-4"="KB 3"
"Flags1360-4"=dword:00000000
"1360-5"="KB 4"
"Flags1360-5"=dword:00000000

Alienfxgui:
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\SOFTWARE\Alienfxgui]
"AutoStart"=dword:00000001
"Minimized"=dword:00000001
"Refresh"=dword:00000000
"LightsOn"=dword:00000001
"Monitoring"=dword:00000000
"OffWithScreen"=dword:00000001
"Dimmed"=dword:00000000
"DimmedOnBattery"=dword:00000000
"OffPowerButton"=dword:00000000
"DimmingPower"=dword:0000005c
"ActiveProfile"=dword:00000001
"GammaCorrection"=dword:00000001
"ProfileAutoSwitch"=dword:00000000
"DisableAWCC"=dword:00000000
"LastActive"=dword:00000550
"CustomColors"=hex:64,b1,ff,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00

[HKEY_CURRENT_USER\SOFTWARE\Alienfxgui\Events]
"Set-1360-0-1"=hex:01,00,00,00,00,02,00,00,00,80,00,ff,00,00,00,80,00,ff,00,00,\
  00,00,00,02,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,02,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,5a,00,00,00,02,00,00,00,00,00,00,00,00,00,00,\
  00,00
"Set-1360-1-1"=hex:01,00,00,00,00,01,00,ff,ff,80,00,ff,00,00,00,00,00,02,00,00,\
  00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,02,00,00,00,00,00,00,00,00,00,\
  00,00,00,00,5a,00,00,00,02,00,00,00,00,00,00,00,00,00,00,00,00
"Set-1360-2-1"=hex:01,00,00,00,00,02,00,00,00,80,00,ff,00,00,00,80,00,ff,00,00,\
  00,00,00,02,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,02,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,5a,00,00,00,02,00,00,00,00,00,00,00,00,00,00,\
  00,00
"Set-1360-3-1"=hex:01,00,00,00,00,02,00,00,00,80,00,ff,00,00,00,80,00,ff,00,00,\
  00,00,00,02,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,02,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,5a,00,00,00,02,00,00,00,00,00,00,00,00,00,00,\
  00,00
"Set-1360-4-1"=hex:01,00,00,00,00,02,00,00,00,80,00,ff,00,00,00,80,00,ff,00,00,\
  00,00,00,02,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,00,02,00,00,00,\
  00,00,00,00,00,00,00,00,00,00,5a,00,00,00,02,00,00,00,00,00,00,00,00,00,00,\
  00,00

[HKEY_CURRENT_USER\SOFTWARE\Alienfxgui\Profiles]
"Profile-1"="Profile 1"
"Profile-flags-1"=dword:00000001
"Profile-app-1-0"=""

The "no supported AlienFX devices" error seems to be quite common. Apart from that, the laptop seems to work fine. I use it for my work and don't play games on it.

T-Troll, 2021-02-09:
Uff... I can't reproduce, GUI works well with you config.

But i see possible source of the issue - `"Dev#1360"="Ø�"`
What locale do you use? This can be termination chars in some.
Try to use Latin device name - you can change it in registry or just remove and edit in GUI.

trlaing, 2021-02-09:
I have a UK keyboard and Windows is English (United Kingdom) if that helps. If I don't run the gui on startup and just use the command
`alienfx-cli set-all=128,128,255`
After booting, the keyboard all changes to the desired colour. The power button doesn't change and the case light looks kind of green. However, I am quite happy with this as I can't see the case light when using it and it is really just the keys I wanted to change. Thank you very much for your time and help with this.

T-Troll, 2021-02-09:
As far as i see, you wave weird device name (some special chars). It's quite interesting how it happened, but you can just change it in registry - GUI should start working.
I recommend you go this way, cause gui have a lot of useful options.
Anyway, the choice is yours, you can also switch Power button colors using `set-power` cli command. (just add it after set-all). Check power button ID in list above (i see it 1).

trlaing, 2021-02-09:
Excellent, thank you.
`alienfx-cli set-power=1,127,127,255,127,127,255`
does set the power light to light blue.
I changed the device name to "AlienFX" in the registry, but alienfx-gui still crashed when I saved it. I'll see how it goes next time I reboot (as I haven't since changing it - all my work programs are now opened for the day).

closed 2021-05-16 by T-Troll