T-Troll/alienfx-tools #10: Flickering keyboard after applying 'rainbow' profile

sirsquishy67, 2021-05-09:
One of the 4 zones is flickering between Green and red right now, I was trying to test out the GUI applet to build a custom 'slower' moving color cycle when the software crashed and ever since then the left zone of the keyboard is flashing between my two test colors. Is there a way to reset the RGB hub to overwrite the settings? I tried a reinstall/wipe of AWCC, dropped the profile data from your tool set, and nothing works. its constant and starts at power on (POST).

T-Troll, 2021-05-09:
Thank you for report.

1. What gear do you have?
2. Can you please issue command `alienfx-cli.exe status high-level status` (then awcc working) and send me results?
3. What chain of operations provide GUI crash?

Yes, you can change the lights back - just start AWCC and select other zone color, then apply. If system in hang state (it happened sometimes if you play with power button settings) - just shutdown/hibernate system, remove power cord and wait 10-15 seconds - it's completely wipe BIOS lights configuration (this trick described in readme, by the way).

sirsquishy67, 2021-05-09:
Oh wow thanks for getting back to me so soon. Was not expecting anything until next week, so I have been doing various testing and trials to get the symptom to stop. So far zero success. 

The laptop is a G5 SE5505 (Ryzen+AMD GPU) and the 4 zone RGB keyboard. 

I reset the CMOS (pulled AC + Main battery + CMOS Battery) - No change
I reflashed the BIOS (1.4.4) - no Change
Reinstalled AWCC+OC controls - No change
I reinstalled windows - no Change
Reapplied profiles with "alienfx-cli.exe" - no change

the Far left zone on the KB keeps on rapidly(and some what randomly) flashing green no matter what (If I set this zone to say RED it will flash Green and flip over to red during the 'off' part of the flashing before flipped back to green).

Videos of the issue
During a reinstall of windows - https://imgur.com/fE1idW5
With the KB RGB turned off (255,255,255) via CLI - https://imgur.com/J4McbZi
AWCC using a red theme on the Left zone - https://imgur.com/aGPBdMv

These are the settings I ran, when the GUI crashed, that caused this issue.
1 - https://imgur.com/MMwOm5k
2 - https://imgur.com/e6AreO5

Here is the output you requested
C:\alienfx-tools>alienfx-cli.exe status high-level status
alienfx-cli v1.0.0
Device ID#1360 - 550 (Active, V3)
  Light ID#0 - 1
  Light ID#1 - 2
  Light ID#2 - 3
  Light ID#3 - 4
API version 5.2.0.0 detected
Devices found: 1
Device #0, Name: Dell G5 5505 Standard KB, Type: Keyboard, Lights: 4
  Light #0, Name: KB Left, Position Unknown, Color Unknown
  Light #1, Name: Keyboard Center, Position Unknown, Color Unknown
  Light #2, Name: KB Right, Position Unknown, Color Unknown
  Light #3, Name: NumPad, Position Unknown, Color Unknown
Done.

I can set the color for all/select-lights via CLI and the GUI with no issue, but the flashing continues.  It seems effects were 'stacked' into the config of the RGB controller as any new setting I push is done on top of the failed settings. Is there anyway to wipe the config of the controller via the SDK to give it a 'factory' reset? The AWCC 'reset' options do it exist/work for this Model Laptop (the setting button does absolutely nothing), so I have not even been able to try it through that method.

sirsquishy67, 2021-05-09:
Small update, so by running

C:\alienfx-tools>alienfx-cli.exe set-action=1360,0,pulse,0,0,0
alienfx-cli v1.0.0
Done.

I was able to reduce the flashing quite a bit. But it's still happening. It feels like one of the really rapid rainbow profiles is still in effect (short/fast duration). See video - https://imgur.com/vjM4bG6

any ideas to get that last bit to stop?

T-Troll, 2021-05-09:
Thank you for reply.

I'm quite interested about to catch the situation GUI crashed, so can you describe what you do (step-by-step from first launch) to reproduce?

We are check G5 (but Intel one) before, it have the same chip as Alienware m15/m17.
You symptoms can rely to one of the 3 situations:
1. You set this light as a hardware power button (what's why blinks). Only light system reset can help in this case.
2. Rainbow mode require more colors (it can support up to 9, AFAIR). You can just try to change amount of colors or light mode.
3. AWCC going crazy and change you light zone. Try to stop AWCCService in computer management and then set another mode.

I'm into the trip around the globe now (yes, Covid can't stop me), but will investigate this issues at the end of this week.

Meanwhile, you can try to do one of 2 things:
1. use `alienfx-cli` to set full rainbow (7 colors) to the light #0 (see readme about commands) or set it to different mode.
2. Reset light system use power trick as i describe above:
> just shutdown/hibernate system, remove power cord and wait 10-15 seconds - it's completely wipe BIOS lights configuration (this trick described in readme, by the way).

sirsquishy67, 2021-05-09:
See below;
1.You set this light as a hardware power button (what's why blinks). Only light system reset can help in this case.
- this is possible, how do I disable the power button light? Its unchecked in the GUI, I am not aware of where else to change this. How can I do a light system reset - This sounds exactly what I want/need to do to fix this. 

2.Rainbow mode require more colors (it can support up to 9, AFAIR). You can just try to change amount of colors or light mode.
- I know, it was a test to see how SLOW I can make the GUI app push color changes, AWCC is limited to 500ms or so and I wanted it slower, LOL

3.AWCC going crazy and change you light zone. Try to stop AWCCService in computer management and then set another mode.
- AWCC service was always disabled during the tool kits setup, I enabled the service so you can see that I have API v5 installed. But all settings pushed through GUI/CLI are with AWCC's service disabled (stop + disabled).

T-Troll, 2021-05-09:
1. Unfortunately, i didn't find a way to unset it. So if you do it once - it will stay until reset/other light assigned as power button. In fact, i'm implement this feature because my m15 didn't have HDD light, so i use power button for it. You can try to assign non-existing light (f.e #4) as a power button (use `set-power` cli command or do reset - see quoted text into my previous message! (one more thing - you need shutdown system power but have battery working, not put system to sleep or clear CMOS battery!).
2. AWCC have limited settings, yes. In fact, i don't use it at all (only thing i need it is to switch TPU). But some of my tools settings can provide unexpected results (although it supported by hardware chip).
3. Ok. But you can keep it running then play with CLI - just don't mess with power button.
4. See above - to reset you need system powered off, power cord out, but battery connected - maybe you need to wait more for light system reset for you gear - try to wait 30 seconds or so. If reset successful, you will see keyboard lights switch to random colors at POST. 

Hmm... Maybe they change light reset procedure for AMD (for Intel G5 it's the same)?

sirsquishy67, 2021-05-09:
See above - to reset you need system powered off, power cord out, but battery connected - maybe you need to wait more for light system reset for you gear - try to wait 30 seconds or so. If reset successful, you will see keyboard lights switch to random colors at POST.

- I think maybe the RGB controller hub got damaged with your software here, doing the reset procedure that you outlined and I do get the full spectrum RGB change over from power off to through POST, but that far left zone flickers still.  See video - https://imgur.com/Rmg36WI , thoughts? Its under warranty so I will work with Dell to figure out what to do next, but if the SDK + your tools are able to damage the RGB controller, thats a huge issue for users of yours that are out of warranty. 

Hmm... Maybe they change light reset procedure for AMD (for Intel G5 it's the same)?
- This is possible, however if the RGB spectrum chance at POST after a power down (AC removed) is what's expected then this G5 is doing the expected reset and still has the issues makes this a hardware failure of the RGB controller (voltage issue maybe?).

sirsquishy67, 2021-05-09:
Ok, I fixed it. It's definitely a hardware issue with the G5 SE 5505 and the SDK calls your software kit was doing. Not only was the power button setting selected (I did not do this), the HDD sensor was also selected on the first zone of the Keyboard. 

[Fix]
I had to move all monitoring with the GUI applet to a phantom '5th' zone for it to finally stop. Then shut down the machine and did the reset method in your document to get the flashing to stop. I then cleared out all settings in the registry to make sure the phantom sensor data is just gone. 

[process to repeat?]
Probably repeatable..maybe...? - On a installed and setup G5 SE 5505 download your current zipfile build, run 'alienfx-probe' to setup the controller(PID 0x 550) and its 4 zones (0-3), then run alienfx-gui. At this point the sensors were not in the GUI and I had to manually enter them. Once they were there and tested to be active (Green LED test) then I went to the Color tab to setup a two color 'rainbow' profile on sensor 0. That is when the Green/black(lights out) profile was applied for that rapid flashing and the GUI crashed. When I loaded the GUI it had no settings and sensor data again. That is when the condition was met and I reached out after some testing (CMOS reset, reinstalled windows, reset/wiped the reg keys for your software kit..etc). 

[symptoms]
You know about the flashing, I dont have crash logs but the CLI did not know what sensor was the 'power-LED', so I set that to a phantom sensor 4, however sensor 4 never showed up in the GUI until I manually added it. Many hours later is when I decided to take every.single.monitoring event on the 'monitoring' tab and applied them all to the 5th (ID#4) phantom sensor and that is when the issue got resolved. Even after applying the GUI elements the CLI did not show the correct status for the power-LED so there is some broken logic between the GUI applet and what the CLI thinks is active. 

I advise considering to building fail safes into your GUI tool and also have ALL of the settings in the GUI tool built into the CLI tool so we can force them down to low-level. Maybe have a warning system in place for the monitoring tab or make it a selectable install on first run with file>configuration options to enable/disable it for machines like this that may run into this issue? 

For example, the power button setting would only work with the GUI and not with the CLI after this condition was met. This is even true after clearing/reinstalling the OS. So your tool sent commands to the RGB controller that persists through CMOS BIOS resets and the 'soft' RGB controller reset (as shown in my last video clip). This is a huge problem and newer G5 users that want to use your project need a well documented way back when(not if) these conditions are met. 

**Lastly, I get this is a project you do in your spare time, I do! But this could be one of the best AlienFX projects out there that we can actually tie into other RGB suites (Razer Chroma for example) using your CLI tool kit!** 

(I do suggest fleshing out your CLI guide a bit, there should be working examples that laymen can use as a reference point. If I share this with my G5 community I will have to buildout a CLI guide for them lol)

T-Troll, 2021-05-14:
Well... I finally landed for another quarantine, so let's continue.

> Ok, I fixed it. It's definitely a hardware issue with the G5 SE 5505 and the SDK calls your software kit was doing. Not only was the power button setting selected (I did not do this), the HDD sensor was also selected on the first zone of the Keyboard.

Uh-oh... For me it looks like you enable Hardware Power for this zone, so it set up this way. But - maybe - It's also effect of advanced effects like Rainbow, i'll check this at my gear.

> run 'alienfx-probe' to setup the controller(PID 0x 550) and its 4 zones (0-3), then run alienfx-gui. At this point the sensors were not in the GUI

It should not be. Probe set the same values in registry as GUI setup page, so they are equivalent. Do you finish Probe run or just press Ctrl-C?

> I went to the Color tab to setup a two color 'rainbow' profile on sensor 0. That is when the Green/black(lights out) profile was applied for that rapid flashing and the GUI crashed.

Aha.... Seems like Rainbow with one-two lights provide this. Ok, let me try to reproduce this.

> When I loaded the GUI it had no settings and sensor data again. 

GUI save data on exit or if "Save" button pressed. So it didn't save you settings if app crash. Ok, got it.

> Many hours later is when I decided to take every.single.monitoring event on the 'monitoring' tab and applied them all to the 5th (ID#4) phantom sensor and that is when the issue got resolved.

Monitoring is a software feature, it only works then GUI app active. It can't provide flashing of light without it... So it's not this.

> Even after applying the GUI elements the CLI did not show the correct status for the power-LED so there is some broken logic between the GUI applet and what the CLI thinks is active.

AFAIR, you hdd and power led didn't controlled by Alienware chip, it's Dell logic. Only keyboard are (instead of "true" Alienware laptops).

----

Ok, as a resume:
1. For me, seems like you check "Power button" at Device tab accidently for zone 0, then uncheck it... But it already set up, so it start to blink then you try to change light other way. It's common behavior for Power button in hardware.
2. I'll add "reset Power" button to set virtual zone as Hardware power light (or maybe do the same if you uncheck Power button in GUI) - it will avoid this situation, thank you for reveal the solution.
3. Please tell me about Probe question above - seems like i need to change data store logic there.
4. The same for GUI - i'll add additional data store points in some situations (f.e. after adding a light).
5. GUI crash.... I will try to reproduce it, will let you know what i found.
6. I will appreciate if you help me to provide more detailed help/readme for tools (not CLI only), it's a bit hard for me to describe all situations (some are trivial for me as an author).

Ok, expect new release in some days, as far as i check p.5 and implement pp. 2,3,4.

T-Troll, 2021-05-15:
Here it is - partially.

1. Yes, it was it. I reproduce you behavior, if light zone set to power (even if disabled later) - it provide this effect.
2. Now checking out "power button" settings will set power button to virtual light zone. BUT - you still need to reset light system after it, i've add a message about it.
3. Still waiting for you answer. Meanwhile, i change Probe logic to more frequent data store.
4. Done too.
5. Can't reproduce the crash. Can you try again from scratch and try to reproduce? It's important to all of us.
6. ...

See 1.1.6 beta in attach with changes described above, you can check it. I still wait for pp. 3 and 5 for some days, and release if we fail to obtain more data.
[alienfx-tools.zip](https://github.com/T-Troll/alienfx-tools/files/6483254/alienfx-tools.zip)

sirsquishy67, 2021-05-30:
Sorry for the super delayed response. I think the crash was related to smartshift and/or the RX5600M sleeping and just taking the program down. I was able to reproduce it a couple times but not on my Alienware test system. The changes you made helped reduce the ability to create the issue, so thanks for that!

I have a couple other G5 SE 5505 users now using your project and with the known issue documented they were able to use the tools perfectly after your last code push!

so we are good to go!

T-Troll, 2021-05-30:
Thanks, try 1.1.9 with some more bugfixes.
I close this issue.

closed 2021-05-30 by T-Troll