T-Troll/alienfx-tools #11: Max. temperature - doesn't trigger?

andyktaylor, 2021-05-24:
Using a Dell G5 5000, light works fine with GPU load, CPU load etc, but when I select Max. Temperature, I can move the slider right over to the left and it still won't trigger the 100% flag so the colour doesn't change on the Desktop LED.

andyktaylor, 2021-05-24:
EDIT - forgot to mention, I don't have AWCC installed, found it overides my Throttlestop settings, so uninstalled it (just in case this perhaps impacts the setting).

T-Troll, 2021-05-24:
Ooops.... Thank you.
Interesting. Left, not right?
The only reason i see you have no temp sensors into performance monitor, so i can't read them.
Please check what do you have in Performance Monitor under "Thermal zone information/Temperature" - i use it for read data.
If you don't have it - please study monitoring tree to locate other temperature data counter(s), i'll add it into monitoring.

PS: You don't need to uninstall AWCC completely in this case. You can just stop AWCCService and XTUOCDriverService (i use ThrottleStop as well), then made them "Disabled". In this case you can run AWCC afer start it back for alter some other settings like fan and sound (not sure you G5 support it, maybe more reasonable to uninstall it in you case).

andyktaylor, 2021-05-24:
Hey there, sorry not sure what you mean. When you sat Performance monitor, i that in AWCC?
I just installed it again, on the FUSION tab. I see the thermometer icon and it's tracking CPU fan, Temp 'C etc. Is that the place?

T-Troll, 2021-05-24:
It's ok.

No, it's Windows tool. Open start menu or search and type "Preformance monitor" - you will see a windows application with this name. Start it, select "Performance monitor" into left tree, then press "+" (add) button at top button bar. You will see the full list of Windows sensord into the new window it's open.

Oh, i see you found then i type.
Ok, add it to graph (use "add" button), close window and select into sensor list under graph. What value does it have?

andyktaylor, 2021-05-24:
Scale 1. 
Last: 301
Average 301
Minimum 301

T-Troll, 2021-05-24:
Yes. Always +28C (they have Kelvin degree). This is why you don't see it.
So we need to locate other sensor for temp there, if you found something - let me know, i'll add monitoring for you.

andyktaylor, 2021-05-24:
Yeah no luck, been through the whole listing. Painful!
Any chance of pointing the reading to a different app, like Core Temp?

T-Troll, 2021-05-24:
Sorry, man, it's no so easy to make integration with 3rd party API. And not so fast.
Look for some tool or driver which can show/add it into perfmon, find an opensource i can use for read it, or wait until i hack entire system (i want to add fan/power control into app).

andyktaylor, 2021-05-24:
No worries, thanks for helping. The tool is still pretty awesome 🥇

T-Troll, 2021-05-24:
Thanks. Feel free to report here if found some bugs/ideas.

andyktaylor, 2021-06-02:
Hey I found a temperature reading in Performance Monitor! 
What do you need in order to add it to the app?

T-Troll, 2021-06-03:
I need full PerfMon path (like current one - \\Thermal Zone Information(*)\\Temperature)
If you share it i'll add it into app and share you beta for testing.

andyktaylor, 2021-06-03:
Hmm not sure how to give you the path.
I see it under

EsifDeviceInfomation
Then the sub-section is Temperature

Is this what you mean?

[image]

T-Troll, 2021-06-03:
Yes. Add it into graph, select, right click and choose "Properties", then "Data" tab.
Copy here what you see inside.

andyktaylor, 2021-06-03:
OK got it,

it's \EsifDeviceInformation(PCI\VEN_8086&DEV_1903&SUBSYS_09BC1028&REV_05\3&11583659&0&20_1)\Temperature

Showing in Celsius

T-Troll, 2021-06-03:
Ok. Wait a bit for beta version.

T-Troll, 2021-06-04:
I have good news and bad news.
Good news - EsifDeviceInformation method works for all hardware, and even provide more temp values (for my gear i can also read SSD and PCB temperature).
Bad news - this counters can only be read if app started as administrator.

So, here is the beta for you - [alienfx-gui.beta.zip](https://github.com/T-Troll/alienfx-tools/files/6595246/alienfx-gui.beta.zip), you can check it. RUN FROM ADMINISTAROR, or temp values will not works!

I'm thinking to do it in release version right way now.

andyktaylor, 2021-06-04:
working well so far, thanks!

andyktaylor, 2021-06-04:
FYI, I created a simple task in Scheduler, so that AlienFX-gui.exe starts on Log-in with elevated priviledges to get around the fact that it won't start on boot otherwise.

T-Troll, 2021-06-05:
> I created a simple task in Scheduler

I'll try to add support for this, instead of adding to start like now. I just want to keep app simple before, don't asking for admin.

Ok, i'll close this task, release 1.2.0 fix all issues.

closed 2021-06-05 by T-Troll