T-Troll/alienfx-tools #12: Possible bug in Profiles - Active program doesn't change the colours if an app already open

andyktaylor, 2021-05-24:
So I created a couple of profiles to test the switching.

Chrome - Green
Citrix Workstation - Blue
CoinOps Next2 - Red

They all work fine on their own, and closing the app moves the colour back to default.

However, if I open Chrome, the light turns green (as it should) then I open Citrix, the light stays green.
If I then close Chrome, the light will change to blue as expected.

Is there a way to make the active window trigger the colour change? So if I tabbed between three open apps, it would dynamically change (this is how AWCC was working for me).

T-Troll, 2021-05-24:
Hmm... Good idea. Now i use first app found in memory as a reference, i mention it in readme, but also keep active app in mind is reasonable.
Ok, i'll try to add this into the next release. But it will be not so soon (some days), now i add support for multiply devices.
I'll drop a message here with the results.

T-Troll, 2021-05-25:
Check new release (1.1.7), now active window application profile always selected (if any).
Also, i add a checkbox "Only then active" at "Profiles" tab - if checked, profile only activates if application window active and de-activated if background.

andyktaylor, 2021-05-25:
Wow, super fast! Works great on initial testing, thanks so much :)

T-Troll, 2021-05-25:
You are welcome.
Please test more and close this issue if ok for you.

andyktaylor, 2021-05-25:
One minor issue.

I set up a profile for Chrome - it seems to stay active when minimised, I have to close it for the default to restore. 

All other apps, when minimised, the default profile kicks in

andyktaylor, 2021-05-25:
File explorer is a bit sticky too, if I open that (with nothing else open) then close it, the profile doesn't reset. If I go into AlienFX, I can hit refresh and it updates.

T-Troll, 2021-05-25:
> I set up a profile for Chrome - it seems to stay active when minimised, I have to close it for the default to restore.
It's normal behavior.  Enable checkbox on "profiles" tab if you want it acting like this.

> All other apps, when minimised, the default profile kicks in
This should not be. Maybe they compete to last selected profile... I check.

> File explorer is a bit sticky too, if I open that (with nothing else open) then close it, the profile doesn't reset. If I go into AlienFX, I can hit refresh and it updates.
"explorer.exe" is also a name for desktop. I don't recommend to assign any profile to it.

andyktaylor, 2021-05-25:
It seems I fixed the Chrome issue by adding Explorer.exe as a profile to override the bug with Chrome :)

T-Troll, 2021-05-30:
I assume it fixed now, re-open if you found some scratch in stability or functional.
Close this issue.

closed 2021-05-30 by T-Troll