T-Troll/alienfx-tools #13: alienfx-ambient seems to crash the rgb controller for my dell g5 5500

lowrck, 2021-05-26:
so i managed to get the lighting zones for the lightbar mapped to names and even set up the ambient screen map how i wanted. it works for about 15-30 seconds but then the bar goes black and stays black no matter what i do. even starting AWCC doesn't allow changes to work. i don't know if this is a known issue or if i am unique but for now ambient is unusable. my guess is it issues commands too quickly and overwhelms the controller causing a unrecoverable crash to happen. well, it can recovery, though only after a system reboot. its a bummer but it looks like the ambient mode just can't be used right now.

T-Troll, 2021-05-27:
1. Can i check you registry entries into HKCU/Software/AlienFX_sdk and HKCU/Software/Alienfxambient?
2. I check at my gear, and seems like it's not hardware hung, but MS alter DXGi capture into recent update. You can try to press "Reset" button into app to resume, or exit app and issue `alienfx-cli set-all=255,0,0` command or so to check.
3. I will try to fix DXGi hung into new release, it's also nice if you confirm p.2

PS: There are 10fps limit into `alienfx-ambient`, so command rate is not so high (haptics use higher rate).

T-Troll, 2021-05-27:
Check new release 1.1.8 - should be more stable now.

closed 2021-06-09 by T-Troll