T-Troll/alienfx-tools #14: Version 2.X.X not working on G5 5590

PhSMu, 2021-06-11:
Hi, its me again!

So, I was on 1.2.0 and almost everything was working well, except for a small-ish bug when I tried to set all colors. Since I only have 4 keyboard zones, it wasn't a big problem. Then v2+ came and...

Well, I can't even **open** the program. Well, I can, but nothing happens.

Anyways, no pressure, just letting you know. And as it ocurred to me as I was writing this, is there a possibility that this happened because I haven't touched the latest versions of AWCC and my keyboard firmware needs update?

T-Troll, 2021-06-11:
Welcome back, glad to see you agian, in fact.

You see, i mostly check at my home m15/13R2, so it can be dragons (i mention it then release 2.0).
So, for everyday - rollback at 1.2.0.

For 2.0 issues - wait a moment, i provide a debug release to check what i broke. Firmware update looks reasonable (last time they broke status), but not sure. Let me think a bit.

PS: Oh, i remember i have this situation before one time. Try to stop gui, run ambient or haptics, then close and restart. If it helps... Shame on me.

PhSMu, 2021-06-11:
Restart didn't work, in fact I have actually formatted the PC and reinstalled windows and remembered to check for updates because of that (I usually wait for email notifications.), but I'll try AWCC and see how it goes.

PhSMu, 2021-06-11:
Yes, it was just that. AWCC installation ran an ELC-something (too fast to fully read) and now Alien-FX tools work.

T-Troll, 2021-06-12:
No. Shame on me, but you reveal a bunch of bugs, so i withdraw 2.0.2 (and, being honest, all 2.0.0 line).
f.e. GUI can't start (have exception) if no device names defined (first start), etc, etc.

Now i test 2.1.0, i fix a lot, plan to publish it later today.

PhSMu, 2021-06-12:
will do!

closed 2021-06-18 by T-Troll