T-Troll/alienfx-tools #16: Corresponding colours do not show correctly

RikPeeters, 2021-07-02:
When you use any of the application the RGB code is not loaded in correct. 
System: Alienware 17 r4

Observations:
255, 0, 0 (red) shows as          255, 0, 0 (red) -> correct
0, 255, 0 (green) shows as      50, 0, 0 (shallow red) -> almost not visible
0, 0, 255 (bleu) shows as        0, 255, 0 (lime green)

255, 255, 0 (yellow) shows as  255,0,0 (red)
0, 255, 255 (cyan) shows as    50, 255, 0 (lime green) -> very minor change
255, 0, 255 (pink) shows as     255, 0, 255 (yellow)

Conclusion:
The red parameter is correct.
The bleu parameter is the green parameter.
The green parameter is a 1:5 scale of the red parameter (it looks like).

T-Troll, 2021-07-03:
Well, you bring me two issues in one.

1. About colors.
Seems like you system accidently detected as APIv2, but really uses APIv1 interface or mix between then. APIv2 have 4-bit colors, not 8-bit, so that's why you have this mess (replace 50 to 64 in you debug - and it fits well).
Can you please show me output of `alienfx-cli status` command - i'll fix this for you gear?
2. v1-v2 uses mask color index, and, by the way i see some mappings in you list i didn't meet before. As a short fix we can try use v1 mappings, but for new lights (tron, lcd) you have we need to check the mask. We can do it if you help me to scan masks using USBPcap.

RikPeeters, 2021-07-03:
C:\Fxtools>alienfx-cli status
alienfx-cli v2.2.5.1
Device ID#1328 - Alienware17R4 (Active, V2)
  Light ID#2 - left
  Light ID#3 - leftm
  Light ID#4 - rightm
  Light ID#5 - right
  Light ID#6 - head
  Light ID#7 - logo
Done.

RikPeeters, 2021-07-03:
Following filter control devices are available:
1 \\.\USBPcap1
  \??\USB#ROOT_HUB30#4&2080db8e&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 1] USB Composite Device
      USB Input Device
        HID-compliant mouse
      USB Input Device
        HID Keyboard Device
      USB Input Device
        HID-compliant consumer control device
        HID-compliant vendor-defined device
    [Port 2] USB Composite Device
      Alienware AW988 Chat
        A-Volute Audio Effects Component
        Headset Earphone (Alienware AW988 Chat)
        Headset Microphone (Alienware AW988 Chat)
      Alienware AW988 Game
        A-Volute Audio Effects Component
        Headphones (Alienware AW988 Game)
      USB Input Device
        HID-compliant consumer control device
        HID-compliant vendor-defined device
        Alienware Wireless Gaming Headset AW988
          Alienware Command Center Win32 Component Device
          Alienware Wireless Gaming Headset Component Device
    [Port 4] USB Input Device
      HID-compliant vendor-defined device
    [Port 5] Qualcomm Atheros QCA61x4 Bluetooth
      Bluetooth Device (RFCOMM Protocol TDI)
      Microsoft Bluetooth Enumerator
      Bluetooth Device (Personal Area Network) #3
      Microsoft Bluetooth LE Enumerator
    [Port 7] USB Composite Device
      Integrated Webcam
    [Port 9] USB Composite Device
      EyeChip
        Tobii Device
          Tobii Eye Tracker HID
      TOBII Video
Select filter to monitor (q to quit):

I have downloaded USBPcap like you can see above, is the the information you wanted? I hope I can help in any way possible, the work you guys are doing means so much to the Alienware community I cannot even comprehend.

T-Troll, 2021-07-03:
> Device ID#1328 - Alienware17R4 (Active, V2)

I was right. It uses 12-bytes (v2) interface, but 8-bit colors. So we have a kind of API2.5 here.
Ok, i will add support for this cases and share test cli app here for you soon.

> 1 \.\USBPcap1

0. Run AWCC.
1. Select the group where "HID-compliant vendor-defined device" located (1, in you case).
2. Select some filename (don't forget to add ".pcup" after name).
3. Set tron/lcd lights to some colors (better different ones, and tell me which one and which order you do it) into AWCC.
4. In USBPCap window, press Ctrl-C.
5. Send me the .pcup file from p.2
I will check this lights masks and add it into the index table.

PS: Low-level masks/index does not follow AWCC ones in general, so it's ok. For 0-1 check more close, it should be power button and, probably, hdd/wifi/capslock indicators, if you have one.

T-Troll, 2021-07-03:
Here is BETA cli for you (it should fix color mess). It also provide me better information for detect systems like yours.
Please, try to set colors using it and send me output.
[alienfx-cli.exe.zip](https://github.com/T-Troll/alienfx-tools/files/6757706/alienfx-cli.exe.zip)

PS: No, it will NOT fix masks yet, so some lights will still be unavailable.

RikPeeters, 2021-07-03:
Left tron - red
right tron - yellow
left lcd - green
right lcd - cyan
rest - bleu

https://drive.google.com/file/d/1155ahRXAl0D6ZjP60vWVOv4yERUUhvVc/view?usp=sharing
you might need to ask permission first to download the file

I tested the beta cli, the colours are correct now!
Also, I have overlooked this before, but the lightning of the micro keys? -> in the file above they were bleu, tell me if I need to run it again when there red.

T-Troll, 2021-07-03:
Please, share me output of cli as well - especially what after "DEBUG:" - i now use hardcode for you PID, but it's not a good way!

Ok, great. If colors is good now, i put this patch into next release of whole toolset (but i have hard bug with it, so took some time).
Meanwhile - by you data - i will try to add mappings for the lost lights.

RikPeeters, 2021-07-03:
alienfx-cli v2.3.0 BETA
DEBUG: VID#6268, PID#1328, Length 12, Version 0
Device VID#6268, PID#1328 - Alienware17R4 (Active, V2)

T-Troll, 2021-07-03:
> DEBUG: VID#6268, PID#1328, Length 12, Version 0

Except PID looks the same like mine 13R2. Uff... Seems like PID selection only. Ok.

Let me check logs, then i come back with questions, if any.

T-Troll, 2021-07-04:
Excusez-moi, but we have an issue.

Both you files have no commands from AWCC to interface (BTW, you can take a look yourself using Wireshark, device address is 1.1.0). Maybe you forget to apply colors, maybe USBPcap buggy on you gear (try to restart you gear).

T-Troll, 2021-07-04:
FJYI: I publish 2.3.0 with you device support, except other lights (still waiting for the data from you).

crim2som, 2021-07-04:
Hi @T-Troll  continue with watching brief to see if my Alienware 17 R5 Dark Mode can be unlocked so that AWCC FX tab works and AlienFX 2.0 is loaded. 
Just downloaded and unpacked alienfx-tools-master.zip, but there are no .exe files. 
How is alienfx-tools-master used? 
Thanks.

RikPeeters, 2021-07-04:
Left tron - green
right tron - red
left lcd - yellow
right lcd - bleu

micro - orange - yellow - bleu -> swapped 3 times

rest - purple

[test3.zip](https://github.com/T-Troll/alienfx-tools/files/6759762/test3.zip)

RikPeeters, 2021-07-04:
@T-Troll it seems like your latest release has done something to the allocation (maybe I didnt notice before):

Light #0 - Keyboard Right         -> Logo
    Light #1 - Keyboard Middle Right  -> shows nothing
    Light #2 - Keyboard Middle Left   -> Keyboard Left
    Light #3 - Keyboard Left          -> Keyboard Middle Left
    Light #4 - Alienhead              -> Keyboard Middle Right 
    Light #5 - Logo                   -> Keyboard Right
    Light #6 - Touchpad               -> Alien head (on the back)
    Light #7 - Left Tron              -> Logo AGAIN!!!
    Light #8 - Right Tron             -> Does not appear in test
    Light #9 - LCD Panel Left         -> Does not appear in test
    Light #10 - LCD Panel Right       -> Does not appear in test

I can control the color of the logo with the 2 different addresses now

T-Troll, 2021-07-04:
Thank you, i will check masks from you data soon.

Well-well-well. Seems like you masks from v1, not v2. Let me check, we fix it. Maybe i switch you 2.5 as "12-byte v1" instead of "8-bit light v2".

PS: I see you fork my code. Take a look at mask8[] and mask12[] arrays into alienfx_sdk.h

T-Troll, 2021-07-04:
@crim2som use [Discussions](https://github.com/T-Troll/alienfx-tools/discussions) for this kind of questions, don't spam issue about other topic!
You need alienfx-tools.zip, not "master" one, it's source code.

T-Troll, 2021-07-04:
@RikPeeters 

Did you power button set at ff9900 = 0000f0? 
If yes, i'll add power button control as well.
I scan the rest codes, it different from other mask. Ok, create new one.

RikPeeters, 2021-07-05:
for the power button, I think I can allocate it, but it is always overwritten by the power state. in usbpcap everything is blue, and I switched the power button from green to red a couple of times.

[power.zip](https://github.com/T-Troll/alienfx-tools/files/6760908/power.zip)

As for the fork, I am trying to get the colours to change on temperature of the cpu/gpu rather then the load, dut didnt really get anywhere just yet.

T-Troll, 2021-07-05:
I have all data needed already, plan to release today - with all masks (even change some mine for R13), BETA power control (some dragons where, i can now change it for battery mode only), and better Pulse/Morph support (i found how to do it).

For temps, switch "Esif temperatures" on at settings (and check ESIF into you Performance monitor) - seems like some older gear didn't support common windows temp - i do it for G5 before, but work on mine 13 as well. 
But it only check maximal temperature across all sensors, not CPU/GPU separately - it's hard to recognize them into WMI data.

T-Troll, 2021-07-07:
Did it work correctly with the latest release?
Please confirm and close the issue.

RikPeeters, 2021-07-07:
the trons and lcd lights did not appear in the the test probe, so couldn't configure them.

T-Troll, 2021-07-07:
1. Remove all light mappings, now it not the same like before.
2. Use "Devices" tab in alienfx-gui - Probe have limited compatibility. Or use Probe option (see readme).
Trons should be OK, as well as other. BTW, use IDs from 0 to 12, it should cover all.

RikPeeters, 2021-07-07:
AHA, it works sheers mate 🍾

closed 2021-07-07 by RikPeeters

T-Troll, 2021-07-07:
Ok, glad to here. 
Bonus for you - see #7 - mask 0x80? At my gear, it can be 3 masks here - 0x80, 0x200, 0x280. They can control all leds, HDD led only or the rest of leds. You can check yourself - did this work for you gear - and provide me mask correction. Experiment!

RikPeeters, 2021-07-07:
#7 doesnt do anything, I do not have any leds for hdd, power, or network

T-Troll, 2021-07-08:
Ah, ok. As you can see, you gear is miss mask 0x200, so it intended to be ^_^. Maybe not used.