T-Troll/alienfx-tools #19: Latest version made my power button go black

RikPeeters, 2021-07-10:
since the latest gui version you posted I have had a issue with my power button. It just wont light up.
additionally the AlienWareFXTester fails to turn it on.
I am 90% sure this is related to your program, just cant figure out how exactly.
I have heard from the tech support that the address the button uses can be the same as the hardware indicator lights you mentioned earlier. Do you have any version where I can change the power button of the AlienWare17 r4?
Thanks
Rik

RikPeeters, 2021-07-11:
update, I did manage to fix the issue.
I fixed it by hard shutting down the PC.
restarting and normal shutdown did not resolve it, only holding power button down did.
I tried to reproduce the issue, but failed as of yet.
since there is no reason to stay this trait open Ill close it for now.

closed 2021-07-11 by RikPeeters

T-Troll, 2021-07-11:
Yes, it is. My experimental power button control can do it in some cases (and even disable battely completely!), see light system restart procedure in read.me to fix!
Now i just disable PBC for v1-v3.