T-Troll/alienfx-tools #23: Alienware 15 R3 Light Help

PCNUB911, 2021-07-17:
Hello! , I found your project and it was my last hope but sadly I can't get it to work. 
I recently bought this laptop off facebook marketplace and everything works fine except there's no lighting on anything.  
I've tried everything I found on google , from fresh install w10 , installing/reinstalling ACC , uplugging battery, updating bios . 

The laptop is a Alienware 15 R3 with a 1080 max-q 

--- Info about Alienware command center (idk if it matters but, going to label it incase you want to skip it)- ---
When I open AlienFX I get a box titled 
"ALIENWARE ALIENFX "

> The theme you are opening is not compatible with this computer. You will not be able to apply the theme.

I would try to change the color and it would give me this error

`System.NullReferenceException: Object reference not set to an instance of an object.
   at AlienLabs.AlienFX.Editor.Controls.ActionColorPickerControl.processSelectedColorChanged(Color primaryColor, Color secondaryColor)
   at AlienLabs.AlienFX.Editor.Controls.ActionColorPickerControl.selectedColorChanged(Color primaryColor, Color secondaryColor)
   at AlienLabs.AlienFX.Tools.Controls.DynamicColorWheel.onSelectedColorChanged(Color primaryColor, Color secondaryColor)
   at AlienLabs.AlienFX.Tools.Controls.DynamicColorWheel.selectColotItem(PathRadioButton element, Boolean rotateIndicator)
   at System.Windows.EventRoute.InvokeHandlersImpl(Object source, RoutedEventArgs args, Boolean reRaised)
   at System.Windows.UIElement.RaiseEventImpl(DependencyObject sender, RoutedEventArgs args)
   at System.Windows.Controls.RadioButton.OnChecked(RoutedEventArgs e)
   at System.Windows.Controls.Primitives.ToggleButton.OnIsCheckedChanged(DependencyObject d, DependencyPropertyChangedEventArgs e)
   at System.Windows.DependencyObject.OnPropertyChanged(DependencyPropertyChangedEventArgs e)
   at System.Windows.FrameworkElement.OnPropertyChanged(DependencyPropertyChangedEventArgs e)
   at System.Windows.DependencyObject.NotifyPropertyChange(DependencyPropertyChangedEventArgs args)
   at System.Windows.DependencyObject.UpdateEffectiveValue(EntryIndex entryIndex, DependencyProperty dp, PropertyMetadata metadata, EffectiveValueEntry oldEntry, EffectiveValueEntry& newEntry, Boolean coerceWithDeferredReference, Boolean coerceWithCurrentValue, OperationType operationType)
   at System.Windows.DependencyObject.SetValueCommon(DependencyProperty dp, Object value, PropertyMetadata metadata, Boolean coerceWithDeferredReference, Boolean coerceWithCurrentValue, OperationType operationType, Boolean isInternal)
   at System.Windows.DependencyObject.SetCurrentValueInternal(DependencyProperty dp, Object value)
   at System.Windows.Controls.Primitives.ToggleButton.OnClick()
   at System.Windows.Controls.Primitives.ButtonBase.OnMouseLeftButtonUp(MouseButtonEventArgs e)
   at System.Windows.RoutedEventArgs.InvokeHandler(Delegate handler, Object target)
   at System.Windows.RoutedEventHandlerInfo.InvokeHandler(Object target, RoutedEventArgs routedEventArgs)
   at System.Windows.EventRoute.InvokeHandlersImpl(Object source, RoutedEventArgs args, Boolean reRaised)
   at System.Windows.UIElement.ReRaiseEventAs(DependencyObject sender, RoutedEventArgs args, RoutedEvent newEvent)
   at System.Windows.UIElement.OnMouseUpThunk(Object sender, MouseButtonEventArgs e)
   at System.Windows.RoutedEventArgs.InvokeHandler(Delegate handler, Object target)
   at System.Windows.RoutedEventHandlerInfo.InvokeHandler(Object target, RoutedEventArgs routedEventArgs)
   at System.Windows.EventRoute.InvokeHandlersImpl(Object source, RoutedEventArgs args, Boolean reRaised)
   at System.Windows.UIElement.RaiseEventImpl(DependencyObject sender, RoutedEventArgs args)
   at System.Windows.UIElement.RaiseTrustedEvent(RoutedEventArgs args)
   at System.Windows.Input.InputManager.ProcessStagingArea()
   at System.Windows.Input.InputManager.ProcessInput(InputEventArgs input)
   at System.Windows.Input.InputProviderSite.ReportInput(InputReport inputReport)
   at System.Windows.Interop.HwndMouseInputProvider.ReportInput(IntPtr hwnd, InputMode mode, Int32 timestamp, RawMouseActions actions, Int32 x, Int32 y, Int32 wheel)
   at System.Windows.Interop.HwndMouseInputProvider.FilterMessage(IntPtr hwnd, WindowMessage msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at System.Windows.Interop.HwndSource.InputFilterMessage(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndWrapper.WndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam, Boolean& handled)
   at MS.Win32.HwndSubclass.DispatcherCallbackOperation(Object o)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)`

-----------------

Using version 3 3.0.0.2 (BETA 3)

I installed alienfx tools ran " alienfx-gui " and I get the error 
"No alienware light devices detected, exiting!" 

Running "alienfx-probe" opens a cmd box for a millisecond and closes

None of the other files show any devices . 
The only one that shows me something is "alienfx haptics"  >input > default input devices 
other than that it doesn't seem to work for me.

Do you think you could help me??

I did notice that ACC stop giving me the error after installing your software. but lights still don't work.

Thank you<3

RikPeeters, 2021-07-17:
Have you tryed the AlienWareFX Test application? 
its located at "C:\Program Files\Alienware\Command Center\AlienwareAlienFXTester.exe"
Additionnaly you should be looking into alienfx-cli, type the following in CMD:
cd "path to fx folder"   (mine was cd C:\Program Files\!small programs\Fxtools)
alienfx-cli.exe status
alienfx-cli.exe high-level
alienfx-cli.exe set-all=255,255,255[,255]

this should set all lights to white

RikPeeters, 2021-07-17:
also reseat the cable please!!!

PCNUB911, 2021-07-17:
> 
> 
> also reseat the cable please!!!

Just did,  Still have the same problem.

PCNUB911, 2021-07-17:
> 
> 
> Have you tryed the AlienWareFX Test application?
> its located at "C:\Program Files\Alienware\Command Center\AlienwareAlienFXTester.exe"
> Additionnaly you should be looking into alienfx-cli, type the following in CMD:
> cd "path to fx folder" (mine was cd C:\Program Files\!small programs\Fxtools)
> alienfx-cli.exe status
> alienfx-cli.exe high-level
> alienfx-cli.exe set-all=255,255,255[,255]
> 
> this should set all lights to white
> MAKE SURE ALL ALIENWARE FX PROGRAMS ARE TERMINATED!!!
> for me left button, right click alien head, exit is sufficient

I tried the tester I get the same error as before
"It seems there is some problem with the AlienFX board."
"current version: " is blank and 'upgrading to" 3.00.00
firmware upgrade: skipped
set leds red : failed 

`C:\Program Files\AlienFX Tools>alienfx-cli.exe status
alienfx-cli v3.0.0
No low-level device found!

C:\Program Files\AlienFX Tools>alienfx-cli.exe  high-level
alienfx-cli v3.0.0
No low-level device found!

C:\Program Files\AlienFX Tools>alienfx-cli.exe set-all=255,255,255[,255]
alienfx-cli v3.0.0
No low-level device found!`

crim2som, 2021-07-17:
[image]

T-Troll, 2021-07-18:
> alienfx-cli.exe high-level

@RikPeeters, do you mean `alienfx-cli high-level status`? high-level alone just switch default API to Dell one (if available).

@crim2som, do you make a twink @PCNUB911? ^_^
I told you many times, here and at Dell forums, if cli and gui can't find USB device - no fun. 
Both my tools and AWCC can't control light device absent into the system.
So you need to check BIOS/cable/motherboard/whatever until you will have USB device working.
There is no other way.

T-Troll, 2021-07-18:
So, this issue not tools-related, and i close it. welcome to [Discussions](https://github.com/T-Troll/alienfx-tools/discussions) if you still need some advice.

closed 2021-07-18 by T-Troll