T-Troll/alienfx-tools #27: Dell Alienware Aurora R7

raould, 2021-07-23:
hi, I can't get alienfx-tools to work yet, seeking help :-)

* Just downloaded from here today, 0.9.0.
* Windows 10 home 10.0.19043
* Aurora R7, 0VDT73, Bios 1.0.23, i3 8100, 8gb ram single stick, GTX 1060 3gb.
* I can successfully run AWCC and change all the lights (there are 3 zones: the power button, the right case panel, the left case panel).
* AlienFX, AlienFusion Windows Service both appear as status Running.
* Doesn't appear to matter if I am running powershell as Administrator.
* I did install the old DirectX stuff and rebooted and that didn't fix it.

```
PS C:\Program Files\AlienFX Tools> .\alienfx-probe.exe
alienfx-probe v3.1.1
For each light please enter LightFX SDK light ID or light name if ID is not available
Tested light become green, and turned off after testing.
Just press Enter if no visible light at this ID to skip it.
Probing low-level access... AlienFX devices not present, please check device manage!
```

T-Troll, 2021-07-24:
Wow!

Being honest, you are the first guy with Alienware desktop here.
Ok, let's check and add support for you gear, if possible.

Can you, please, do this for me:
0. Download latest version (3.1.3) from [here](https://github.com/T-Troll/alienfx-tools/releases/tag/3.1.3) - 0.9.0 WAY too old!
1. After downloading, you can use `alienfx-cli` - it have Dell high-level support as well - try `alienfx-cli high-level status` or `alienfx-cli high-level set-all=255,0,0`
2. For low-level support, i need you help:
  - Open "Device manager"->"Human interface devices".
  - For each "HID-Compliant vendor-defined device", open properties, switch to "Details", switch "Property" to "Hardware ids" and tell me VID and PID.
3. If they use re-marked chip, i add it, but if it's something OEM - we need to scan interface, but it's a next step.

raould, 2021-07-24:
Thanks for your note! Apologies, no idea how I got the old version by mistake, my fault. I did just install v3.1.3. When I try "set-all=0,255,0" the lights do *not* change from their current red to the intended green. I also tried as Administrator, still no change.

There are a few too many HID entries to manually go through, there's 8 of them and that's annoying, so I am trying to find a way to export those but of course Windows UI/UX is sorta lame and frustrating this way... :-) I will add a reply when i get the information!

```
PS C:\Program Files\AlienFX Tools> .\alienfx-cli.exe high-level status
alienfx-cli v3.1.3
No low-level devices found, trying high-level...
Dell API ready
Dell API version 2.2.0.0 detected
Devices found: 1
Device #0, Name: Alienware AuroraR7 Chassis, Type: Desktop, Lights: 3
  Light #0, Name: Alienhead, Position Unknown, Color (0,0,0), Brightness: 0
  Light #1, Name: Side Left, Position Unknown, Color (0,0,0), Brightness: 0
  Light #2, Name: Side Right, Position Unknown, Color (0,0,0), Brightness: 0
Done.
PS C:\Program Files\AlienFX Tools> .\alienfx-cli.exe high-level set-all=0,255,0
alienfx-cli v3.1.3
No low-level devices found, trying high-level...
Dell API ready
Done.
```

raould, 2021-07-24:
```
 Get-WmiObject Win32_PNPEntity| Out-File -Append -Force "hwdump.txt"
```
then massaged it in emacs (i am not a windows person, really), then extracted the vendors...
```
type .\hwdumpB.txt| findstr vendor |Out-File vendors.txt
```

```
NEXT_ENTRY; __CLASS                     : Win32_PnPEntity; __SUPERCLASS                : CIM_LogicalDevice; __DYNASTY                   : CIM_ManagedSystemElement; __RELPATH                   : Win32_PnPEntity.DeviceID="HID\\VID_046D&PID_C537&MI_01&COL04\\B&1ED82C98&0&0003"; __PROPERTY_COUNT            : 26; __DERIVATION                : {CIM_LogicalDevice, CIM_LogicalElement, CIM_ManagedSystemElement}; __SERVER                    : DESKTOP-B66FIOT; __NAMESPACE                 : root\cimv2; __PATH                      : \\DESKTOP-B66FIOT\root\cimv2:Win32_PnPEntity.DeviceID="HID\\VID_046D&PID_C537&MI_01&COL04;                               \\B&1ED82C98&0&0003"; Availability                : ; Caption                     : HID-compliant vendor-defined device; ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}; CompatibleID                : ; ConfigManagerErrorCode      : 0; ConfigManagerUserConfig     : False; CreationClassName           : Win32_PnPEntity; Description                 : HID-compliant vendor-defined device; DeviceID                    : HID\VID_046D&PID_C537&MI_01&COL04\B&1ED82C98&0&0003; ErrorCleared                : ; ErrorDescription            : ; HardwareID                  : {HID\VID_046D&PID_C537&REV_3400&MI_01&Col04, HID\VID_046D&PID_C537&MI_01&Col04, ;                               HID\VID_046D&UP:FF00_U:0001, HID_DEVICE_UP:FF00_U:0001...}; InstallDate                 : ; LastErrorCode               : ; Manufacturer                : (Standard system devices); Name                        : HID-compliant vendor-defined device; PNPClass                    : HIDClass; PNPDeviceID                 : HID\VID_046D&PID_C537&MI_01&COL04\B&1ED82C98&0&0003; PowerManagementCapabilities : ; PowerManagementSupported    : ; Present                     : True; Service                     : ; Status                      : OK; StatusInfo                  : ; SystemCreationClassName     : Win32_ComputerSystem; SystemName                  : DESKTOP-B66FIOT; PSComputerName              : DESKTOP-B66FIOT
NEXT_ENTRY; __CLASS                     : Win32_PnPEntity; __SUPERCLASS                : CIM_LogicalDevice; __DYNASTY                   : CIM_ManagedSystemElement; __RELPATH                   : Win32_PnPEntity.DeviceID="HID\\VID_046D&PID_C537&MI_01&COL05\\B&1ED82C98&0&0004"; __PROPERTY_COUNT            : 26; __DERIVATION                : {CIM_LogicalDevice, CIM_LogicalElement, CIM_ManagedSystemElement}; __SERVER                    : DESKTOP-B66FIOT; __NAMESPACE                 : root\cimv2; __PATH                      : \\DESKTOP-B66FIOT\root\cimv2:Win32_PnPEntity.DeviceID="HID\\VID_046D&PID_C537&MI_01&COL05;                               \\B&1ED82C98&0&0004"; Availability                : ; Caption                     : HID-compliant vendor-defined device; ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}; CompatibleID                : ; ConfigManagerErrorCode      : 0; ConfigManagerUserConfig     : False; CreationClassName           : Win32_PnPEntity; Description                 : HID-compliant vendor-defined device; DeviceID                    : HID\VID_046D&PID_C537&MI_01&COL05\B&1ED82C98&0&0004; ErrorCleared                : ; ErrorDescription            : ; HardwareID                  : {HID\VID_046D&PID_C537&REV_3400&MI_01&Col05, HID\VID_046D&PID_C537&MI_01&Col05, ;                               HID\VID_046D&UP:FF00_U:0002, HID_DEVICE_UP:FF00_U:0002...}; InstallDate                 : ; LastErrorCode               : ; Manufacturer                : (Standard system devices); Name                        : HID-compliant vendor-defined device; PNPClass                    : HIDClass; PNPDeviceID                 : HID\VID_046D&PID_C537&MI_01&COL05\B&1ED82C98&0&0004; PowerManagementCapabilities : ; PowerManagementSupported    : ; Present                     : True; Service                     : ; Status                      : OK; StatusInfo                  : ; SystemCreationClassName     : Win32_ComputerSystem; SystemName                  : DESKTOP-B66FIOT; PSComputerName              : DESKTOP-B66FIOT
NEXT_ENTRY; __CLASS                     : Win32_PnPEntity; __SUPERCLASS                : CIM_LogicalDevice; __DYNASTY                   : CIM_ManagedSystemElement; __RELPATH                   : Win32_PnPEntity.DeviceID="HID\\VID_046D&PID_C537&MI_01&COL06\\B&1ED82C98&0&0005"; __PROPERTY_COUNT            : 26; __DERIVATION                : {CIM_LogicalDevice, CIM_LogicalElement, CIM_ManagedSystemElement}; __SERVER                    : DESKTOP-B66FIOT; __NAMESPACE                 : root\cimv2; __PATH                      : \\DESKTOP-B66FIOT\root\cimv2:Win32_PnPEntity.DeviceID="HID\\VID_046D&PID_C537&MI_01&COL06;                               \\B&1ED82C98&0&0005"; Availability                : ; Caption                     : HID-compliant vendor-defined device; ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}; CompatibleID                : ; ConfigManagerErrorCode      : 0; ConfigManagerUserConfig     : False; CreationClassName           : Win32_PnPEntity; Description                 : HID-compliant vendor-defined device; DeviceID                    : HID\VID_046D&PID_C537&MI_01&COL06\B&1ED82C98&0&0005; ErrorClea [... cut ...] ompliant vendor-defined device; PNPClass                    : HIDClass; PNPDeviceID                 : HID\VID_258A&PID_002A&MI_01&COL03\A&375BC78&0&0002; PowerManagementCapabilities : ; PowerManagementSupported    : ; Present                     : True; Service                     : ; Status                      : OK; StatusInfo                  : ; SystemCreationClassName     : Win32_ComputerSystem; SystemName                  : DESKTOP-B66FIOT; PSComputerName              : DESKTOP-B66FIOT
```

raould, 2021-07-24:
(note that the "status" could not discover the current RGB settings since it reported (0,0,0) for all 3 zones, but they are currently set to red.)

raould, 2021-07-24:
(oh looks like originally i was using 3.1.200 up above at the top i think, i typed up the original notes wrong, not sure what was 0.9.0)

T-Troll, 2021-07-24:
> Dell API version 2.2.0.0 detected

Wow! Ancient!
Lights SHOULD change color by Dell API, but, most probably, AWCC switch it back. Try to use "loop" command after set-all, it block AWCC change (but also keep cli running).

VID_046D - Logitech. Do you have mouse or keyboard from them?
VID_258A - Seems like here it is. OEM from SinoWealth, they make LCD controllers. I didn't support it yet.

So... Easy way didn't work (but maybe AW become re-marked SW later).
Can you please install USBPcap, run it, then change some lights by AWCC, then send me a log? 
I'll add support.

raould, 2021-07-24:
I updated to 3.1.301 just now as well.

No idea if this usbpcap will be of any help. I went into AWCC and changed the zone lights around in various ways. 

https://drive.google.com/file/d/13wMdeZk2YgVc8-JCtgT9ZwCOcjY9RklZ/view?usp=sharing

raould, 2021-07-24:
You are correct that adding "loop" *does* make the lights change to the color I set.

T-Troll, 2021-07-24:
About USBPcap - you can attach files right here, doesn't need use GD. Ok, i'll check.

Yes, AWCC try to restore colors after app exit in some cases.

T-Troll, 2021-07-24:
I check you pcap.... Seems like you are choose wrong controller (need one with VID_258A devices inside). So no data.

raould, 2021-07-24:
Oh no! Ok I will try again. The USBPcap user interface doesn't seem very clear to me i guess, heh.

T-Troll, 2021-07-24:
You can use it as a part of wireshark, then UI will be a bit better.

raould, 2021-07-24:
I think I was capturing the correct interface. There are 2 listed by USBPcap/Wireshark, and only the 1st one has devices on it, and only the 1st one shows any data when i capture. Also, the number of "vendor" things matches the count from Device Manager.

https://drive.google.com/file/d/1Q1db4SstkAPeDK5Llum78_C8g2CE6C2W/view?usp=sharing

I don't understand the dump because I searched for the vendor ID (had to get the endian order right ha ha?) and it looks to me like it was 1.6.0 but then there's nothing from 1.6.0 after the configuration. The only data looks to be USB_INTERRUPT with 1.9.1, which is maybe the hub for the mouse & keyboard.

T-Troll, 2021-07-25:
Uh-oh. It means light device is not USB in this case, like the rest of devices.
Let me RTFM for a while...

T-Troll, 2021-07-25:
... According to [the specs](https://www.dell.com/support/kbdoc/en-us/000179513/alienware-command-center-quick-guide), it should have classical v2-v3 chip inside.
But also have 3rd-party USB controller (ASMEDIA).
So we have 2 possibilities:
- Chip is still connected to USB, but hidden. Can you look for ANY USB device with VID_187c?
- Chip is connected to serial i/o interface (most probably). In this case, we need to find it's address. Can you check SMBIOS info at the any tool like rweverything or hwinfo?

Good news interface is the same like v2-v3 notebooks, we just need to find chip interface. ^_^

raould, 2021-07-26:
Nothing in the device dump that is 187c or anything like that as far as I can tell.

hwinfo / Motherboard / SMBIOS DMI / Port Connectors / Port Connector / type: unknown, internal reference: LED CONTROLLER. 

rweeverything:
```
[Port Connector Information] (Type 8)
08 09 23 00 01 FF 00 00 FF 4C 45 44 20 43 4F 4E	..#......LED CON
54 52 4F 4C 45 52 00 00                        	TROLER..        
Type	0x08 (8)
Length	0x09 (9)
Handle	0x0023 (35)
Internal Reference Designator	String1 - "LED CONTROLER"
Internal Connector Type	0xFF - Other 
External Reference Designator	NULL
External Connector Type	0x00 - None 
Port Type	0xFF - Other 
```

i don't see how to find any more info however?!?!?

T-Troll, 2021-07-26:
> Handle	0x0023 (35)

This is enough for now. Let me work for some days, then i drop you a probe to check.

T-Troll, 2021-07-27:
Uh-oh. I got some info and code.

Good news - There are Dell SMBIOS driver, even in source code. And i plan to add it anyway for fan control.
Bad news - it will be not a short story, so i can't predict then i done.

PS: BTW, all Aurora up to R6 use notebook USB chip. But they decide to play hard in you R7. And we still have probability it's USB, just some OEM like DARFON (AKA RGB keyboard).

T-Troll, 2021-07-31:
I need you help.
I'm working on fan/power control as well as you gear support now.
Can you please install/run `rweverything`, open "ACPI" page, save all data and send it to me?
I want to check you structures and how far it from other ones.
T.I.A.!

WildOne69, 2021-07-31:
I'm gonna be following this. I have an Aurora R7 as well as an Aurora R8 that I can provide info off my systems to you as well. the r8 just supports 8th and 9th gen Intel Core I chips.

T-Troll, 2021-08-01:
@WildOne69 
Welcome!
Is R8 doesn't have USB light device as well as R7? If it have, try to use `alienfx-probe` app and share the probe output, please!
In any case, i will be appreciated for ACPI dump from `rweverything`.

raould, 2021-08-01:
https://drive.google.com/file/d/1oMbYhgoZfEudqmF4BaG6hSc9OkmKY_-Z/view?usp=sharing please let me know when you've downloaded it :-)

T-Troll, 2021-08-02:
Ok, thank you! I remove a link.

T-Troll, 2021-08-07:
Status update - i make a new tool for direct ACPI access, this can be used for you lights too. It doesn't support you hardware yet, but will do. You can check early beta [here](https://github.com/T-Troll/alienfan-tools).

T-Troll, 2021-09-23:
Status update - i support fan control now, but still looking for ACPI function to control lights. Seems like R7 the only one requires this kind of access, so still looking for examples.

raould, 2021-09-23:
>  "Serial IO I2C host controller"? Should have Alienware VID.

which app should i use to check? in windows system information, i see only GTX and Audio irq's when searching for "Serial IO I2C..."

T-Troll, 2021-09-24:
Common Windows "Device Manager". Then "Properties->Details->Hardware IDs".
Look for VID_187C.

I'm trying to find the control interface inside AWCC ACPI methods now, but still don't locate it.

raould, 2021-09-24:
Device Manager > View > Devices by container > Desktop XXXXX > Intel(R) Serial IO I2C Host Controller

A2E0 and A2E1

PCI\VEN_8086&DEV_A2E0&SUBSYS_08581028&REV_00
PCI\VEN_8086&DEV_A2E0&SUBSYS_08581028
PCI\VEN_8086&DEV_A2E0&CC_118000
PCI\VEN_8086&DEV_A2E0&CC_1180

PCI\VEN_8086&DEV_A2E1&SUBSYS_08581028&REV_00
PCI\VEN_8086&DEV_A2E1&SUBSYS_08581028
PCI\VEN_8086&DEV_A2E1&CC_118000
PCI\VEN_8086&DEV_A2E1&CC_1180

T-Troll, 2021-09-24:
Switch to "Devices by connection" and check what is connected to these controllers, please!
As i said before, look for VEN_187C.

T-Troll, 2021-09-24:
PS: If we still not locate device, let's try to grab AWCC to interface communication. I'll drop you the instruction.

raould, 2021-09-24:
Honestly unfortunately nothing found yet with 187C from all the GUI or complete device dumps I've done thus far, so I am wondering if maybe there's another vendor ID that alienware/dell could be using? 

"Devices by connection" > DESKTOP-XXXXX > ACPI x64-based PC > Microsoft ACPI-Compliant System > PCI Express Root Complex > Intel(R) Serial IO I2C Host Controller {A2E0,A2E1} ... those do not show any sub-tree, they do not show any children connected to them, they are leaf nodes in the GUI.

T-Troll, 2021-09-24:
According to Dell specs for your gear, you should have a common Alienware light control chip, v4 in my API.
I donno how (and for what) they hide it so deep.... AFAIK other desktops have common HID interface for it.

Ok, let's try to catch them.
1. Install [USBPcap](https://desowin.org/usbpcap/).
2. Install, reboot, run it, choose one of the controllers (which one - we need to find) and file name (abcd.pcap, for example).
3. Run AWCC, change some colors.
4. Stop USBPcap.
5. Check file - if it has small size (some kb) - it's not what we are looking for. Try other controller. It should be about 100kb+ in size.
6. If you found it - you can take a look yourself using [WireShark](https://www.wireshark.org/) and send it to me for analysis.

raould, 2021-09-24:
hm didn't we do this before up above in this thread, or will this be different data?

T-Troll, 2021-09-24:
Yes, we do. I forget, sorry.
Buuuut.... Seems like i found in ACPI.
Take a look at your dump, search for `Method(SETC`!
Seems like it's what we're looking for.

Bad news, i can't include it into AlienFX directly. Good news, i can use this into AlienFan.

raould, 2021-09-24:
When I look for the SETC I see it in a few places in the ACPI dumps. There's
C5SETC, _SB_AMW1SETC, CM72pCM73bICPCSETC
dunno if that means anything/useful. :-)

T-Troll, 2021-09-24:
No, look for EXACT "Method(SETC".
Looks like it is. Do you have 3 lights?

raould, 2021-09-24:
I am not finding "Method(SETC" anywhere, what should I use to dump it? Note that RW-Everything gives me an error when I try to use it to get ACPI data, so I've used msinfo32 and devcon, I have not seen that string in the output. 

From devcon I did see:
ACPI\PNP0C14\AWCC
    Name: Microsoft Windows Management Interface for ACPI
    Driver is running.

T-Troll, 2021-09-24:
I don't need it, i use direct ACPI method calls.

So.... I provide you the test version of `alienFAN-cli` soon (today or tomorrow). You can check did it change lights and some another things i found.
If all works - i'll think how to integrate it. But seems like it will be part of alienfan project, not alienfx ('cause it require ACPI access).

T-Troll, 2021-09-24:
So, here are test version for you - [alienfan-cli.zip](https://github.com/T-Troll/alienfx-tools/files/7223654/alienfan-cli.zip)

Unpack it (antivirus can complain!), then run `alienfan-cli` with some parameters. First, you can check is your hardware supported or not for fan control (it should be) - use, for example, `alienfx-cli rpm` or without parameters - you will see in output.
Even if it says "not supported", try to issue these commands:
- `alienfan-cli resetcolor` - it should reset your color system, but i don't know what stage it will have after.
- `alienfan-cli setcolor=light,r,g,b` - it should set light(s) to given color. WARNING! "Light" here is not the number, but bitmask. So use 1,2,4 (or combination for set more, than one).
- `alienfan-cli setcoloreffect=mode,num` - I have no idea what it does, but something related to color system. "mode" can be in (1,3,4,6,7,9,10,12,13,15), "num" is 0 or 1 (enable/disable?).

Please, share results with me.

raould, 2021-09-24:
hi,
"resetcolor" opens a new command output window and says
```
AlienFan-cli v1.2.2.0
Supported hardware v4 detected, 3 fans, 3 sensors, 2 power states.
```
but then hangs and i can't even kill the window or process, have to manually power cycle.

T-Troll, 2021-09-25:
Interesting...
So detection passed, but calling method loop inside. Ok, let me check.

PS: Did you check other commands?

T-Troll, 2021-09-25:
Upd: Well, it's possible if light system busy (AWCC doing this?). The code inside `\_SB.AMW1.SRST`:
```
Store(One, IC6C)
While(One)
{
	If(LEqual(IC9C, One))
	{
		Break
	}
}
```
So it waits endlessly until light system reset done.
Noted. Let me think how to cut it from wait.

T-Troll, 2021-09-25:
Oh, i found system lights init:
```
		\_SB.AMW1.ICPC()
		\_SB.AMW1.SETC(Zero, Zero, Zero, 0x04)
		\_SB.AMW1.SETC(Zero, Zero, Zero, 0x02)
		\_SB.AMW1.RCPC()
		\_SB.AMW1.SETB(0x0F, Zero)
```
So i need to modify code to keep this sequence. Wait a little.

T-Troll, 2021-09-25:
Ok, try this - [alienfan-cli.zip](https://github.com/T-Troll/alienfx-tools/files/7228990/alienfan-cli.zip) (NB: Reboot your system before use, i change the driver code, so need to reload it).
I've adds waiting timeout, so method call will fail if not finished in reasonable time.
Also, i change `resetcolor` and `setcolor` logic. Seems like `resetcolor` not needed, try `setcolor` first.
And seems like `setcoloreffect` should set brightness, not effect.

raould, 2021-09-25:
it always opens another command window to show the output which then goes away as soon as the command finishes so it is hard to see what it is reporting and hard to copy-paste it here.

BUT, success with "alienfan-cli.exe setcolor=2,0,255,0" to change the side lights, it really worked!

light 1 doesn't seem to work.
light 2 is the right hand side.
light 4 is the left hand side.
light 8 doesn't seem to work.

T-Troll, 2021-09-25:
> it always opens another command window to show the output

It's because it requires admin. Start administrator command prompt and run it from it.

> BUT, success with "alienfan-cli.exe setcolor=2,0,255,0"

Great. You can also try to mix lights - use 3 or 5 or 7 as number. It's strange 1 not works (you have it, but seems like it is power button, maybe need some additional magic). 8 and higher will not work at all, as i see into code.

What about "setcoloreffect"? What it does?

T-Troll, 2021-09-25:
Now i think how to integrate it... I don't want always share ACPI code. Maybe additional library for this, so fan and gui will use it. But not sure about ambient and haptics.

raould, 2021-09-27:
Thanks for working on this. Lighting commands were ok but things like "rpm" gave empty output.

T-Troll, 2021-09-27:
Thank you for feedback, i'll check. What about temperatures and fan control?
What "setcolormode" do? (please use the latest release). Seems like it should be brightness and on/off lights.

raould, 2021-09-29:
what do you mean by latest release? the zip from just up above? where is the "setcolormode" command docs? 

other things look to work
```
C:\Users\x\Downloads\alienfan-cli>.\alienfan-cli.exe rpm
AlienFan-cli v1.2.2.0
Supported hardware v4 detected, 3 fans, 3 sensors, 2 power states.
Fan#0: 5237
Fan#1: 1103
Fan#2: 400

C:\Users\x\Downloads\alienfan-cli>.\alienfan-cli.exe temp
AlienFan-cli v1.2.2.0
Supported hardware v4 detected, 3 fans, 3 sensors, 2 power states.
CPU Internal Thermistor: 37
GPU Internal Thermistor: 31
FAN#3 sensor: 34
```

T-Troll, 2021-09-29:
Thank you. What about `getfans` and `setfans` (but stop AWCCServiece first!)?
Can you please try `alienfan-cli rpm unlock setfans=50,50,50 getfans rpm`?

Last release is 1.3.0 is at GitHub - [here](https://github.com/T-Troll/alienfan-tools/releases) (i publish it yesterday).
At your beta, this command name is `setcoloreffect`, but it is not right, so i change it into release.
I need to check - is this command control lights brightness (seems like this) and on/off state?

T-Troll, 2021-10-12:
Thanks. `setcolormode` have 2 parameters. Should be brightness and on/off or some effects.
Can you try `setcolormode=12,1`, `setcolormode=7,0`, `setcolormode=1,1`, `setcolormode=1,0`?
If this doesn't change anything - the task is closed.

PS: Your system light control already integrated into `alienfx-gui`, but you need to enable fan control in "Settings" to use it (just set Power mode to 2nd value at "Fans" tab if you don't want to control fans).

T-Troll, 2021-10-12:
Ooops!

It's a bug, sorry, thanks to reveal it. I will fix it at next release.
Please add 2 more arguments for now (any value, it will not be used) - `setcolormode=1,0,1,1`

raould, 2021-10-12:
setcolormode *does* change the lights, it is doing something at least. :-)

T-Troll, 2021-10-12:
Full support added in [AlienFx-Tools 5.0.4](https://github.com/T-Troll/alienfx-tools/releases/tag/5.0.4) and [AlienFan-Tools 1.3.1](https://github.com/T-Troll/alienfan-tools/releases/tag/1.3.1)

closed 2021-10-12 by T-Troll