T-Troll/alienfx-tools #29: Dell G7 7500 laptop sorta works, sorta doesn't :-)

raould, 2021-07-24:
it sort of does, sort of does not work :-)
- The GUI doesn't seem to work correctly, or I don't understand the user interface. 
- I did successfully change all the colors using the CLI with set-all=0,255,0.
- The status always says Position Unknown and Color Unknown however.

- Dell G7 7500 laptop, 0VF32T, bios 1.9.0
- Windows 10 Version	10.0.19042 Build 19042
- v3.1.3 just installed
- AWCC 5.3.2.0

```
c:\Program Files\AlienFX Tools>alienfx-cli.exe high-level status
alienfx-cli v3.1.3
Low-level device ready
Dell API ready
Dell API version 5.2.0.0 detected
Devices found: 1
Device #0, Name: Dell G7 7500, Type: Notebook, Lights: 16
  Light #0, Name: Light Bar Zone 1, Position Unknown, Color Unknown
  Light #1, Name: Light Bar Zone 2, Position Unknown, Color Unknown
  Light #2, Name: Light Bar Zone 3, Position Unknown, Color Unknown
  Light #3, Name: Light Bar Zone 4, Position Unknown, Color Unknown
  Light #4, Name: Light Bar Zone 5, Position Unknown, Color Unknown
  Light #5, Name: Light Bar Zone 6, Position Unknown, Color Unknown
  Light #6, Name: Light Bar Zone 7, Position Unknown, Color Unknown
  Light #7, Name: Light Bar Zone 8, Position Unknown, Color Unknown
  Light #8, Name: Light Bar Zone 9, Position Unknown, Color Unknown
  Light #9, Name: Light Bar Zone 10, Position Unknown, Color Unknown
  Light #10, Name: Light Bar Zone 11, Position Unknown, Color Unknown
  Light #11, Name: Light Bar Zone 12, Position Unknown, Color Unknown
  Light #12, Name: KB Left, Position Unknown, Color Unknown
  Light #13, Name: KB Center-Left, Position Unknown, Color Unknown
  Light #14, Name: KB Center-Right, Position Unknown, Color Unknown
  Light #15, Name: KB Right, Position Unknown, Color Unknown
Done.

c:\Program Files\AlienFX Tools>alienfx-cli.exe status
alienfx-cli v3.1.3
Low-level device ready
Dell API ready
Device VID#6268, PID#1360 - Device #550 (Active, V4)
Done.
```

raould, 2021-07-24:
```
__GENUS                     : 2; __CLASS                     : Win32_PnPEntity; __SUPERCLASS                : CIM_LogicalDevice; __DYNASTY                   : CIM_ManagedSystemElement; __RELPATH                   : Win32_PnPEntity.DeviceID="HID\\INTC816&COL0F\\3&36A7043C&0&000E"; __PROPERTY_COUNT            : 26; __DERIVATION                : {CIM_LogicalDevice, CIM_LogicalElement, CIM_ManagedSystemElement}; __SERVER                    : MACHINEXYZ; __NAMESPACE                 : root\cimv2; __PATH                      : \\MACHINEXYZ\root\cimv2:Win32_PnPEntity.DeviceID="HID\\INTC816&COL0F\\3&36A7043C&0&000E"; Availability                : ; Caption                     : HID-compliant vendor-defined device; ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}; CompatibleID                : ; ConfigManagerErrorCode      : 0; ConfigManagerUserConfig     : False; CreationClassName           : Win32_PnPEntity; Description                 : HID-compliant vendor-defined device; DeviceID                    : HID\INTC816&COL0F\3&36A7043C&0&000E; ErrorCleared                : ; ErrorDescription            : ; HardwareID                  : {HID\INTC816&Col0F, HID\VID_8087&UP:FFB3_U:0010, HID_DEVICE_UP:FFB3_U:0010, ;                               HID_DEVICE_UPR:FF00-FFFF...}; InstallDate                 : ; LastErrorCode               : ; Manufacturer                : (Standard system devices); Name                        : HID-compliant vendor-defined device; PNPClass                    : HIDClass; PNPDeviceID                 : HID\INTC816&COL0F\3&36A7043C&0&000E; PowerManagementCapabilities : ; PowerManagementSupported    : ; Present                     : True; Service                     : ; Status                      : OK; StatusInfo                  : ; SystemCreationClassName     : Win32_ComputerSystem; SystemName                  : MACHINEXYZ; PSComputerName              : MACHINEXYZ
__GENUS                     : 2; __CLASS                     : Win32_PnPEntity; __SUPERCLASS                : CIM_LogicalDevice; __DYNASTY                   : CIM_ManagedSystemElement; __RELPATH                   : Win32_PnPEntity.DeviceID="HID\\INTC816&COL11\\3&36A7043C&0&0010"; __PROPERTY_COUNT            : 26; __DERIVATION                : {CIM_LogicalDevice, CIM_LogicalElement, CIM_ManagedSystemElement}; __SERVER                    : MACHINEXYZ; __NAMESPACE                 : root\cimv2; __PATH                      : \\MACHINEXYZ\root\cimv2:Win32_PnPEntity.DeviceID="HID\\INTC816&COL11\\3&36A7043C&0&0010"; Availability                : ; Caption                     : HID-compliant vendor-defined device; ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}; CompatibleID                : ; ConfigManagerErrorCode      : 0; ConfigManagerUserConfig     : False; CreationClassName           : Win32_PnPEntity; Description                 : HID-compliant vendor-defined device; DeviceID                    : HID\INTC816&COL11\3&36A7043C&0&0010; ErrorCleared                : ; ErrorDescription            : ; HardwareID                  : {HID\INTC816&Col11, HID\VID_8087&UP:FFB3_U:0010, HID_DEVICE_UP:FFB3_U:0010, ;                               HID_DEVICE_UPR:FF00-FFFF...}; InstallDate                 : ; LastErrorCode               : ; Manufacturer                : (Standard system devices); Name                        : HID-compliant vendor-defined device; PNPClass                    : HIDClass; PNPDeviceID                 : HID\INTC816&COL11\3&36A7043C&0&0010; PowerManagementCapabilities : ; PowerManagementSupported    : ; Present                     : True; Service                     : ; Status                      : OK; StatusInfo                  : ; SystemCreationClassName     : Win32_ComputerSystem; SystemName                  : MACHINEXYZ; PSComputerName              : MACHINEXYZ
__GENUS                     : 2; __CLASS                     : Win32_PnPEntity; __SUPERCLASS                : CIM_LogicalDevice; __DYNASTY                   : CIM_ManagedSystemElement; __RELPATH                   : Win32_PnPEntity.DeviceID="HID\\VID_046D&PID_C537&MI_01&COL06\\7&3058295C&0&0005"; __PROPERTY_COUNT            : 26; __DERIVATION                : {CIM_LogicalDevice, CIM_LogicalElement, CIM_ManagedSystemElement}; __SERVER                    : MACHINEXYZ; __NAMESPACE                 : root\cimv2; __PATH                      : \\MACHINEXYZ\root\cimv2:Win32_PnPEntity.DeviceID="HID\\VID_046D&PID_C537&MI_01&COL06\\7&305;                               8295C&0&0005"; Availability                : ; Caption                     : HID-compliant vendor-defined device; ClassGuid                   : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}; CompatibleID                : ; ConfigManagerErrorCode      : 0; ConfigManagerUserConfig     : False; CreationClassName           : Win32_PnPEntity; Description                 : HID-compliant vendor-defined device; DeviceID                    : HID\VID_046D&PID_C537&MI_01&COL06\7&3058295C&0&0005; ErrorCleared                : ; ErrorDescription            : ; HardwareID                  : {HID\VID_046D&PID_C537&REV_3400&MI_01&Col06, HID\VID_046D&PID_C537&MI_01&Col06, ;                               HID\VID_046D&UP:FF80_U:0080, HID_DEVICE_UP:FF80_U:0080...}; InstallDate                 : ; LastErrorCode               : ; Manufacturer                : (Standard system devices); Name                        : HID-compliant vendor-defined device; PNPClass                    : HIDClass [... cut ...]   : HID-compliant vendor-defined device; PNPClass                    : HIDClass; PNPDeviceID                 : HID\VID_046D&PID_C537&MI_01&COL05\7&3058295C&0&0004; PowerManagementCapabilities : ; PowerManagementSupported    : ; Present                     : True; Service                     : ; Status                      : OK; StatusInfo                  : ; SystemCreationClassName     : Win32_ComputerSystem; SystemName                  : MACHINEXYZ; PSComputerName              : MACHINEXYZ
```

T-Troll, 2021-07-24:
> Low-level device ready

It means it supported. Should be no issues.

Unfortunately, i have a serious bug in 3.1.0-3.1.3 - you can't add lights/effects/profiles correctly (compiler broke all ListView if optimization enabled ^). I release a patch yesterday (3.1.3.1) fixing this issue.
So update, start GUI, add light names and feel free to drop down AWCC. Tell me if any issues.
You can also add lights for low-level using Probe CLI as a linux way. ^_^

closed 2021-07-24 by raould