T-Troll/alienfx-tools #46: Dell G5 15 5515

Hellohi3654, 2021-09-11:
Hi I have a 
[https://www.dell.com/en-au/shop/gaming-and-games/g15-ryzen-edition-gaming-laptop/spd/g-series-15-5515-laptop/hng551503au](https://www.dell.com/en-au/shop/gaming-and-games/g15-ryzen-edition-gaming-laptop/spd/g-series-15-5515-laptop/hng551503au)
and I cannot get your tool to work
not sure if it's compatible just checking

Thanks

Hellohi3654, 2021-09-11:
Oh btw I only really want to use your tool for fan control
Thanks

T-Troll, 2021-09-11:
Well, seems like Dell start to utilize Alienware design. How cute!

If you only need fan control, better use my [Alienfan-tools](https://github.com/T-Troll/alienfan-tools) project, it's pure fan control without anything other.

Ok, back to business. I never meet Ryzen-based notebooks, so i don't surprised it have some difference.
Can you please:
- Install [RW Everything](http://rweverything.com/)
- Run, press "ACPI" button (it can take some time)
- Save all dump ("save all" button)
- Send it to me (you can attach it at reply here)

I'll take a look and will try to support it.

Hellohi3654, 2021-09-11:
ok thanks for the reply I'll do that now

Hellohi3654, 2021-09-11:
https://drive.google.com/file/d/1-IlbLN6L7w2iW-FBQpk_nVHqXILmc2J6/view?usp=sharing

Hellohi3654, 2021-09-11:
Also when I try and run Alienfan-tools I get this error
[image]
I have made sure that my antivirus is disabled and that it's running as administrator

T-Troll, 2021-09-11:
Aha. Ok, go it, thanks, please give me some time to analyze (it's later evening here, so expect my reply tomorrow).

Hellohi3654, 2021-09-11:
> Aha. Ok, go it, thanks, please give me some time to analyze (it's later evening here, so expect my reply tomorrow).

Don't worry it's late for me too

Good night

T-Troll, 2021-09-11:
So... I check your dump.

It's close to Alienware, but different in small details - so i need to check why you got error.
By the way, you have 2 power modes, but only one can be selected by power select call!

... Ok, wait for the new release (maybe beta) to test.

PS: Also, maybe admin will not need anymore if driver already run.

Hellohi3654, 2021-09-11:
> So... I check your dump.
>
> It's close to Alienware, but different in small details - so i need to check why you got error.
> By the way, you have 2 power modes, but only one can be selected by power [...]

Ok thanks
I have a button on my keyboard that enables "high performance mode"
Not sure if that switches power modes

Also how would I log the error when I open Alienfan-tools

Thanks

T-Troll, 2021-09-11:
Ok. Can you please run this debug version - [Alienfan-Debug.zip](https://github.com/T-Troll/alienfx-tools/files/7148666/Alienfan-Debug.zip)

It has both GUI and CLI tools.

First, try GUI. If it is not started, you will have one of 2 possible error messages:
- "Can't install driver, check system configuration!" - in this case, the issue is into your system configuration. What you can do for test:
  - Disable Secure Boot in BIOS. Is it work after it?
  - Check readme and issue TestMode command, reboot. Is it work?
  - If both not work, tell me for deep diagnostic.
- "Supported hardware not detected!" - this will mean i have bug in hardware detection, but - according to your dump, you device should be detected. You can try other way:
  - Open Command prompt as Administartor
  - Run CLI - `alienfan-cli direct=13,1,32` and show me output.

PS: In fact, `direct` command is forced access to hardware interface. You can open your dump, search for "Method(WMAX" string, then check what parameters are available.

Hellohi3654, 2021-09-12:
> Ok. Can you please run this debug version - [Alienfan-Debug.zip](https://github.com/T-Troll/alienfx-tools/files/7148666/Alienfan-Debug.zip)
>
> It has both GUI and CLI tools.
>
> First, try GUI. If it is [...]

ah so I run the GUI from alienfandebug and it made my laptop BSOD

Hellohi3654, 2021-09-12:
I ran the GUI again and it didn't BSOD it said Supported hardware not detected

so I followed your steps 
[image]

Hellohi3654, 2021-09-12:
I also ran the normal cli
[image]

Hellohi3654, 2021-09-12:
opened the dump and found this 
[image]

T-Troll, 2021-09-12:
Wow! BSOD is possible, but very interesting how you got it. 
Hmm, "Direct call failed". So you have function, but access denied.
Seems like ACPI chain is different for your gear.
Ok, now i understand the issues now, let me think a bit and provide other test version.

PS: Yes WMAX is a fan/power control routine from Alienware. I only use this commands, so it's safe for hardware.

T-Troll, 2021-09-12:
UPD: Ohh.... I found.

Path is different - my device have _SB.AMW1.WMAX, yours - _SB.AMW3.WMAX !
Ok, i'll patch this.

Hellohi3654, 2021-09-12:
> UPD: Ohh.... I found.
> 
> Path is different - my device have _SB.AMW1.WMAX, yours - _SB.AMW3.WMAX !
> Ok, i'll patch this.

oh cool thanks

Hellohi3654, 2021-09-12:
> Wow! BSOD is possible, but very interesting how you got it.
> Hmm, "Direct call failed". So you have function, but access denied.
> Seems like ACPI chain is different for your gear.
> Ok, now i understand [...]

ok thanks

T-Troll, 2021-09-12:
Let's try [this one](https://github.com/T-Troll/alienfx-tools/files/7148796/Alienfan-Debug.zip)

First, try `alienfan-cli direct=13,1,32` - it should return 1. Also, it should detect fans, powers and sensors (you only have 2 sensors instead of 7 on mine...).

If it's ok, try other CLI commands, like "rpm". If ok, try GUI.

Hellohi3654, 2021-09-12:
sorry for late reply been away this morning

Hellohi3654, 2021-09-12:
i'll try now

Hellohi3654, 2021-09-12:
so it made my laptop BSOD again when trying to do `alienfan-cli direct=13,1,32`

Hellohi3654, 2021-09-12:
tried again and it returned value 1

T-Troll, 2021-09-12:
Yes, it should work now.
But BSOD... I think we need to investigate it. Can you help me?

Hellohi3654, 2021-09-12:
> Yes, it should work now.
> But BSOD... I think we need to investigate it. Can you help me?

Yeah sure it seems to only do it the first time I try to do `alienfan-cli direct=13,1,32`

Hellohi3654, 2021-09-12:
Also when I try opening the gui it just opens in the background and nothing is displayed
[image]

Hellohi3654, 2021-09-12:
it also seems to fully load one thread
[image]

T-Troll, 2021-09-12:
1. What BSOD about? You can check in Event Viewer. Is it protection or driver fault?
2. Let's try to install driver manually:
- Reboot to unload it
- At folder you unpack debug, it's sub-folder named KDU.
- Copy hwacc.sys (better from release version, but from this OK too) into this folder
- Issue the command `kdu.exe -prv 6 -scv 3 -drvn HwAcc -map HwAcc.sys"
If it BSOD, let me check other ways. If not - i'll to to check into app.

Hmmm... So it hangs or looped before UI. Did tray icon visible?
Ok, kill it and let me check and add some tests.

Hellohi3654, 2021-09-12:
Log Name:      System
Source:        Microsoft-Windows-WER-SystemErrorReporting
Date:          12/09/2021 3:00:59 PM
Event ID:      1001
Task Category: None
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      DESKTOP-J718USC
Description:
The computer has rebooted from a bugcheck.  The bugcheck was: 0x0000001e (0xffffffffc0000005, 0xfffff8023ad56ebf, 0x0000000000000000, 0x0000000000000044). A dump was saved in: C:\Windows\MEMORY.DMP. Report Id: 920be000-14b7-46de-8a63-aa070df98a22.
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Microsoft-Windows-WER-SystemErrorReporting" Guid="{ABCE23E7-DE45-4366-8631-84FA6C525952}" EventSourceName="BugCheck" />
    <EventID Qualifiers="16384">1001</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2021-09-12T05:00:59.6493349Z" />
    <EventRecordID>15138</EventRecordID>
    <Correlation />
    <Execution ProcessID="0" ThreadID="0" />
    <Channel>System</Channel>
    <Computer>DESKTOP-J718USC</Computer>
    <Security />
  </System>
  <EventData>
    <Data Name="param1">0x0000001e (0xffffffffc0000005, 0xfffff8023ad56ebf, 0x0000000000000000, 0x0000000000000044)</Data>
    <Data Name="param2">C:\Windows\MEMORY.DMP</Data>
    <Data Name="param3">920be000-14b7-46de-8a63-aa070df98a22</Data>
  </EventData>
</Event>

Hellohi3654, 2021-09-12:
> 1. What BSOD about? You can check in Event Viewer. Is it protection or driver fault?
> 2. Let's try to install driver manually:
>
> * Reboot to unload it
> * At folder you unpack debug, it's sub-folder [...]

I restarted then
I copied the .sys file in kdu and ran `kdu.exe -prv 6 -scv 3 -drvn HwAcc -map HwAcc.sys`
it didn't BSOD and returned value 1

T-Troll, 2021-09-12:
Ok, try new version - [Alienfan-Debug.zip](https://github.com/T-Troll/alienfx-tools/files/7149028/Alienfan-Debug.zip)

It will provide message boxes upon loading to trace where it loops.

Also, can you run CLI without parameters and show me output? (Hint: you can copy text, not picture - let's save the nature %)

UPD: BSOD is access violation from my driver. Quite strange it only happened from time to time, should be stable in this case. I check.

Hellohi3654, 2021-09-12:
Ran CLI without parameters
```
C:\Windows\system32>cd C:\Alienfan\Alienfan-Debug

C:\Alienfan\Alienfan-Debug>alienfan-cli.exe
AlienFan-cli v1.1.1.0
[#] Kernel Driver Utility v1.1.1 started, (c)2020 - 2021 KDU Project
[#] Build at Fri May 14 22:25:32 2021, header checksum 0x3E810
[#] Supported x64 OS : Windows 7 and above
[*] Windows version: 10.0 build 19042
[*] SecureBoot is enabled on this machine
[+] Selected provider: 6
[*] Driver mapping using shellcode version: 3
[+] Driver object name: "HwAcc"
[+] No driver registry key name specified, driver object name will be used instead
[+] Input driver file loaded at 0x00007FF6CB000000
[+] Provider: G.SKILL Trident Z Lighting Control, Name "EneIo64"
[+] Drivers database "drv64.dll" loaded at 0x0000016DDC4B0002
[+] Extracting vulnerable driver as "C:\Alienfan\Alienfan-Debug\EneIo64.sys"
[+] Vulnerable driver "EneIo64" loaded
[+] Vulnerable driver opened
[+] Executing post-open callback for given provider
[+] Driver device security descriptor set successfully
[+] Victim driver map attempt 1 of 3
[+] Extracting victim driver "PROCEXP152" as "C:\Users\zachh\AppData\Local\Temp\PROCEXP152.sys"
[+] Victim driver loaded, handle 0x00000000000000D0
[+] Reading FILE_OBJECT at 0xFFFFD409DCA3DAE0
[+] Reading DEVICE_OBJECT at 0xFFFFD409D9BCAA70
[+] Reading DRIVER_OBJECT at 0xFFFFD409D3770850
[+] Victim IRP_MJ_DEVICE_CONTROL 0xFFFFF804D8892220
[+] Victim DriverUnload 0xFFFFF804D8893280
[+] Loaded ntoskrnl base 0xFFFFF80480400000
[+] Ntoskrnl.exe mapped at 0x7FF77C0E0000
[+] Resolving kernel import for input driver
[+] Resolving payload import
[*] IoDriverObjectType 0xFFFFF804810FB5A0
[*] ObCreateObject 0xFFFFF80480A0E8D0
[*] ObInsertObject 0xFFFFF80480A0D1E0
[*] ObMakeTemporaryObject 0xFFFFF80480B0A9C0
[*] ZwMakeTemporaryObject 0xFFFFF804807F58A0
[*] ZwClose 0xFFFFF804807F38A0
[+] Resolving base shellcode import
[*] MmSectionObjectType 0xFFFFF804810FB510
[*] ExAllocatePoolWithTag 0xFFFFF80480DB2160
[*] ExFreePoolWithTag 0xFFFFF80480DB2140
[*] IofCompleteRequest 0xFFFFF80480684F30
[*] ZwMapViewOfSection 0xFFFFF804807F3BC0
[*] ZwUnmapViewOfSection 0xFFFFF804807F3C00
[*] ObReferenceObjectByHandle 0xFFFFF804809F5BC0
[*] ObfDereferenceObject 0xFFFFF80480608270
[*] KeSetEvent 0xFFFFF80480686B50
[+] Bootstrap code size = 0x4E1
[+] Driver IRP_MJ_DEVICE_CONTROL handler code modified
[+] Run shellcode
[~] Shellcode result: NTSTATUS (0x0)
[+] Victim driver unloaded
[+] Vulnerable driver unloaded
[+] Vulnerable driver file removed
[+] Return value: 1. Bye-bye!

```

Hellohi3654, 2021-09-12:
Tried running alienfan-gui.exe 
[image]

T-Troll, 2021-09-12:
Press OK for continue ^_^ - it will be a pack of messages.

Hellohi3654, 2021-09-12:
> Press OK for continue ^_^ - it will be a pack of messages.

nothing displays after I press ok

T-Troll, 2021-09-12:
Aha! Loop in probe.

Can you run `alienfan-cli rpm`? What output did it have?

Hellohi3654, 2021-09-12:
Ran `alienfan-cli rpm`
```
AlienFan-cli v1.1.1.0
[#] Kernel Driver Utility v1.1.1 started, (c)2020 - 2021 KDU Project
[#] Build at Fri May 14 22:25:32 2021, header checksum 0x3E810
[#] Supported x64 OS : Windows 7 and above
[*] Windows version: 10.0 build 19042
[*] SecureBoot is enabled on this machine
[+] Selected provider: 6
[*] Driver mapping using shellcode version: 3
[+] Driver object name: "HwAcc"
[+] No driver registry key name specified, driver object name will be used instead
[+] Input driver file loaded at 0x00007FF7EA940000
[+] Provider: G.SKILL Trident Z Lighting Control, Name "EneIo64"
[+] Drivers database "drv64.dll" loaded at 0x0000017F94490002
[+] Extracting vulnerable driver as "C:\Alienfan\Alienfan-Debug\EneIo64.sys"
[+] Vulnerable driver "EneIo64" loaded
[+] Vulnerable driver opened
[+] Executing post-open callback for given provider
[+] Driver device security descriptor set successfully
[+] Victim driver map attempt 1 of 3
[+] Extracting victim driver "PROCEXP152" as "C:\Users\zachh\AppData\Local\Temp\PROCEXP152.sys"
[+] Victim driver loaded, handle 0x00000000000000D0
[+] Reading FILE_OBJECT at 0xFFFFDB082193BA30
[+] Reading DEVICE_OBJECT at 0xFFFFDB081EA69E10
[+] Reading DRIVER_OBJECT at 0xFFFFDB081B60E7B0
[+] Victim IRP_MJ_DEVICE_CONTROL 0xFFFFF805A5EB2220
[+] Victim DriverUnload 0xFFFFF805A5EB3280
[+] Loaded ntoskrnl base 0xFFFFF8054E600000
[+] Ntoskrnl.exe mapped at 0x7FF7F5080000
[+] Resolving kernel import for input driver
[+] Resolving payload import
[*] IoDriverObjectType 0xFFFFF8054F2FB5A0
[*] ObCreateObject 0xFFFFF8054EC0E8D0
[*] ObInsertObject 0xFFFFF8054EC0D1E0
[*] ObMakeTemporaryObject 0xFFFFF8054ED0A9C0
[*] ZwMakeTemporaryObject 0xFFFFF8054E9F58A0
[*] ZwClose 0xFFFFF8054E9F38A0
[+] Resolving base shellcode import
[*] MmSectionObjectType 0xFFFFF8054F2FB510
[*] ExAllocatePoolWithTag 0xFFFFF8054EFB2160
[*] ExFreePoolWithTag 0xFFFFF8054EFB2140
[*] IofCompleteRequest 0xFFFFF8054E884F30
[*] ZwMapViewOfSection 0xFFFFF8054E9F3BC0
[*] ZwUnmapViewOfSection 0xFFFFF8054E9F3C00
[*] ObReferenceObjectByHandle 0xFFFFF8054EBF5BC0
[*] ObfDereferenceObject 0xFFFFF8054E808270
[*] KeSetEvent 0xFFFFF8054E886B50
[+] Bootstrap code size = 0x4E1
[+] Driver IRP_MJ_DEVICE_CONTROL handler code modified
[+] Run shellcode
[~] Shellcode result: NTSTATUS (0x0)
[+] Victim driver unloaded
[+] Vulnerable driver unloaded
[+] Vulnerable driver file removed
[+] Return value: 1. Bye-bye!

```

T-Troll, 2021-09-12:
It's log from KDU, not from alienfan-cli! Ah... I see... Funny....
Ok, let me check.

T-Troll, 2021-09-12:
Found. My ACPI return -1 in case of error, yours - 0. Ok, wait for new version.

Hellohi3654, 2021-09-12:
> Found. My ACPI return -1 in case of error, yours - 0. Ok, wait for new version.

oh ok thanks

T-Troll, 2021-09-12:
[Alienfan-Debug.zip](https://github.com/T-Troll/alienfx-tools/files/7149071/Alienfan-Debug.zip)

Better try CLI first, if it shows data below KDU - try gui.

Hellohi3654, 2021-09-12:
> [Alienfan-Debug.zip](https://github.com/T-Troll/alienfx-tools/files/7149071/Alienfan-Debug.zip)
> 
> Better try CLI first, if it shows data below KDU - try gui.

ok thanks trying now

Hellohi3654, 2021-09-12:
yeah cli seems to be working fine now
not showing any kdu stuff

T-Troll, 2021-09-12:
KDU is used one time per session (until reboot) to load the driver. But did it finish or still in loop?

Hellohi3654, 2021-09-12:
ok everything seems to be working now
what does the gpu slider do?

Hellohi3654, 2021-09-12:
actually the cpu fan doesn't seem to be working at all

Hellohi3654, 2021-09-12:
gpu fan works fine

T-Troll, 2021-09-12:
> actually the cpu fan doesn't seem to be working at all

Can you explain please? What do you mean "don't work"?

GPU slider alter GPU power limit for some systems (i can check yours as well) - left one no limit, right one high limit. At my gear left one 90W, right one 65W.

Hellohi3654, 2021-09-12:
so if I set Power to manual and set the gpu to 0 it will be low power but if I set it to 4 it will be high power?
I only have level 1 and manual in my Power dropdown

Oh also I don't think fan control is actually working at all
When I set temp 20 boost 100 and temp 100 boost 100 on both my fans they do nothing

Hellohi3654, 2021-09-12:
> so if I set Power to manual and set the gpu to 0 it will be low power but if I set it to 4 it will be high power?
> I only have level 1 and manual in my Power dropdown
>
> Oh also I don't think fan [...]

once I close the GUI fan#1 follows my fan curve but fan#0 is still off

T-Troll, 2021-09-12:
It depends of hardware.
For mine - power mode 4 and gpu 4 is power-save.
For you.... I have bad news. You have 3 power modes (manual, code 97, code 99) - but you only can switch to manual. The rest require some other stuff i don't know (but check later on). For GPU - it have 4 modes support, but they do nothing at all (check "Method(GPS" function 0x13 - they just return "Ok" but do nothing).

As far as i see, boosts should work (they set it into hardware, at least). What do you expect they doing? In real, minimal RPMs defined in BIOS and better not being altered. Boost 100 adds about 1000 rpm ABOVE current bios level.
At current hardware, CPU can't work at fan off - it's so hot and they save so much money at cooling system.

Any fan will only follow curve if GUI running - it's a software feature, not hardware. Fans will follow BIOS instead.

T-Troll, 2021-09-12:
Ok, so i release this version as a BASIC support for G15, mess in power modes need to be studied.
You can also play with direct calls in CLI to find other control ways meanwhile - possible combinations can be found in you ACPI dump.

Hellohi3654, 2021-09-12:
> It depends of hardware.
> For mine - power mode 4 and gpu 4 is power-save.
> For you.... I have bad news. You have 3 power modes (manual, code 97, code 99) - but you only can switch to manual. The rest [...]

Oh ok

Well I think fan control is not working at all then

When I set every temp to boost 100 it doesn't change anything until I close it
Even then it's only fan#1 that's running 
The CPU is at like 95C and the GPU is cool because fan#1 is working
Fan#0 only works if I restart and then don't open alienfan
I have also tried setting the boost to a lower value but that doesn't seem to do anything either
Also when I only select fan#0 (Fan 1 in the GUI) the graph to change the fan curve displays nothing and I can't change anything
But when I select both Fan 1 (Fan#0 in CLI) and Fan 2 (Fan#1 in CLI) or just Fan 2 it shows the graph and I can change the fan curve but it doesn't actually affect the fan until I close the GUI (but that's probably just it going back to normal and realizing that the GPU is too hot)
Even after closing the GUI the CPU fan doesn't spin and the CPU temp is 90+ (which means it is probably thermal throttling) and my laptop has to be restarted for Fan 1 to spin again

Hellohi3654, 2021-09-12:
> Ok, so i release this version as a BASIC support for G15, mess in power modes need to be studied.
> You can also play with direct calls in CLI to find other control ways meanwhile - possible combinations can be found in you ACPI dump.

ok cool thanks

T-Troll, 2021-09-12:
> Well I think fan control is not working at all then

I see. Seems like you have set not the boost, but something other. Let me study you ACPI dump more (and i'll ask you to issue some `alienfan-cli direct=` commands to check).
We can control, but now need to make it the right way.

Hellohi3654, 2021-09-12:
yeah somethings definitely not right
cli says
```
C:\Alienfan\Alienfan-Debug>alienfan-cli getfans
AlienFan-cli v1.1.1.0
Supported hardware detected, 2 fans, 2 sensors, 2 power states.
Fan#0 now at 0
Fan#1 now at 0
```
but AWCC says
[image]

T-Troll, 2021-09-12:
No, it's ok - you in auto mode, boosts is 0. AWCC show percents (BTW, not so precise), i can do it as well.

Okk.... Interesting... Try to issue real chord: `alienfan-cli rpm unlock setfans=50,50 getfans rpm`

T-Troll, 2021-09-12:
PS: And, by the way, stop AWCCService - it can reset fans...

Hellohi3654, 2021-09-12:
> PS: And, by the way, stop AWCCService - it can reset fans...

ok thanks
 I will try doing `alienfan-cli rpm unlock setfans=50,50 getfans rpm`

Hellohi3654, 2021-09-12:
ok so I closed AWCC service and everything to do with it in task manager then I tried doing `alienfan-cli rpm unlock setfans=50,50 getfans rpm` and it works perfectly with both fans
I then tried opening the GUI and both fans instantly stopped 
I then tried to set fan curves and it still doesn't work
I went back to the cli and got the RPM of the fans and it says 0 for both
I think the issue is to do with the GUI the cli seems to be working fine

[57 more replies]

closed 2021-09-23 by T-Troll