T-Troll/alienfx-tools #48: Dell G5SE - "Supported hardware not detected"

LittleVulpix, 2021-09-14:
**Describe the bug**
Seems G5 SE 5505 (4800H / 5600M) is not supported even though it seems like it should be

**To Reproduce**
Steps to reproduce the behavior:
I used latest version (4.2.2), UAC disabled, ran alienfan-gui / cli as admin, got the "supported hardware not detected" error.

**Expected behavior**
Able to control fans of my laptop.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: W10 x64 20H2 (19042.1202)

**Additional context**
I read about needing test mode but also read that you don't need it anymore due to some kind of driver hack. I didn't get any warning or so so I assume the hack is working - but the control is not.

I can help with providing any dumps that may be needed.

T-Troll, 2021-09-15:
Well... Only Intel-based G-series supported for now, i have this into readme. As well as Radeon not supported.
Now we fight with Ryzen-based G15 [here](https://github.com/T-Troll/alienfx-tools/issues/46), but still not works well.

If you have "Hardware not supported" - it means my SDK can't find command interface. Driver issues have other message - "Can't install driver".

So... Can you please install [RWEverything](http://rweverything.com/), save your ACPI dump and share it with me?
I'll check what we can do.

LittleVulpix, 2021-09-15:
Oh, my bad. I couldn't find the info in readme because I was looking at alienFX readme and not the alienfan one. My bad!

Here's the dump:

[AcpiTbls.zip](https://github.com/T-Troll/alienfx-tools/files/7166357/AcpiTbls.zip)

Thank you very much!

T-Troll, 2021-09-15:
It's my fault - so lazy to organize documentation right way...

So, i check your ACPI dump.

Bad news - Dell never provide 2 notebook lines with the same settings (as well as desktop ones)! You have other device NAME again - Mine and all Alienware has _SB.AMW1, G15 have _SB.AMW3, yours G5 - _SB.AMWW!
Also, they drop the whole control group i use for fan detection, so it will be different for your gear.
As well as make a mess with temperature readings from AWCC interface, i don't sure it will work correctly.
And, of course, you will not have GPU power control - it's for Nvidia only, but you can try to find analog for Radeon yourself.

Good news - The rest looks ok, you have 4 power modes (including manual), and boost read/write.

So let me add your interface version into the code, but better wait, then we are finish with G15 for precise fan control.

I'll drop you beta for compatibility test soon.

T-Troll, 2021-09-15:
So, here is beta - [alienfan-tools.zip](https://github.com/T-Troll/alienfx-tools/files/7166559/alienfan-tools.zip) with you gear support.

I recommend do not use GUI for now, but check CLI instead - start it from admin command prompt and play with switches to be sure all works well.

Also, read my last post into #46 and check your boost type - it should be PWM as well, so let me know which type works for you.

LittleVulpix, 2021-09-15:
Thank you for the test version!

I tried launching the cli - it starts but hangs right after announcing its version.

```
C:\Programs\Programs\alienfx-tools>alienfan-cli.exe
AlienFan-cli v1.1.1.0
```

In fact I don't use AWCC right now (just nbfc, but the problem is I cannot control the GPU fan based on GPU temperature so I wanted to give your tool a try).

Anything else I can try to provide to help?

T-Troll, 2021-09-15:
Thanks for support, it means i have a bug into new detection system.
Let me fix it and share updated version.

PS: I don't use AWCC as well - it was my target, then i develop alienfx-tools. ThrottleStop works better for under volt, and fx gui better control lights and fans. I even write LightFX emulation for games ^_^.

T-Troll, 2021-09-15:
Try this version - [alienfan-cli.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7166710/alienfan-cli.exe.zip)
Copy it into the same folder, overwrite old one (some files needed).

I add some checks and debug trace for detection process to see why it loops for your gear.

LittleVulpix, 2021-09-15:
This new version just works, no issue on initialization:

C:\Programs\Programs\alienfx-tools>alienfan-cli.exe getfans
AlienFan-cli v1.1.1.0
Fan 50 found.
Fan 51 found.
Sensor 257 found.
Sensor 262 found.
Power mode 151 found.
Power mode 152 found.
Supported hardware detected, 2 fans, 2 sensors, 4 power states.
Fan#0 now at 0
Fan#1 now at 0

C:\Programs\Programs\alienfx-tools>alienfan-cli.exe temp
AlienFan-cli v1.1.1.0
Fan 50 found.
Fan 51 found.
Sensor 257 found.
Sensor 262 found.
Power mode 151 found.
Power mode 152 found.
Supported hardware detected, 2 fans, 2 sensors, 4 power states.
CPU Internal Thermistor: 42
GPU Internal Thermistor: 41

I tried setting the fans, it's a bit wonky with just setfans:

41 - not spinning
42 - 5pct
50 - 22pct
80 - 59pct
100 - 71pct

But with pwm I really don't know how that works at all... I was not able to get it to spin at any other speed than 66.5% 

Seems none of the values do anything (even if the fan speed did change).

NBFC is able to control both fans reasonably fine, though it doesn't seem like it can bring them above 80pct. But control up to 80 works flawlessly. I don't know if the below thing tells you anything but that's my NBFC setting for fan control itself:

[image]

I tried setting some of the values between the ones you see with pwm but as mentioned, no real change, just 66.5pct on the random values I tried (even if the value of the getfanpwm kept changing).

T-Troll, 2021-09-15:
Aha, so temps looks (can you check then system loaded?) ok... Great.

Most of the fan control tools uses EC access, but i have different approach and control trough ACPI calls - all Alienware and G have interface for AWCC, i'm just RE it and use. It's safe and quite universal (that's why i only spend some minutes to add your system).

According to EC control, you have PWM-based approach, but seems like AWCC interface re-count values for common boost approach. My current PWM code limited to 91 as a maximum, so that's why you have 70%.
Let me check you ACPI code more close, as well as some other checks.

For now, good news this approach works, bad are we still need to configure out limit values.

T-Troll, 2021-09-15:
Oh, i check. You have 0..100 as a boost.
But real speed depend on power/unlock settings, it's indirect. So you can't fry CPU/GPU stopping fans under heavy load.

T-Troll, 2021-09-15:
[alienfan-tools.zip](https://github.com/T-Troll/alienfx-tools/files/7166877/alienfan-tools.zip)

Here you can try both GUI and CLI (i remove debug print as well).
But to check TOP fan speeds you need boost at 100 and hot system!

LittleVulpix, 2021-09-15:
Aha, right. It was the same way in AWCC too - no direct speed, just "offsets". It seems to be working in the gui, I tried and GPU is controlled by GPU temp and CPU by CPU temp. Thank you so much!

T-Troll, 2021-09-15:
> It was the same way in AWCC too - no direct speed, just "offsets".

Yes. As i said, i use AWCC hardware interface, in fact. But this is a more safe way, and my app more flexible.
BTW, look at your cooling system first - There are no CPU and GPU fans, in common, they are just cool CPU or GPU mostly.
So i recommend always set both fans, but in different proportions.

T-Troll, 2021-09-15:
NB: If you still want to have direct fan control also, try to play with AWCC interface:
- Search your AWCC dump for "Method(WMAX" string.
- It has 3 parameters, but only 2 of them used.
- Read the AML code inside it, found suspicious combinations (most probably in group 0x15).
- Try to issue it into hardware using `alienfan-cli unlock direct=xx,xx,xx,xx` command (f.e. `alienfan-cli direct=14,5,32` return your CPU fan RPMs, and `alienfan-cli direct=15,2,33,100` set max boost for GPU fan).

If you found command for your gear to set RPM/PWM directly, i'll support it.

You can also check your dump for Radeon power control method (for Nvidia it's `\\_SB.PCI0.PEG0.PEGP.GPS`, so look into the same zone), and i'll add GPU control as well.

LittleVulpix, 2021-09-15:
Two things I suppose; the power control here is very complex, primarily because this laptop is one of the very few laptops that have smartshift. I don't think I want to play around with it too much.

Secondly, (and this I kind of noticed in AWCC as well), but the cooling doesn't really work well for the CPU temperature. As mentioned, in AWCC as well, the CPU temp from the EC was always a lot lower than the actual CPU temp reported by CPU, which makes sense if they are measured from some surface rather than internally, and the G5SE AMD version is notoriously known for bad cooling solution.

That said I need to control the CPU fan more proactively based on this so I'm going to play around with combination of nbfc and your tool.

And also yes I know, the fans are not really "separate", so I always set them both to the same thing; but it can be that I have some load that only stresses the GPU (or only the CPU) and in such case I want to spin up the fans more prominently "on that side", so to speak.

T-Troll, 2021-09-15:
I got all power control modes from ACPI. You have 4 of them:
- ID 0 (manual)
- ID 97
- ID 98
- ID AB (BTW, it's hidden in modes list, and only available into the function)

I have no idea what PLs they set for CPU/GPU, you can check yourself using HWINFO.
I got it from function 0x15,0x1 of WMAX - you can check yourself.

Yes, fans is a menace. Dell make notebooks for cut the steak, not for work fast and long work load. For Alienware m-series overheat is near permanent issue. Question are - is BIOS fan control + boost squeeze all juice from fans at full load? For my gear, it is (but i still think i need to use custom fans, i don't care about noise).

I don't get temps from EC, but from ACPI. Maybe it's wrong as well, but for my gear is close to real. Additionaly, at my gear i have 5 more temp sensors, even ambient one, and can tune fans for it.

Yes, different fan settings for different load. And yes, set them to 100% all the time not a good idea (as Performance mode do). But set opposite fan at 40-50 boost at asymmetric loading helps to cool too.
It was one of my target then i design this app. Any sensor now can control any fans at different level (resulted one will be highest). It's what i miss into AWCC, as well as multi-point curve.

BTW, please keep in mind - dynamic fan boost only works then GUI running - it's software, not hardware!

T-Troll, 2021-09-16:
I assume that's all for now.

closed 2021-09-16 by T-Troll

LittleVulpix, 2021-09-16:
Yes it is! Thanks, it's working quite well - and you now have one more machine type you support! :D