T-Troll/alienfx-tools #56: AW3420DW support

Ronny011, 2021-10-03:
Is there any way to get support on AW monitors? It's separate from the laptop and desktop drivers from what I can tell.

The are a lot of light all around this monitor and its stand so I had this dream where ambient mode plus these lights
will serve as sort of an extension of the screen and look real cool, Phillips Hue style.

If there any way I can help with making this happen I would be glad :)

T-Troll, 2021-10-03:
I will be glad to add support for other devices, but i need your support for this.
First, we need to figure out how it connected to AWCC. Did it use USB? Or DP/HDMI is enough?
If USB, we need to locate monitor device and catch and RE interface.

So... If you are ready to help me (i will ask you to do some tests later... and more tests.... and even more) - check this question. 

If it's USB - check how it will present into your system - i need VID/PID for start.

Ronny011, 2021-10-03:
I was offered this driver for my monitor:
https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=3ch8r&oscode=wt64a&productcode=dell-aw3420dw-monitor

So I assume AWCC is present.
Secondly, I have a USB connection from monitor to PC.
screenshot of HID details:
https://imgur.com/a/7GgbtbO

T-Troll, 2021-10-03:
Thank you.

This device is USB hub integrated into the monitor. Can you please select it, then switch to "Device by connection" and check devices connected into it?

Ronny011, 2021-10-04:
Sorry for the late response. I've added the details you've asked for to the imgur link (https://imgur.com/a/7GgbtbO)

T-Troll, 2021-10-04:
Thanks. 

Aha, we found it - it's a software component under hub (first one). What's why i can't detect it.

Now the next step - we need to grab the interface between AWCC and device for analysis.
To doing so, please follow these steps:
1. Install [USBPcap](https://desowin.org/usbpcap/).
2. Reboot, run it, choose one of the controllers (you need the one with monitor device inside) and file name (abcd.pcap, for example), keep it runnning.
3. Run AWCC, change some colors.
4. Stop USBPcap.
5. Send me the file for analysis (you can attach it right here - see bottom line).
6. You can take a look inside yourself using [WireShark](https://www.wireshark.org/) if you like.

Ronny011, 2021-10-11:
I will send another file as soon as possible. Cool that there is this amount of control!

Ronny011, 2021-10-30:
I'm not sure if I'm missing something, but the exe doesn't open any windows and I don't thing that I'm getting any noticable output or some new file in my desktop.

T-Troll, 2021-10-30:
Start it from command line (cmd).

Ronny011, 2021-10-31:
Would that be enoug
[image]
h?

T-Troll, 2021-10-31:
Good, but better copy all text - i need to check all devices.

Ronny011, 2021-11-01:
Actually this is the only thing showing for me.

===== New device VID0424, PID2745 =====
Version 512, Blocksize 12
Output Report Length 65, Input Report Length 65, Feature Report Length 0
Usage ID 1, Usage Page ff00
Output caps 1, Index 64
+++++ Detected as: Microship,  +++++

T-Troll, 2021-11-01:
Ok, let's try this - [testtools.zip](https://github.com/T-Troll/alienfx-tools/files/7453376/testtools.zip)

1. Run DeviceTester first. It should show "Microchip, APIv6" at "+++" string if detected correctly.
2. Run SampleApp. There are will be some output, i need it all. It also will change the color of 2 lights on the monitor if we are lucky.

T-Troll, 2021-11-06:
I create support server in Dicsord, you can [join it](https://discord.gg/XU6UJbN9J5) for rapid communication.

Ronny011, 2021-11-21:
Great. I can do Saturdays and Sundays. At what time would you prefer?

T-Troll, 2021-11-21:
Fortunately, we are made a great process for mouse support (see #73 ), it uses the same interface (but different protocol).
So all we need to do - test about commands and chains are correct.

I'll provide you a test tool today, as well as instructions. Run and share results, please.

T-Troll, 2021-11-22:
So, here is the testing app - [AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7577778/AlienFX.Sample.App.exe.zip)

What should i ask you to do:
- Stop AWCCService in Services
- Open CMD (command prompt), then run the app.
- It should set 1st light to red, second to green, third to blue.
- Then it waits for you to press enter, and turn all lights off.
- Then it waits for you to press enter, and turn all lights dimmed.
- Then it waits for you to press enter, and turn all lights undimmed.

Can you, please, run it, then share its log and visible result for me?

Ronny011, 2021-11-28:
After running the app from CMD, what I get is the device info and all the three light changes hapenning all at once in the console, with no actual change in the lights themselves. Then I press Enter and the app crashes I think, as It just goes back to CM
[image]
D.

Ronny011, 2021-11-28:
Is there some assumed initial light setting or should it work whether the light are on or off? Does the Initial color make a difference?

Ronny011, 2021-11-28:
AWCCService is stopped BTW

Ronny011, 2021-11-28:
Tried runnig CMD as Administrator, it didn't change the outcome.

T-Troll, 2021-11-28:
> Is there some assumed initial light setting or should it work whether the lights are on or off? Does the Initial color make a difference?

No, it doesn't matter. But better disable effects before run.

> ll the three light changes hapenning all at once in the console, with no actual change in the lights themselves

Not so good. At least, i see interface accept commands.

Crashed? Very interesting. Maybe dragons here.

Ok, thank you, let me think and provide you the new app.

PS: No, it doesn't need Admin.

T-Troll, 2021-11-29:
Uff... I'm going a bit deep into interface... Seems like not so easy as i expect - they use some weird scheme.
So... What i can do fast - make ALL lights change/on/off. For per-light, need to think more.
BTW, you have one more hidden light Maybe for suspend or bottom one have two.

T-Troll, 2021-11-30:
Ok, let's try this one - [AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7622083/AlienFX.Sample.App.exe.zip) then you have time, please.

It will show what it tries to do in console, check, and press Enter for the next step.
What do i need to know:
- Is light changed?
- Is it changed according to app?
- What color lights have after application exit?

PS: Don't forget to stop AWCCservice before testing!

Ronny011, 2021-12-09:
[image]
After running the app, none of the commands worked unfortunately. Tried one time with lights on and once time with lights off.
Should the fact that I can also control the lights from the display itself make any difference?

T-Troll, 2021-12-09:
No. But next time keep it on (i don't enable it before use right now), please.
Ok, negative result is a result too.

Let me think more. Being honest, i don't understand why all lights command fail - it's the same as AWCC one.
Can you please start USBPcap, run my app and drop me the log?

NoCopy, 2022-05-05:
Just picked up the 38" version of this monitor and the ambient lighting feature is exactly what I was looking for (can't believe that isn't available out of the box). What's the status of this ticket? I realize it's showing as still open but did you need some more information as it seems the original poster stopped helping you / etc. 

Are the instructions above still relevant for what you need? or what can I do to help get this completed?

T-Troll, 2022-05-06:
Well... The guy who test it for me is busy and don't reply, and i can't resolve this myself - i have no hardware.
In case you are in - ok, let me remember the point (lights interface was RE, but not working correctly yet) and generate test tool for check.
Meanwhile, please install USBPcap for grabbing interface dumps for me.
After we are continuing from same point.

NoCopy, 2022-05-06:
Sounds good. I have some time on the weekend coming up here so I will run through that guide and send you over the data.

T-Troll, 2022-05-07:
Ok. 
First, please check device IDs (for monitor USB device, you should have it into your system).
Then grab simple AWCC log using USBPcap.

This needed to make sure interface for your monitor the same as before.

If it's the same, we start to play with test app (i found it and rebuild already) to make sure it can be controlled.
If not, i need to RE commands and build another interface version.

NoCopy, 2022-05-08:
The device ID looks to be slightly different from the original poster. I will get you the details soon here but running usbcap shows that I have my mouse and keyboard on the same hub as the monitor so getting lots of interference. It did show that I have a hub with nothing on it however - but briefly changing around my devices I couldn't get it on the different HUB. That probably means the extra hub is not connected to my MB right now (need to hook up the cable). 

So I am going to check my MB and re-arrange my devices this week to see I can isolate the monitor only for proper testing.

T-Troll, 2022-05-08:
Oh, i forget to mention it.
But it's OK for me - i can filter other devices. You can too - USBPcaps logs can be viewed by WireShark.

T-Troll, 2022-09-01:
Continue in #194.

closed 2022-09-01 by T-Troll