T-Troll/alienfx-tools #72: Aurora no longer works with Alienware Laptops

Billybobertjoe, 2021-11-11:
it says in the aroura devices and wrappers tab that light FX is initialised, but my keyboard and mouse lights dont change. i use an alienware M15 R4 and an alienware AW610M mouse

T-Troll, 2021-11-11:
It can be AWCC bug. Try to downgrade AWCC to older version or just play with LightFX.dll component.

Billybobertjoe, 2021-11-11:
what version should i downgrade to?

T-Troll, 2021-11-11:
Any older ("Previous"). Maybe something before May, 2020 will be ok.

By the way - I can't fix Aurora - i'm not the Aurora developer. I can only add you mouse support into AlienFX-Tools, and it's SDK.
Aurora using Dell one, not mine for now, AFAIK.
So if you want the tool to control lights outside AWCC - yes, we can do. But if you're looking for Aurora support - i'm not the right person for it!

closed 2021-11-11 by T-Troll