T-Troll/alienfx-tools #73: Alienware AW610M mouse support.

T-Troll, 2021-11-11:
VID_0461&PID_4EC0.
Software component for AWCC.

T-Troll, 2021-11-11:
Next step:
- Install USBPcap.
- Run USBPcap, select the device this mouse attached into, choose filename (like `mouse.pcap`).
- Run AWCC
- Set some mouse light(s) for some basic colors (like red, green, blue) 1-3 times.
- Stop USBPcap
- Attach USBPcap log file here.

Billybobertjoe, 2021-11-11:
[mouse.zip](https://github.com/T-Troll/alienfx-tools/files/7516937/mouse.zip)

Billybobertjoe, 2021-11-11:
i had to compress it because it said the file itself wasnt supported

T-Troll, 2021-11-11:
Thank you. Let me check.

T-Troll, 2021-11-11:
So, at first glance:
- You have m15, not M15 with per-key RGB keyboard.
- You notebook have 2 light devices - one Alienware v4 - tron, power button; one Darfon v5 - RGB keyboard. Both supported.
- I got you mouse lights input, but need some time for analysis. You can help me telling what exactly you do (which lite which color), protocol is uncommon.

Now i going for analysis, you can help me providing more logs (the same way) for different operations:
- Go dark, Go light, Go dim, undim (for mouse only, it's ok if for all devices as well) (log 2).
- Set one, the same mouse light for different light effects - color (like here), then pulse red-green, then morph green-blue (if you have other effects - it's nice to do it as well, but use different basic color combination for all of them).

T-Troll, 2021-11-11:
PS: You can install WireShark and take a look inside logs yourself, if you want - you devices:
- 1.4.0 - Alienware chip,
- 1.7.0 - Darfon chip,
- 1.11.4 - Mouse chip.

Billybobertjoe, 2021-11-11:
[mouse1.zip](https://github.com/T-Troll/alienfx-tools/files/7517267/mouse1.zip)

T-Troll, 2021-11-11:
Here is your mouse basic protocol:
```
Commands:
ack:
c002

color:
401a07000aff0000
-------- Command
        -- Light ID
          ------ Color

control:
40100c000164002aaaff
-------- Command
        -- ???
          -- Brightness
            --------??? (always the same)

update:
40600700c04e000001
update2:
40600700c04e000100

report:
4003010001

Command chain:
report->ack->[color->update]->report->control->update2 - bulk color set
report->control - set device brightness
```

T-Troll, 2021-11-11:
What's next?
First, let me implement it (1-2 days), then we go to testing:
- I will share simple app for you, the main goal is to check device detected correctly.
- I will share simple app which set some lights. The goal is to make sure device accept commands and execute it correctly.
- Then i release tools with basic support and continue with effects (easier).

So wait to me until i publish test apps for you.

T-Troll, 2021-11-11:
So, this is first test app - [AlienFXDeviceTester.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7517647/AlienFXDeviceTester.exe.zip)

Can you, please:
- Open command line (cmd)
- Run it
- Copy all output it provide (it could find about 6 devices)
- Paste it here

I will check device properties and make sure it detected correctly.
T.I.A!

Billybobertjoe, 2021-11-12:
[Uploading AlienFXTester output.txt…]()

Billybobertjoe, 2021-11-12:
[AlienFXTester output.txt](https://github.com/T-Troll/alienfx-tools/files/7528751/AlienFXTester.output.txt)

T-Troll, 2021-11-13:
Great, thank you!
So detection is correct.
BTW, you have 10(!) lights at mouse - indexes from 1 to 10.

So, stage 2. Run [this app](https://github.com/T-Troll/alienfx-tools/files/7531258/AlienFX.Sample.App.exe.zip), please (from the command line as well).
What i need to know:
1. Is device init ok? (it should output about it)
2. Is device call OK? (it should have output like "SetColor result: X". If X = 1 it's ok, if it's 0 - it requires more tests).
3. If both ok, did it change light color (it should set light#1 to blue, and light#2 to cyan)?

In case all 3 points ok, we have basic support, but most probably we are fail at #2. In this case, i will provide you another testing app, with the different approach.

Billybobertjoe, 2021-11-13:
both failed

T-Troll, 2021-11-13:
So, did it tells "Can't init device!" or just 0 in results and lights not changed?

T-Troll, 2021-11-13:
Ok, anyway, let's try this one:
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7531573/AlienFX.Sample.App.exe.zip)
The task is the same.

Billybobertjoe, 2021-11-14:
3 device(s) detected.
VID: 0x461, PID: 0x4ec0, API v7
API v7
Let's try to set some colors...
SetColor result: 0
SetColor 2 result: 0

Billybobertjoe, 2021-11-14:
seems to have failed

T-Troll, 2021-11-14:
Yes. %-(
Result 0 means driver didn't accept the command.
Ok, let's try this way - [AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7533240/AlienFX.Sample.App.exe.zip)

Task is the same. If fail again, i provide another one, but if next one fail - it will be paused until i configure out the issue.

Billybobertjoe, 2021-11-14:
3 device(s) detected.
VID: 0x461, PID: 0x4ec0, API v7
API v7
Let's try to set some colors...
SetColor result: 1
SetColor 2 result: 1

Billybobertjoe, 2021-11-14:
i didnt see any color change on my mouse or keyboard though

T-Troll, 2021-11-14:
Aha! And did colors changed (on mouse only)? Ok, got it.

Anyway, please, try this one - [AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7533264/AlienFX.Sample.App.exe.zip)

One of them should change colors.

Billybobertjoe, 2021-11-14:
3 device(s) detected.
VID: 0x461, PID: 0x4ec0, API v7
API v7
Let's try to set some colors...
SetColor result: 0, error 57
SetColor 2 result: 0, error 57

Billybobertjoe, 2021-11-14:
hmm

T-Troll, 2021-11-14:
Hardware error.
Ok, what about this (i don't need log, only error code if any non-zero and result) - [AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7533273/AlienFX.Sample.App.exe.zip)

T-Troll, 2021-11-14:
By the way... Stop AWCC service first, it can alter colors.

Billybobertjoe, 2021-11-14:
oh ok

Billybobertjoe, 2021-11-14:
VID: 0x461, PID: 0x4ec0, API v7
API v7
Let's try to set some colors...
SetColor result: 1, error 0
SetColor 2 result: 1, error 0

Billybobertjoe, 2021-11-14:
still no color change

Billybobertjoe, 2021-11-14:
my keyboar and mouse are changing colors with AWCC service disabled. not from the app u sent though. just fro awcc spectrum setting

Billybobertjoe, 2021-11-14:
what now?

T-Troll, 2021-11-14:
Ok, now on pause - let me think more.
I'll share another test app later on.
Yes, hardware effects not depends on AWCC... Hmm... But can interfere with color set!
Can you disable effects on mouse and check again?

Billybobertjoe, 2021-11-14:
how do i disable the effects?

T-Troll, 2021-11-14:
Open AWCC, select mouse and choose global effect (or per-button, don't sure it has global) for "color".

T-Troll, 2021-11-14:
Also, please check another version (but first previous one!): 
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7533292/AlienFX.Sample.App.exe.zip)

Billybobertjoe, 2021-11-14:
i see no "global effect" or "per-button" option

T-Troll, 2021-11-14:
They name it "static button". Left one above colors circle.

Billybobertjoe, 2021-11-14:
I.. still dont see anything like that. Example picture?

T-Troll, 2021-11-14:
[image]

Billybobertjoe, 2021-11-14:
[image]

Billybobertjoe, 2021-11-14:
im on version 5.4.7.0

T-Troll, 2021-11-14:
Should be OK. Maybe in "Settings" now....
Anyway, let's try - both previous and last versions.

Billybobertjoe, 2021-11-16:
No luck

T-Troll, 2021-11-19:
Does it return `Result: 0` or just don't change colors?

Billybobertjoe, 2021-11-19:
doesnt change colors

Billybobertjoe, 2021-11-19:
Be heard that dell removed the “enable 3rd party access” setting. Idk if that has something to do with this

T-Troll, 2021-11-19:
So, let's do another round.

Use this test app - [AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7573245/AlienFX.Sample.App.exe.zip)

Please, follow this process:
- Set MOUSE to steady color (no effects) into AWCC.
- Close AWCC, go to "Services", stop "AWCCService" (no, just close AWCC is not enough for now).
- Run app.
- If colors not changed, look at "SetColor result:" values. If it's zero, we are deep in troubles. But if it's 1, let's check more:
  - Start USBPcap (the same way like before)
  - Run app
  - Stop USBPcap, send me the log to understand what's going wrong at low level.

> I heard that dell removed the “enable 3rd party access” setting. Idk if that has something to do with this

If they enable it by default, it's not an issue. If they remove support - i have their library emulation ^_^

Billybobertjoe, 2021-11-19:
[mouselighttest.zip](https://github.com/T-Troll/alienfx-tools/files/7573339/mouselighttest.zip)

Billybobertjoe, 2021-11-19:
theres the log

T-Troll, 2021-11-19:
Well... Good news - commands coming into device successfully and in right form.
Now i need to check is this CORRECT commands, and maybe i miss the protocol format and need to study more.

T-Troll, 2021-11-19:
Seems like i found an issue.

Let's try this (keep your eye at "SetColor result:", please):
[AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7573467/AlienFX.Sample.App.exe.zip)

Billybobertjoe, 2021-11-21:
3 device(s) detected.
VID: 0x461, PID: 0x4ec0, API v7
Let's try to set some colors...
SetColor result: 1, error 0
SetColor 2 result: 1, error 0

after disabling the AWCC service, the colors changed on my mouse. are the lights on my keyboard supposed to also change too?

Billybobertjoe, 2021-11-21:
so it worked!!!

T-Troll, 2021-11-21:
Greeeat! So we hack it!

What we need to do next:
- First, i drop you another test app - i need to check lights off/on/dim works well.
- If OK, i release new package with BASIC mouse support, and you can use it.
- Then we test light effects and release full support version.

In fact, this is about 85% of full support for now.

No, your keyboard colors should not change, as you can see, i use mouse device only.
But both keyboard and tron already supported, so just download and start last tools release, define light names, and have fun.

After we are finish support, i tickle Aurora guys about AW support as well.

Billybobertjoe, 2021-11-21:
Ok ill do this tomorrow

Billybobertjoe, 2021-11-21:
when can we start? also, is there a way i can integrate this into aurora?

T-Troll, 2021-11-21:
I have GMT+7. ^_^

Ok, let's do the latest test in this round. Here is the app: [AlienFX Sample App.exe.zip](https://github.com/T-Troll/alienfx-tools/files/7577259/AlienFX.Sample.App.exe.zip)

As usual, disable awcc service and start it. What it should do:
- Set 2 mouse lights to white.
- Wait you to press Enter.
- Off all lights and wait for Enter.
- Dim (to half) all lights and wait for Enter.
- Return lights to full brightness and... wait for Enter pressed.

If all steps doing well (keep your eye what color lights have after it's back on), that's all for basic support.

About Aurora, yes, but some effort needed - they use slightly weird structures for light modules.

Billybobertjoe, 2021-11-21:
3 device(s) detected.
VID: 0x461, PID: 0x4ec0, API v7
Let's try to set some colors...
SetColor result: 1, error 0
SetColor 2 result: 1, error 0
Ok, let's try to set lights off...

Now set for half brightness...

... then return back to full...
Done!

Billybobertjoe, 2021-11-21:
it all worked fine

Billybobertjoe, 2021-11-21:
at the end, the color changed to blue

[41 more replies]

closed 2022-05-13 by T-Troll