T-Troll/alienfx-tools #75: Fan Boost Value Changed

Hellohi3654, 2021-11-11:
Hey just wondering if you would be able to set the max fan boost to 255

I'm not sure what has changed but it seems now that boost 255 is now the max RPM for my fans
Boost 160 is about 4800RPM while boost 255 is around 5700RPM

Thanks

Btw if you have forgotten which laptop I have it's the Dell G5 15 (AMD)
Thanks again

T-Troll, 2021-11-11:
Even better. Latest release have special `alienfan-overboost` tool.
Run it, and it set all boosts to maximal possible level (check readme_alienfan for details).

T-Troll, 2021-11-11:
PS: and no, boost can't be more, then 255 (it's a byte value inside BIOS).

Hellohi3654, 2021-11-11:
ok thanks

Hellohi3654, 2021-11-11:
> Even better. Latest release have special `alienfan-overboost` tool. Run it, and it set all boosts to maximal possible level (check readme_alienfan for details).

Awesome very helpful

closed 2021-11-11 by Hellohi3654