T-Troll/alienfx-tools #77: UI Bugging

bloworlf, 2021-11-15:
I just installed it and opened it. When I click on the tabs, nothing happens. But when I hover the pointer over elements, they start popping up.

My config:

- Alienware m15r5 Ryzen Edition
- Windows 11 x64

[image]

T-Troll, 2021-11-15:
Thank you for report.

Seems like tab switch issue, let me examine it. Hope it not W11-specific, 'cause i use W10.

bloworlf, 2021-11-15:
Okay, thank you. After starting alienfx-probe, the gui works fine (navigating thru tabs, playing with rgb) but when trying to setup the fan control, it said something like it couldn't install watherver it was trying to and BSODed me with a "KMODE EXCEPTION NOT HANDLED". Didn't have time to take a screenshot.

T-Troll, 2021-11-15:
I catch and fix this visual bug at latest release [5.2.5.1](https://github.com/T-Troll/alienfx-tools/releases/tag/5.2.5.1).
Please check and close this issue if it works for you.

Wow! So my ACPI driver crash in Windows 11!
Please do no try to use it in this case. I added a warning about it into documentation as well.
Unfortunately, i will not plan to update my system to WIndows 11 until it will be stable enough, so don't expect rapid fix for this bug. But i open another issue about it to remind me, thank you for pointing to this.

bloworlf, 2021-11-15:
Yes, thank you. I just tested it (I uninstalled the previous version and deleted its folders from registry) and the UI is fixed.

You're doing a great job here. A better job than that manufacturer I bought that PC from. Keep up the good work.

closed 2021-11-15 by bloworlf

T-Troll, 2021-11-23:
Thank you for your feedback.
In fact, it's my pet project, so i don't need any donate or anything, but i glad some people like it.
By the way, can you check #79 and comment about your Windows 11 settings i mentioned there?