T-Troll/alienfx-tools #79: ACPI fan control diver is incompatible with Windows 11

T-Troll, 2021-11-15:
As reported by testers, ACPI control driver used for fan/power control is incompatible with Windows 11.
It provides BSOD with "Unhandled kernel exception" then used.

I have no W11 system to check it right now, so this issue postponed until i update one of my systems.

T-Troll, 2021-11-23:
@bloworlf, thank you for feedback.

Can you please check one Windows 11 setting for me?

Go to "Settings/Privacy & Security/Windows Security/Device Security" and check "Memory Integrity".

If it's on, it can be the source of the driver issue (and also provides significant performance decrease), so can you please turn it off and try again (be careful, system can be crashed again!)?

bloworlf, 2021-11-23:
Okay, I just updated windows, I'm on build 22000.348 now. I opened Alienfx Control, went to settings to enable fan control. And it works now.

[image]

[image]

You're doing great work.

T-Troll, 2021-11-23:
Funny. I Don't change anything!

Thank you for the information, but i still recommend disabling the setting mentioned, as it well-known to cause incompatibility and significant performance drop.

PS: Also, i recommend disabling "Auto refresh lights" into app settings - this option is for working with AWCC active at the same time (but you set it disabled), and provide a massive light change overhead instead. Maybe i need to disable it automatically if "Disable AWCC" active.

T-Troll, 2021-11-23:
This is how temperature readings looks for my gear:
[image]

bloworlf, 2021-11-23:
Maybe I will fresh install the app (delete all the files and registries) and check again. I just installed the new version over the old one.
I don't know how many temperature sensors my laptop has but I have optimus. I can switch the NVIDIA GPU ON an OFF.

--my system is in French--
[image]

T-Troll, 2021-11-23:
I understood French un peu.

No, it's not about it (but +2C on GPU can). You only have AWCC sensors, but no other one detected in BIOS.
Can i check your ACPI BIOS dump (use RWEverything tool to dump it)?

bloworlf, 2021-11-24:
The ACPI Tables. Don't know if it's all the information you needed.

bloworlf, 2021-11-24:
Here's what I got from RWEverything's ACPI tab

[ACPI Tables](https://drive.google.com/file/d/1gjQOMaG6VYzdD-Ip5dp960tzHRqrcdYu/view?usp=sharing)

T-Troll, 2021-11-24:
Thanks, let me check. I remove the link (and recommend you to remove the file as well), 'cause it has some sensitive data inside (not personal, but about the system).

T-Troll, 2021-11-24:
Uff... I check. What a fine mess inside, looks like they are in a hurry to finish it.
A lot of debug info still in release BIOS...

The layout is similar not to mXX, but to Dell G-series, by the way.

First, you have to mess into CPU power control. BIOS reports 3 modes to my app, but.... only one can be set, and weird way. So it will not work, in fact.
Second, GPU control present, but do nothing. Good news, they have direct GPU power control - i think i'll add support for this.

And yes, they just "forget" to set any thermal zone (so both my tool and Windows can't see anything). You can check PerfMon "Esif" data - maybe Windows guys can catch some other source.
You have some suspicious variables in EC (in fact, all 2 temperatures i got is from here in Alienware function), but i can't check it myself - can only share where to look for you.

bloworlf, 2021-11-25:
Okay, thank you for the information. If you need anything to debug further, just tell me. I'll send perfmon data when I get back to my PC.

bloworlf, 2021-11-25:
Weird thing, AWCC doesn't seem to offer "breath, pulse, spectrum" effect. And I couldn't get it to work on you app either. I only used Haptics and Ambiant. I just turned the PC on and all of my RGB is getting colorful 😂

Where do I get PerfMon "ESIF" data?

T-Troll, 2021-11-25:
This effect is only available for v4 (tron&power) devices. Your keyboard have v5 (and global effects) instead.
I'm trying to give a more flexibility how to set your lights, i'm using different profiles for different situations.

Start Performance Monitor, select "Performance Monitor" into it, press "+" (add) button. Wait a moment, and you will have a HUGE list of performance indicators available. Select "ESIF Device information", and look into this group, you should have "Temperatures" there. Select it, I'm interested what do you have into "Instances" list.

bloworlf, 2021-11-25:
I cannot find those anywhere. Only thermal I found was empty:
[image]

If you want, I can grant you remote access via teamviewer or anydesk to check.

T-Troll, 2021-11-25:
The place is right (in fact, it's there i got some temperature sensors for my app), just look for "ESIF Device information" (another source) into the same list. It's a pity if you don't have it.

> If you want, I can grant you remote access via teamviewer or anydesk to check.

Thanks, but i don't need. Seems like you system really don't have more sensors into ACPI. As i say, it can be the issue of the BIOS, so just wait for update (or downgrade it for previous one).

T-Troll, 2021-12-11:
Fixed in [5.3.6](https://github.com/T-Troll/alienfx-tools/releases/tag/5.3.6)

closed 2021-12-11 by T-Troll