T-Troll/alienfx-tools #83: Unable to switch on the lights on m17xR4

ramana16, 2021-11-23:
**Describe the bug**
Unable to set the lights using both the GUI or the CLI tool
Even when using the command center it says not detected, really very strange.
can you please help me to get the lights working?

**To Reproduce**
When I execute the CLI I get the message as - 
"
C:\Program Files\AlienFX Tools>alienfx-cli.exe lightson
alienfx-cli v5.3.0
No low-level devices found, trying high-level...
No high-level devices found!
[image]

Both low-levl and high-level devices not found, exiting!
"

**Expected behavior**
Was hoping to be able to change the color

[AcpiTbls.zip](https://github.com/T-Troll/alienfx-tools/files/7587844/AcpiTbls.zip)
[image]
[image]
[image]

T-Troll, 2021-11-23:
> No low-level devices found, trying high-level...

This means you hardware not detected - it can be both doesn't work or my detection fault.

Do you have old 17x or new m17? Anyway, check Device Manager, under "Human Interface Devices". You should have "Vendor-defined device" with VID_187C. If you don't have - it means light system have a hardware issue. If you have one - we need to check why it doesn't detect.

ramana16, 2021-11-24:
I dont seem to be able to any device like that under HID.
Here are the screenshots from windows 7 and windows 10, can it be a firmware issue which is preventing the device from being recognised?
In Windows 10, I see there are some un-recognised devices. How do I identify it?
[image]
[image]

T-Troll, 2021-11-24:
First, my apps is not compatible with Windows 7 for some reasons.

Second... Yes, device manager in W10 looks weird, system have some issues. Most probably chipset drivers not installed correctly.

You can check unknown devices 2 ways: First, select it and switch device manager to "View->Device by connection". This give you a hint what kind of device is it - system bus, USB, etc. Second, check Hardware ID's at unknown device properties. If it's VID_8080 - it's something from Intel, and it can have some child devices not revealed until parent driver works. You can check other VIDs online as well to understand. We are looking for VID_187C - Alienware.

closed 2021-12-06 by T-Troll