T-Troll/alienfx-tools #89: Bug Fan Controller

Djioup, 2021-12-05:
**Describe the bug**
I have the following problem: when I restart my computer after using alienfx, when I restart alienfx, the fan controller does not work, and in the worst case, I have a blue screen and the pc restarts

**To Reproduce**
1. Start Alienfx with fan controller 
2. Restart windows
3. Restart Alienfx with fan controller
4. In the worst case, when I click OK to close the error window, blue screen and reboot.

**Expected behavior**
a window with a message that says the fan controller is not working

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Firefox
 - Version alienfx : Last one

**Additional context**
Alienware m15 R5

T-Troll, 2021-12-06:
Can you check GUI version? ("About" from the top menu)?
Is it pre-5.3.4 or 5.3.4?

I'm asking about it because i change some driver parameters in 5.3.4 to avoid this situation, but it can provoke this.

Can i also have details about your gear configuration:
- BIOS: Do you use UEFI boot?
- BIOS: Is "Secure boot" enabled?
- Windows: What release do you have (Settings->System->About)?
- Do you have some external devices, especially from MSI or Asus?

PS: Fan disabling means my tool can't start driver correctly. BSOD means the same, but it happened at driver loading.
PPS: As a temporary workaround, you can switch your system into "Test mode" - read instructions [here](https://github.com/T-Troll/alienfx-tools/tree/master/alienfan-tools#requrements).

Djioup, 2021-12-06:
_Can you check GUI version?_ : **5.3.4.**
_Do you use UEFI boot?_ : **Yes**
_BIOS: Is "Secure boot" enabled?_ : **Yes** 
_Windows: What release do you have ?_ : **21H2**
_Do you have some external devices, especially from MSI or Asus?_ **No**

I hope I gave you the information you were looking for.
Thank you very much for your work!

Djioup, 2021-12-06:
Ok well the problem was the secure boot. My problem is solved since I disabled it. Sorry for the waste of time. Thanks again

T-Troll, 2021-12-06:
It's still in issue, in fact.
F.e. Windows 11 didn't work without it.
I use other guy's hack code for loading, drop this to him as well.
Or maybe MS changes the policy about driver signing (that's why i forced to use hack).

Djioup, 2021-12-06:
Correction: I just got the bug again :'(

T-Troll, 2021-12-06:
Hmm... Please check antivirus log. Is any complaint about app start?
What do you have in system log (Event Viewer->Windows logs->Application/System, look for "Error" entry)?
Also, let me check something, it's possible one of your system's drivers interfere.

PS: I use the same Windows version, secure boot enabled, only have it one time at initial start.

T-Troll, 2021-12-07:
We are discussed this issue with other guys at Discord support channel.
Seems like this issue can be a result of antivirus action.
Can you please try to add `kdl.dll`, `HwAcc.sys` and `drv64.dll` into antivirus exceptions and see what happens?

Djioup, 2021-12-08:
I just tried, adding the 3 files to the antivirus exceptions. But the problem persists :)

Djioup, 2021-12-08:
I'm sending you the information you asked for tonight

T-Troll, 2021-12-08:
Can you also provide me system minidump, please?
Also try the latest version, i change the driver a bit, it could help.

T-Troll, 2021-12-08:
PS: The strangest thing, BSOD happened after you press "OK". But driver already fail to load, so i don't do anything, in fact.

Djioup, 2021-12-08:
**Windows log error** : Nom de l’application défaillante AWCC.exe, version : 5.4.7.0, horodatage : 0x6166ddbf
Nom du module défaillant : KERNELBASE.dll, version : 10.0.19041.1348, horodatage : 0x76fcd692
Code d’exception : 0xc000027b
Décalage d’erreur : 0x000000000010b302
ID du processus défaillant : 0x345c
Heure de début de l’application défaillante : 0x01d7ec7898c33ce8
Chemin d’accès de l’application défaillante : C:\Program Files\WindowsApps\DellInc.AlienwareCommandCenter_5.4.7.0_x64__htrsf667h5kn2\AWCC.exe
Chemin d’accès du module défaillant: C:\Windows\System32\KERNELBASE.dll
ID de rapport : fe21f636-0063-47ba-80bf-3ce78659101a
Nom complet du package défaillant : DellInc.AlienwareCommandCenter_5.4.7.0_x64__htrsf667h5kn2
ID de l’application relative au package défaillant : App

HwAcc.sys : Size 9.67 Ko/ 12 Ko (on disk); version 1.3.1.0; time : 01.12.2021 (21h58)

System minidump : automatic memory dump ( don't know if this is what you asked me ^^)

_BSOD happened after you press "OK"_ : I haven't had a BSOD since I disabled secure boot

T-Troll, 2021-12-08:
Yes, AWCC will crash if run then AWCCService disabled. In general, it's not a bright idea to run it with my GUI running - then have a challenge for the same resources, so the result is a bit unpredictable. By the way, you can enable AWCC service back, but also enable "Disable AWCC" into GUI settings - in this case GUI will stop it then start and start back then exit. So you can use AWCC as well (but not the same time).

I see you have old driver, the latest one is 1.4.0. Do not update right now, i have a serious bug in the latest version, plan to fix it today, so wait for 5.3.5.2.

I need a minidump from the latest BSOD (if you still keep it). It's located into `C:/Users/<you profile name>/AppData/Local/CrashDumps`. Latest system crash also into `C:/Windows/minidump.dmp`.
Check the file timestamp to find the right one. I require it to check which driver provides the crash and it's conditions.

Djioup, 2021-12-08:
[AWCC.exe.12508.dmp.zip](https://github.com/T-Troll/alienfx-tools/files/7679236/AWCC.exe.12508.dmp.zip)
[KNDBWM.exe.9172.dmp.zip](https://github.com/T-Troll/alienfx-tools/files/7679239/KNDBWM.exe.9172.dmp.zip)

_Latest system crash also into C:/Windows/minidump.dmp_ : **empty**

T-Troll, 2021-12-08:
Unfortunately, it's not what i looking for. AWCC dump is not needed, i know about it.
The second one... You have issue with Killer Networking adapter, so i recommend you to update it's drivers. ^_^

Djioup, 2021-12-08:
Hmm ... I can't reproduce the problem since I reactivated AWCC and the "disabled AWCC" of the alienGui. It already did it once so I'm still on my guard, but it's a good sign

T-Troll, 2021-12-08:
Great. So let's keep our eye. I'll also ask you to check latest driver later on, please!

Djioup, 2021-12-08:
Ok, count on me ! Thanks a lot for your help and for your great work

T-Troll, 2021-12-08:
And disable secure boot it's not a good idea (especially for W11), so we need to check.

I use latest W10 (21H2) with secure boot enabled and have no issues. But i don't use any AV (my router act as a primary AV, and i execute scan-before-run in case i download something from unconfirmed source).

T-Troll, 2021-12-08:
> Ok, count on me ! Thanks a lot for your help and for your great work

NP, i have a lot of fun doing it.

Djioup, 2021-12-08:
I enabled secure boot, did 2 reboots in a row with AlienFan enabled (which is normally enough to cause the error), and I have no problem. It looks good.

T-Troll, 2021-12-08:
Ok, i release hotfix, so you can update and check about new driver.

Djioup, 2021-12-08:
I had a BSOD again xD
I haven't downloaded the hotfix yet

[120821-8000-01.zip](https://github.com/T-Troll/alienfx-tools/files/7679447/120821-8000-01.zip)

T-Troll, 2021-12-08:
Great, let me analyze it....

Djioup, 2021-12-08:
I downloaded the hotfix, did a series of reboots and got a blue screen again 

[120821-8906-01.zip](https://github.com/T-Troll/alienfx-tools/files/7680920/120821-8906-01.zip)

T-Troll, 2021-12-08:
I got it already:

```
FAILURE_BUCKET_ID:  0x1E_c0000005_R_nt!IopUnloadDriver
```

Uh-oh. Crazy. I'm going to check WTF.

T-Troll, 2021-12-08:
Try this driver - [HwAcc.zip](https://github.com/T-Troll/alienfx-tools/files/7681021/HwAcc.zip)

Djioup, 2021-12-08:
I tried the new HwAcc, but I got a BSOD
[120921-8453-01.zip](https://github.com/T-Troll/alienfx-tools/files/7681059/120921-8453-01.zip)

I notice that the error occurs only when I start the computer after a total shutdown, but not when I restart the computer

Djioup, 2021-12-08:
I notice something else too: since I have the hotfix2, alienGUI doesn't launch automatically anymore, I have to launch it manually

T-Troll, 2021-12-08:
> I tried the new HwAcc, but I got a BSOD

Ok, thanks, let me check.

> I notice something else too: since I have the hotfix2, alienGUI doesn't launch automatically anymore, I have to launch it manually

Just go to the settings, then disable "Start with windows" and enable it back. Or check Task Scheduler - something to remove/disable the task there.

Djioup, 2021-12-08:
Another remark : I have BSOD with or without the secure boot

T-Troll, 2021-12-09:
Ok, let's made another try.
Here is both driver and GUI app - [bsod_test.zip](https://github.com/T-Troll/alienfx-tools/files/7681257/bsod_test.zip)

I completely disable any unloading, both in driver and SDK. I check this both at restart and after shutdown, don't have BSOD...

Djioup, 2021-12-09:
ok, I'll test tomorrow, I have to go to sleep :(

T-Troll, 2021-12-09:
Also try to disable DriverVerifier, as described here - https://www.drivereasy.com/knowledge/fixed-ntkrnlmp-exe-blue-screen-error/

Djioup, 2021-12-09:
Hi!
So I tried the new driver and the new GUI. Here is the result (no blue screen for the moment)
[image]

Djioup, 2021-12-09:
Excellent! This error window used to mean a BSOD 100% of the time when I was about to click OK or close. Now I can close it without a BSOD. This is a huge improvement
[image]
.

T-Troll, 2021-12-09:
So i was right - you have BSOD every time i just try to remove driver service (!). Incredible!
Now even more interesting. Code 3 means i can access the device, but probe can't detect its type.
Let me share the version with more diagnostic for you.

T-Troll, 2021-12-09:
Ohh... I have an idea.
Can you please:
- Download latest AlienFan (not AlienFx) release from [here](https://github.com/T-Troll/alienfan-tools/releases)
- Make a folder with simple name at the root of "C" drive
- Unpack it into this folder
- Run `alienfan-gui`

Keep `alienfx-gui` stopped before this.
Just to check the idea app can't locate driver file.

Djioup, 2021-12-09:
here are the results
[image]
[image]

T-Troll, 2021-12-09:
Thanks. AWCC errors is not interesting for this, by the way, leave it alone.
Let me provide you the version with extended debug, i still doesn't understand why it happened.

T-Troll, 2021-12-09:
Ok. Here is the test tool - [alienfan-test.zip](https://github.com/T-Troll/alienfx-tools/files/7689014/alienfan-test.zip)

Can you please:
1. Unpack it to the same folder as alienfan (or alienfx) - it needs driver and dlls.
2. Open command line (CMD) FROM ADMINISTRATOR.
3. Run.
4. Copy output here.

Djioup, 2021-12-09:
Is this it ?

Opening ACPI...
Device not activated, trying to load driver...
Loaing driver from C:\ALIEN\alienfan-tools(1)\HwAcc.sys... KDL loades, trying... Driver loaded, trying to open it... Loading complete ,failed.
Done
Goodbye!

T-Troll, 2021-12-09:
Wow! Crazy. Driver was loaded successfully, but you system block driver access!
How did it possible... let me check.

T-Troll, 2021-12-09:
Can you please try [this one](https://github.com/T-Troll/alienfx-tools/files/7689123/alienfan-test.zip)?

Procedure is the same, but it has more extended diagnostic (and maybe i fix some).

Djioup, 2021-12-10:
Opening ACPI...
Driver access failed
Device not activated, trying to load driver...
Loaing driver from C:\ALIEN\alienfan-tools(1)\HwAcc.sys... KDL loades, trying... Done
Goodbye!

T-Troll, 2021-12-10:
Please reboot. 2nd driver loading failed.

Djioup, 2021-12-10:
I just rebooted, same result

Opening ACPI...
Driver access failed
Device not activated, trying to load driver...
Loaing driver from C:\ALIEN\alienfan-tools(1)\HwAcc.sys... KDL loades, trying... Done
Goodbye!

T-Troll, 2021-12-10:
Are you run from Administrator console? Looks like not.

T-Troll, 2021-12-10:
Anyway, let's try this one - [alienfan-test.zip](https://github.com/T-Troll/alienfx-tools/files/7689324/alienfan-test.zip)

Djioup, 2021-12-10:
last result with the last test in admin mode ^^

Opening ACPI...
IOControl OK for 1
Probing...
Probing devices... Device type 1 found.
Goodbye!

T-Troll, 2021-12-10:
Ooook! We fix it. Now wait for next release.

Djioup, 2021-12-10:
Great !! Thank you so much for your help. I'm looking forward to the release !

T-Troll, 2021-12-10:
Try this GUI - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/7691617/alienfx-gui.zip)

Please, let me know - if it's OK, i release tomorrow.

Djioup, 2021-12-10:
For the moment, this version works when the alienfan gui does not. I think that means it's good.

I'll do some more testing tonight

Djioup, 2021-12-10:
By the way, do you want to develop what you said here:  _BTW, this situation is not good, you will lose about 15k of kernel RAM every time it happened._
I'm not sure I understood everything.Are there any long term consequences?

T-Troll, 2021-12-10:
No. Nothing except profiles in registry stay after reboot.
But if my app doesn't detect the driver, it tries to load it again in kernel. Nothing serious, in fact, if it is not happened again and again.

I'm specially design apps to work (as they can) in any situation. 
No Internet connection? Ok, we don't check for update.
No admin? Ok, we disable fans (and some other).

Djioup, 2021-12-11:
After a number of reboots and shutdowns, I haven't noticed any problems yet.

T-Troll, 2021-12-11:
Ok, great!

closed 2021-12-11 by T-Troll