T-Troll/alienfx-tools #100: Does it support R13?

lixiang1994, 2021-12-23:
I replaced 3 Noctua NF12x25 fans, But the speed seems uncontrollable.

T-Troll, 2021-12-24:
Did you check readme?

I use the same method (but with more flexibility) like AWCC do. So fan control is not direct, but indirect - BIOS still control fans, but both AWCC and my tool can define offset to BIOS value.
In general, it's more safe way, it can also spin fans more, then BIOS maximal value.

If you want to stop fans under some temperature, first switch cooling method to "Passive" into Windows Power Scheme - in this case, fans will stop under 40-50C (depend on BIOS setting) and 0 boosts.

PS: Maybe manual fan mode fails for your gear. Please check data around fan curve - is boost value working correctly (if you set above zero for this temp - it should show both dot and value above zero)?

T-Troll, 2022-01-04:
Don't sure about R13, but R10-R12 now 100% supported. Can you check please and close this issue?

lixiang1994, 2022-01-04:
OK, Thank you

closed 2022-01-04 by lixiang1994