T-Troll/alienfx-tools #102: AlienFan Control - Expose more gpu/cpu temp sensor for control by fan curve

Tyfui90, 2021-12-24:
**Is your feature request related to a problem? Please describe.**
I have been looking for a way to control the front fans for my R12 via the gpu temps. The gpu temp exposed via the fan control is the gpu internal thermister (fig1 34C) . I was hoping to be able to use the memory junction temp (fig1 92C) or whatever hwinfo exposes as the gpu temperature (fig1 53C) 

[image]
fig1 - HWiNFO64 GPU Temperature & Memory Junction Temperature, AlienFan Control

**Describe the solution you'd like**
If possible, more Temperature Sensors exposed to control the fan curves. Specifically the gpu memory junction and the gpu temperature.

**Describe alternatives you've considered**
I can currently use the gpu temp exposed via the fan control to control the fans via the gpu temp. But this temperature is much different that the temps reported by HWInfo64, and so doesnt provide the direct fan to temperature connection that I was initially hoping

**Additional context**
PS, I love this utility. the amount of tuning you can do when setting a fan curve is great; compared to AWCC this is already a much better tool.

Alienware R12 - Z590 chipset motherboard
RTX 3080

Let me know if there is additional information I can provide, or specific investigation I can do that may help this request

T-Troll, 2021-12-25:
Thank you.

Ok, let's see...

You have the same sensor list as R10. Please keep in mind, sensor names are fake now (i take it from notebook mappings, because there is no way to get it from BIOS at your gear) - i'll fix this later. So 2nd one can be not GPU temp, in fact (let me check). Also, power modes are not finished yet for your gear (it will require serious changes in detection process).
According to BIOS, both R10 and R12 have 3 unused thermistor connectors (-1 temperatures), exposed by the BIOS, but (instead of notebooks and R7) have no thermal zones defined in BIOS.

I utilize the same BIOS functions like AWCC using, the bright side BIOS update (and even new model) didn't break it (or AWCC will be broken as well). But dark side is i limited to the same information most of the time.

What can we do?

First, i don't want to go deep into EC sensors - it's highly model-specific, so it will require tuning between models and, sometimes, BIOS updates.

But there is another way, if your system support it.
Please, open PerfMon (Performance Monitor), select "Performance monitor" at left column, then press green "+" button.
Wait for a couple of seconds until list is filled, then try to locate "ESIF device information" entry.
Open it, a look for "Temperature" entry inside.
In case you found it (you can also add all temperatures into PerfMon graph here and check) - i can read and use this values as well. But i will need your help for name mapping - which sensor is from which point?
In case there is no such entry - it will require deep ACPI BIOS study, and it not guarantees we are will be successful.

PS: One trick - you can overboost your fans (set more RPM than 100% BIOS). Just stop GUI, open Administrator command line, navigate to alienfx-tools folder and run `alienfan-overboost` utility. It should run for some minutes, then you can see maximal available boost and RPM values for your fans into its log. It will be stored and used by GUI later.

PPS: Is lights control available for your gear? If yes, which API version it used?

Tyfui90, 2021-12-28:
Sorry for the delay, the Christmas period slowed me down!

Looking into Perfmon I didnt find anything under "ESIF device information" unfortunately
[image]

So it sounds like the ACPI BIOS study may be required. Though im not entirely sure what that would entail.
It looks like you're able to report the Dell EC temps, but these dont line up with the temps that are actaully experienced by the cpu & gpu, so not sure what they're actually measuring.
[image]

As for the lights control API version, I think its v4
[image]
Just in case this is useful, heres the version for the AWCC.service thats running on my system
[image]
and the AWCC Driver version from Device Manager
[image]

When I ran the AlienFX Control detected the R12 as a the m15 notebook. and selecting that I was able to control the Alienhead Power button light, and the front ring light. So i think AlienFX Control found all it can and is working as expected.
[image]
And running the alienfx-probe it detected the front ring and power button as well

ps, looks like the power mode detection is now working as this dropdown has values in it now.
[image]

T-Troll, 2021-12-28:
Merry Christmas and happy New Year! (and yes, it's busy time for all :)

Thank you for wide analysis, let me comment and discuss what we can do.

First, about lights. This chip with PID_0550 (APIv4 by my SDK) is used widely by Dell for all latest products - all m- and x- series notebooks, and even G-series. It's quite nice, with up to 9 hardware effects per light and hardware dim/off.
So that's why i detect it as a notebook - they use other chip for Aurora before. It's mentioned into [5.4.1 release notes](https://github.com/T-Troll/alienfx-tools/releases/tag/5.4.1).

`alienfx-probe`, as you can see in log, can use Dell's API device and lights name, that's why it detected like this. But be careful, the order of lights can be different between Dell and hardware!
`alienfx-gui` uses my own device database, and it's not so wide. By the way, i will appreciate if you share your mapping names for me to add into this database (use "Save lights" button at Devices tab).

Now about fans/power modes (it's a bit strange, you have 3 modes - R10 have 4. Is it my bug?).

It's really not good, Dell stopped to expose sensors both into Thermal Zone and ESIF. By the way, m15R6 have the same issue.
EC method is way more complex, because it's model-specific - so it requires complex analysis and testing for each new hardware. I only use it if there are no other choice (like for old 15Rx and 13Rx).
By the way, it means the rest of sensors does not participate into BIOS fan speed control.

So... Can you share your ACPI BIOS dump for me - let's try to find BIOS-based functions for sensor access?
You can do it using [RWEverything ](http://rweverything.com/) tool.

PS: I can share for you which methods and functions from ACPI i use now, if you're interested.

Tyfui90, 2021-12-28:
Merry Christmas and happy New Year as well :)

>PS: I can share for you which methods and functions from ACPI i use now, if you're interested.

Yeah, If you could let me know that would be great. Im a little intimidated looking at the RWEverything options. unsure exactly how to do the ACPI bios dump.

>By the way, i will appreciate if you share your mapping names for me to add into this database (use "Save lights" button at Devices tab).

Heres the csv mapping for my R12:
```
'0','6268','1360','Alienware Aurora R12'
'1','0','0','Front Ring'
'1','1','0','Alienhead/Power Button'
```
[AuroraR12AlienFXLightMapping.csv](https://github.com/T-Troll/alienfx-tools/files/7782340/AuroraR12AlienFXLightMapping.csv)

>I'm also recommend stopping AWCCService

I have set that service to no longer start via msconfig. I assume i dont need it anymore as im not using AWCC anymore.

T-Troll, 2021-12-28:
> Heres the csv mapping for my R12:

Thank you, i'll add it into next release. By the way, maybe it's not so bright idea to use power button control here - it's for notebooks, they have 7 modes (ac, battery, charging, etc.), but you desktop not. So it can work incorrectly.

> Yeah, If you could let me know that would be great.

Ok, here are some hints.
First, open "ACPI" into RWEverything and press "Save all" button. You will have .rw file with full ACPI dump. In fact, it's a text file with disassembled ACPI functions code - so you can open it using any text viewer/editor.

What to look for:
- First, custom Alienware device. Search for "Device(AMW". All functions AWCC using are inside.
- Most interesting one is function named WMAX (search for "Method(WMAX") - it's main control function.
- I call it using different parameters - for example (0x14, 0x2) reveal all fan, sensor and power modes mapping. And (0x14,0x4) returns sensor temperature. You can check full list into `alienfan-controls.h` file int this repo.
- Also look for Thermal Zone blocks - it should have "_TZ" name into device tree.
- Another big block is `\\_SB.PCI0.LPCB.EC0` (or EC1) - it's all EC-controlled data mappings. Other systems have a lot of sensor devices inside as well. Here you can check about other sensor access methods as well as direct fan control (if BIOS have it), and EC variables as well. As a hint, look at sensors names into `alienfan-controls.h` - maybe you found them for your system as well.
- And last, but not least, is `\\_SB.PCI0.PEG0.PEGP.GPS` method. It's all about Nvidia GPU control. It can vary for your system, this one from any Nvidia-based notebook.

So... Good luck to study it!

closed 2022-03-21 by T-Troll