T-Troll/alienfx-tools #105: Add support for AW 15R2 [Comes under Echo series 13R2, 17R3]

vasudev-gm, 2021-12-29:
**Describe the bug**
Didn't detect it as supported device eventhough it uses same lighting as 13R2 and 17R3. Also had to copy LightFX.dll to root folder for alienfx-prober 26[number of zones] to detect the lighting zones properly.

**To Reproduce**
Steps to reproduce the behavior:
1. Unusual delay while detecting the zones via prober in cmdline
2. Sometimes the color changes correctly and sometimes it doesn't
3.Attached a screenie with device pid, api version and vid.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 21H2
 - Browser: Edge 96.x, Firefox 95.x
 

**Additional context**
None

T-Troll, 2021-12-29:
Ok, let's check...

1. Probe does not require LightFX.dll. But it can scan names from it if original Dell's one available. So it's useless to copy my emulator before names defined (it will have 0 lights).
2. Yes, some pins not connected to the lights. In probe, just press "Enter" to skip this, you don't need to have all 28. Also, you can press CTRL+C to stop probe if all lights defined.
3. Indicator lights (hdd, Wi-Fi, capslock) can be only visible if active - so keep your Wi-Fi connected and capslock on to detect it.
4. Yes, i see you have the same v2 protocol like mine old 13R2. But light mappings can vary from model to model.
5. After setting light names, switch to "Color" tab and define default color for it - otherwise it will stay at initial color.
6. What's wrong with colors? Can you, please, be a bit more specific?

PS: In fact, probe is for bulk set from command line, in case you don't want to use GUI. I recommend use `alienfx-gui` to define lights instead.
PPS: Most probably, fan control will not work for your gear, even if you can enable it. You can help me to implement it later on, but it's tricky for old systems.

vasudev-gm, 2022-01-03:
> Ok, let's check...
>
>     1. Probe does not require LightFX.dll. But it can scan names from it if original Dell's one available. So it's useless to copy my emulator before names defined (it will have [...]

Thanks.
I think I have found the issue with color not changing when I have assigned multiple color morphs and timings along with events due to older AlienFX SDK which does not allow w/o crashing the app. I can set all zones to single color w/o any issue.
So whenever I try to change the status LED and power LED the alienfx simply doesn't apply the changes. I think AFX SDK 2.7 redist dll is too old. I tried replacing it with AWCC 4.5.19 but it didn't work. Too bad Dell has locked out the SDK. 
Also I am able to adjust the fans but its buggy due to EC and functions similarly to HWINFO fan control.
Even seeing better battery life with light enabled w/o AWCC.
I did complete uninstall and reinstall and completely replaced it with alienfx-tools since its lighter and efficient. Now no more waiting or 10 mins for AWCC splash screen.

T-Troll, 2022-01-03:
Hmm.... my tools doesn't need any DSK DLL, it works at direct hardware level. However, i publish my own LightFX.DLL EMULATOR - so you can eliminate AWCC completely, if you wish. But older version doesn't install the service, so just quit it is enough.

Yes, you need to choose - hardware effects OR software effects (monitoring, ambient, haptics). It can't work at the same time.

Fan control didn't work because your gear have other EC mapping - i only support 13R2 between old models now. But my app detected you have an old Alienware system, so it's trying to enable it anyway. It just doesn't work. As i said, i can add support for your system as well, but it will require some effort from you as well for helping me to define mappings and other ACPI data.

Uff... There are many questions from the beginner, so i plan to open another WIKI page about "how to start".

Anyway, i glad you configure things out, feel free to ask questions if you have.

vasudev-gm, 2022-01-03:
I don't think hardware effects, ambients and haptics are supported on Echo series. Software effects is enabled. Due to old AFX SDK its quite buggy.
Thanks.

T-Troll, 2022-01-03:
it's supported into my tools.
You v2 chip have 4-bit per color (BTW you can correct white point for more natural white), hardware Pulse, Morph, and it's combinations.
Software effects are supported at any API version, but consume some CPU (not so much).

vasudev-gm, 2022-01-03:
Hmm but that color correction still has slight pink/blue hues when applying to all zones so I went with Light Blue almost similar to MBP or Surface keyboard light.
Couldn't get SW effects to apply correctly for power or status zones. Whenever I apply them few zones get applied to pink or different color not chosen and starts to flicker. Without Dell SDK dll afx tools never applies lights properly at all and defaults to OFF state.
One thing I forgot to mention is, I am able to get the fan to work correctly based on hotspot/sensors from EC esp. DRAM, PCH, GPU, CPU etc and was also able to adjust fan graph. Only issue is if I want to revert anything I need to Select Power Limit to Level 1 otherwise the fans keep spinning until PC is restarted or shut down.

T-Troll, 2022-01-03:
White point: Yes. it's not always possible to set clear white - as i said, your system only have 4-bit (16 grades) per color, so mixing is hard. I can do it correctly at mine 13R2 by decreasing red. And yes, keyboard zones mix each other, so it's a bit tricky.

SW effects: looks strange for me. First, i don't use Dell SDK, so it completely doesn't matter did it present or not. Power button can be tricky (did you mark it as a power button?), but indicator lights should be the same as others (the only difference it doesn't disabled then other lights off). Power off lights should be set upon exit/sleep/shutdown/hibernate to default settings in Color tab. But sometimes (for long power off) the system reset the chip to factory defaults. There is another way to set default lights, but it took 3-5 seconds and provide the same result.

Fans: Well... I wonder it works - it should not. Can i see your ACPI BIOS dump?
Power modes is tricky. First, i store the power mode you have at start of the application, and then set it back at exit. Seems like it doesn't work for your gear. As a result, it stays at Manual mode all the time. Die to the fact your system have direct fan control (RPMs instead of boost at newer systems), fans stuck at last RPMs defined in app. Until reboot or system will be really hot (emergency mode will be initiated in this case). We can try to disable manual mode if it is possible into your BIOS.

T-Troll, 2022-01-04:
Ok... Check the [latest release](https://github.com/T-Troll/alienfx-tools/releases/tag/5.4.4). What important for you:
- Now you can switch fan mode using ALT+CTRL+0..1 (0 manual, 1 auto) then app running.
- Application forced to switch to Auto mode at quit/shutdown for your gear.

vasudev-gm, 2022-01-07:
Have installed the latest release; will test the hotkey for fan control.
The chip is faulty or broken with 3rd party. I had different colors being applied by games and they won't switch/restore to custom colors. I'm not fussed with custom colors working and all set all LEDs to white but some zones apply Pink Left Keyboard, Alienhead logo is orange.

T-Troll, 2022-01-08:
Ok, let's check it.

First, there is no difference between AWCC and my tools - both share the same hardware interface, so results should be the same.
But my tools have wider possibilities, so it can interfere sometimes.
For your case, please check 3 things:
- Is light defined in Device and set to white at Color tab? (just for sure)
- Disable software effects (set it to "off" into top-left dropdown into application window).
- Check groups - they have priority above single light. Try to remove light from group(s).

In general, you can use "Info" button at Color tab to estimate final light color (and how it generated).

If it still has the wrong color, i will need to check your settings - please export registry tree from `HKCU/SOFTWARE/Alienfx_SDK` and `HKCU/SOFTWARE/Alienfxgui` and send it to me (or attach here).

PS: Pink is possible as some diodes can have different specter (and v2 always mix a bit more red), but yellow... This means NO red in light at all. Is both of them have correct colors into AWCC?
PPS: I have a hardware issue with tron lights at mine 13R2 before, but i lost blue (in fact, it just becomes dark) for all tron lights. This was visible in any software.

T-Troll, 2022-01-08:
Also, what about games? Do you use mine LightFX or Dell one?

vasudev-gm, 2022-01-09:
[afx_regfiles.zip](https://github.com/T-Troll/alienfx-tools/files/7835193/afx_regfiles.zip)
@T-Troll I don't use the LightFX library in games/apps that supports AFX. With AWCC I had to disallow it and some light combos flashing caused severe headaches for me.
Have attached the regfiles requested.
My Right Tron light is off since it won't display white.

T-Troll, 2022-01-10:
Yes, i see.
Setting looks correct, effects off, all lights to white (but it can be not white for this 4-bit lights).
So... One keyboard zone can be a bit red die to this. Try to switch to "Devices", press "Test White" to switch all lights to white, then play with white point (just decrease red one) until it will be more white. In fact, it's for this, for my 13R2 is 176-255-255 for clear white.

But maybe in your case, i need per-light white point.... I'll think about it.

Is right tron have the same non-white into AWCC? As i said, i have this before, then have broken cable to screen. And, by the way, lost blue as well (just like you). It's easy to check - set it to red, green, blue (basic) and check brightness - if blue is darker -this is like this. Check cables and replace.

PS: If you just require basic colors, you can disable "Gamma correction" in settings - it makes them close to the screen colors, but brightness curve modified.

vasudev-gm, 2022-01-11:
> Yes, i see. Setting looks correct, effects off, all lights to white (but it can be not white for this 4-bit lights). So... One keyboard zone can be a bit red die to this. Try to switch to "Devices", [...]
Yeah even my macro cable was broken when I was testing the new oem battery and the system was very hot and after touching the cable which was brittle and instantly broke off.
I will try that RGB values.

closed 2022-02-19 by T-Troll