T-Troll/alienfx-tools #114: Alienfan Not working Dell G5 15 5515

Hellohi3654, 2022-02-19:
**Describe the bug**
When trying to open alienfan-gui.exe it says can't install driver, check system configuration!
I am using the latest version as of writing this 
I started happening after I updated to the latest windows insider dev build.

**To Reproduce**
1. Download latest windows 11 insider dev build and try and run alienfan-gui.exe
2. See pop-up saying can't install driver, check system configuration!

**Screenshots**
[image]

T-Troll, 2022-02-19:
Uff... Quite bad news.

This is mean as it means - i use driver loading hack to load unsigned driver, and seems like this new build block my hack way.

I use 3rd-party library for this, so let me check about it updates.

Hellohi3654, 2022-02-20:
would it be possible for me to brute force and replace the driver manually?
Just as a quick fix for now?

Hellohi3654, 2022-02-20:
couldn't it also be that something has changed meaning it doesn't recognize my laptop anymore
As it says supported hardware not detected

Hellohi3654, 2022-02-20:
hwinfo is still able to control my fans but they use a different method I think

T-Troll, 2022-02-21:
It's not a hardware issue, but a driver loading issue.
And no, it's not possible to load it other way (i remove it because it can provide random BSODs).

What you can TRY to do:
- Disable secure boot (but i don't sure W11 allow it)
- Switch system into test mode
- Roll back to previous release
...
- Wait until updated driver loader for this release will be avaliable

Hellohi3654, 2022-02-21:
disabling secure boot didn't help
what driver what I roll back to the previous release?
or do I have to roll back a windows update?
Thanks

Hellohi3654, 2022-02-21:
would I roll back the intel driver?
Don't really know why there even is one
I have AMD and Nvidia
[image]

T-Troll, 2022-02-21:
Driver (hwacc.sys) was not changed for the long time.
So you should roll back release... Or wait until loader update will be published.

Hellohi3654, 2022-02-21:
hwacc.sys ok

T-Troll, 2022-02-21:
But now it can't be loaded this way. And i will not pay M$ $150 per year to sign it.

Hellohi3654, 2022-02-21:
that's fair

Hellohi3654, 2022-02-21:
where is hwacc.sys located?

Hellohi3654, 2022-02-21:
hmm
[image]

Hellohi3654, 2022-02-21:
[image]

Hellohi3654, 2022-02-21:
am I looking in the right spot?

T-Troll, 2022-02-21:
The issue is not the driver itself, but the fact i use hack method to load it (old one, it works from XP until W11) - because W10+ doesn't allow loading unsigned drivers without serious effort from the user (from W10).

hwacc.sys is my own driver used to access ACPI BIOS calls, it's located into app folder.

Hellohi3654, 2022-02-21:
oh ok
well I'm willing to allow unsigned driver loading is there a tutorial I can follow

T-Troll, 2022-02-21:
It doesn't help. Trying to load it without Test Mode provide BSOD in W11, so i disable this way.

Thank you to reveal me this point, but for now you can only roll back insider release to previous one.
... or, like me, stay at W10.

Hellohi3654, 2022-02-21:
I was going to stay on win 10 because there were issues with AMD CPU's but then all of my Microsoft store games broke and the only way I could fix them was to update to win 11 unfortunately.

Also would it be possible for you to be able to control the fans the same way hwinfo does since it still seems to work fine

Hellohi3654, 2022-02-21:
also how do I disable alienfan from starting on boot manually since I can't open it anymore

T-Troll, 2022-02-21:
> Also would it be possible for you to be able to control the fans the same way hwinfo does since it still seems to work fine
No, it's a dangerous way, as well as many gears not compatible with it at all.

> also how do I disable alienfan from starting on boot manually since I can't open it anymore
Open start menu, start to type "task scheduler", open it, remove `AlienFan-GUI` entry. I'll add automatic remove at next release.

T-Troll, 2022-02-21:
> I was going to stay on win 10 because there were issues with AMD CPU's but then all of my Microsoft store games broke and the only way I could fix them was to update to win 11 unfortunately.

I broke it from time to time as well, it's two shell commands return it back.

T-Troll, 2022-02-21:
Ok... I'll keep this task opened, and drop a message here, then i have updates.

Hellohi3654, 2022-02-21:
> > I was going to stay on win 10 because there were issues with AMD CPU's but then all of my Microsoft store games broke and the only way I could fix them was to update to win 11 unfortunately.
> 
> I broke it from time to time as well, it's two shell commands return it back.

Yeah I tried to use those shell commands but everything I tried didn't fix it
I ended up finding out that there was a cumulative update that was meant to fix my issue so I had to get win 11 to get it.

Hellohi3654, 2022-02-21:
> Ok... I'll keep this task opened, and drop a message here, then i have updates.

Ok thanks

Hellohi3654, 2022-02-21:
> No, it's a dangerous way, as well as many gears not compatible with it at all.
>
>
> Open start menu, start to type "task scheduler", open it, remove `AlienFan-GUI` entry. I'll add automatic remove at [...]

Thanks

T-Troll, 2022-02-22:
Ok, you can try to find a way to load driver yourself.
I use [this](https://github.com/hfiref0x/KDU) library to load my driver.

So you can try to load utility from there, then try to load my driver using it with different parameters (provider/shellcode).
In case you can load driver successfully (you can run alienfan-gui/cli for check), please share provider/shellcode IDs for me, i can alter app settings.

If you can't load it with any combination - we have no choice but wait until guys update it.

WARNING: using incorrect provider/shell can provide BSOD. Close all other apps, use with care!

Hellohi3654, 2022-02-27:
I experimented with kdu and found that this command loads your driver just fine
no BSODs and I have tested it and it seems to be 100% working
"kdu.exe -prv 1 -scv 3 -drvn HwAcc -map HwAcc.sys"

T-Troll, 2022-02-27:
Great news!
Ok, i'll check it at W10 and then publish update.

You can load driver by this command meanwhile before `AlienFan-GUI` start.

Hellohi3654, 2022-02-27:
Yeah I have setup a bat file that runs on startup that loads the driver
then opens alienfan-gui using task scheduler

T-Troll, 2022-02-28:
Ok, let's try the easy way first.
This is [updated loader DLL](https://github.com/T-Troll/alienfx-tools/files/8150233/kdl.zip).
Please overwrite kdl.dll into app folder by it and try (NB: don't forget to disable your current loading script and reboot before it!).

If it not works i need more deep library update, it took some days...

Hellohi3654, 2022-02-28:
your new version 5.5.4 seems to work fine

Hellohi3654, 2022-03-15:
Have you tried using version 5.5.5.1
5.5.7 seems to break my fans 
Haven't tried 5.5.6

T-Troll, 2022-03-16:
@VinyLimaZ What gear do you have and what exactly happened?

@Hellohi3654 Oh... Can you open an issue about it at alienfx-tools with more details? I have no changes in fan control (except some for G5, but it should not break anything) after 5.5.5.1, so let's check and fix.

VinyLimaZ, 2022-03-16:
> 

@Hellohi3654 No I tried the same version you used (5.5.4) but didn't work :/ I'll try that 5.5.5.1 later! TY

@T-Troll WDYM by gear? My AW? It's a M15 R6 (2021), nothing happened, to me never worked in the Win11, On 10 it's okay, I upgraded to 11 and stopped work :(

VinyLimaZ, 2022-03-16:
M15 R6 (2021)
Nothing I mean that to me never worked on Win11, I received that error message saying about the fan driver error and I'm trying to work around it, because I don't want to use the AWCC... The problem with the AWCC is that shit audio driver that bugs my mic in my stream :(

T-Troll, 2022-03-17:
Ok, let's check.
First, try [5.5.4](https://github.com/T-Troll/alienfx-tools/releases/tag/5.5.4). In case it works, i will need your help to catch what is broken in latest.
Second, try to run `alienfan-cli rpm` from Administrator command line. Keep your eye on error message (it has many of them), it helps me to configure out what exactly happened.
In case it complains about "System configuration issue" - try to load driver manually ([as defined above](https://github.com/T-Troll/alienfx-tools/issues/114#issuecomment-1047348423) into this thread).

VinyLimaZ, 2022-03-17:
5.5.4 didn't work, BUT I'll try to run that command to see the errors message and return to you (I'll try others versions too)

T-Troll, 2022-03-21:
Any feedback?

VinyLimaZ, 2022-04-09:
```
PS C:\Program Files\AlienFX Tools> .\alienfan-cli.exe
AlienFan-cli v5.6.0
System configuration issue - see readme.md for details!
```
I'll try to load manually the driver later, but the commando doesn't return anything that helps... 

The strange part is that when I reinstalled the WIN11 at first the app worked, but after the update stoped :/

T-Troll, 2022-04-09:
Yes, it is. I'm a bit tired from this - M$ make a lot of effort to block open source drivers to load into W11.
AFAIR, situation will be even worse later.

All we can do is to try to find a way to load it anyway, it's wrong to follow their way.

So, latest KDU didn't help? Error code and driver not loaded? Or driver loaded but can't be accessed by tools?

JessloRz, 2022-04-18:
Version 5.5.5.0 worked for me on windows 11, the only problem i had with that version is that, when i click the minimize button of the window, it crashes.
Minimizing from app button works fine

T-Troll, 2022-04-19:
A mystery!
The only change is in alienfx-gui, but the driver, loading, and fan SDK are the same!

Maybe some Visual Studio update broke it? Hmm...
Can you try to load .zip from the latest release, then copy hwacc.sys, kdl.dll and drv64.dll from 5.5.5.0?

JessloRz, 2022-04-19:
It works!

[image]

T-Troll, 2022-04-19:
Great!
So, as a hotfix, you can always use these files from 5.5.5 (No changes inside, in fact).
I'll try to configure out what happened and maybe generate correct ones for test.

T-Troll, 2022-04-19:
Mystery continues:

```C:\work\DevStudio\alienfx-tools\x64\Release\t>fc HwAcc.sys ../t2/HwAcc.sys
Comparing files HwAcc.sys and ../T2/HWACC.SYS
FC: no differences encountered
C:\work\DevStudio\alienfx-tools\x64\Release\t>fc kdl.dll ../t2/kdl.dll
Comparing files kdl.dll and ../T2/KDL.DLL
FC: no differences encountered
C:\work\DevStudio\alienfx-tools\x64\Release\t>fc drv64.dll ../t2/drv64.dll
Comparing files drv64.dll and ../T2/DRV64.DLL
FC: no differences encountered
```

It's between 5.5.5 and 5.5.5.1.
I have 2 ideas what happened:
1. AV broke something in installer
2. Installer broken by itself

Can you please check original 5.5.8 not from .msi, but from .zip?

JessloRz, 2022-04-19:
Yeah, it seems like the problem is with the installer version, i didn't try downloading the .zip until now so didn't notice.

[image]

T-Troll, 2022-04-19:
Uff... This kind of issue is so hard to pinpoint.
Thank you for help!

Seems like W11 installer broke/ignore something from .msi into the process. I'll check.

JessloRz, 2022-04-19:
> Yeah, it seems like the problem is with the installer version, i didn't try downloading the .zip until now so didn't notice.
> 
> [image]

Wait, this is actually pretty weird, tried installing again with the .msi and it works. Maybe installing 5.5.5.0 first has something to do with it? I can only help testing though haha...

T-Troll, 2022-04-19:
A puzzle inside the mystery.
No, it's not a difference in saved data or something between it.
The only reason in this case is GitHub/Windows AV issues (broken .msi or AV block/broke something during installation).
I bet on 2nd.

Anyway, let's keep our eye, and we know the solution now.

T-Troll, 2022-04-19:
Ohhh! WAIT!
Reboot now with 5.5.8 active. It can be a surprise!
(Driver only loaded at first app start, then stay in RAM until reboot, so if you load it once - it doesn't touch anymore).

JessloRz, 2022-04-19:
It keeps working. I also tried uninstalling and installing a few more times but it worked every single one.
I'm out of ideas haha...

Hellohi3654, 2022-06-11:
I think it's better if I post this here

I think a new windows update has broken the fan control again
same behavior as before

"kdu.exe -prv 20 -scv 3 -drvn HwAcc -map HwAcc.sys"
this command now works

T-Troll, 2022-06-12:
Even more funny. I got a BSOD today at W10 - in RTCore as well. Seem like latest Windows update.
Ok, i'll switch apps to provider 20.

T-Troll, 2022-06-12:
Uff... A bit tricky with this provider. Can you check other?

Hellohi3654, 2022-06-17:
> Uff... A bit tricky with this provider. Can you check other?

Only provider that works for me

T-Troll, 2022-06-27:
@Hellohi3654 - Can you please check:
- Disable CDU loading
- SHUTDOWN (not restart) system
- Start it again
Is driver loaded in this case?

Hellohi3654, 2022-08-06:
> Uff... A bit tricky with this provider. Can you check other?

Checked the other ones almost two months later sorry about that

`kdu.exe -prv 13 -scv 3 -drvn HwAcc -map HwAcc.sys`

works along with 

`kdu.exe -prv 20 -scv 3 -drvn HwAcc -map HwAcc.sys`

Hopefully 13 easier to implenment 

> @Hellohi3654 - Can you please check:
> 
> * Disable CDU loading
> * SHUTDOWN (not restart) system
> * Start it again
>   Is driver loaded in this case?

No nothing I do makes it work still getting same error using

`kdu.exe -prv 1 -scv 3 -drvn HwAcc -map HwAcc.sys`

and just opening alienfan-gui doesn't work

T-Troll, 2022-08-07:
Thank you. As you can see at Discord, i plan to drop driver access and use the other way.

Hellohi3654, 2022-08-07:
> 

oh cool

[12 more replies]

closed 2022-08-15 by T-Troll