T-Troll/alienfx-tools #116: no lights detected 15 R4

gamerjag, 2022-02-20:
no lights detected when i launch the program i tried this as a last resort because awcc won’t turn on my lights neither, they all went black after i reinstalled awcc with “go dark” on

T-Troll, 2022-02-21:
My apps are universal, and it's not possible to auto-detect all versions of hardware (especially one published after release).
I can have a small database from volunteers used to auto-assign known systems.
If your system not into this database, you always have an option to define light manually.

But R4 IS into it.

So what happened - is any devices (you should have 2 of them) detected, or you have warning about no devices?

T-Troll, 2022-05-13:
No feedback, closed.

closed 2022-05-13 by T-Troll