T-Troll/alienfx-tools #123: Add GPU boost for Dell G5 15

Hellohi3654, 2022-03-20:
**Is your feature request related to a problem? Please describe.**
I used to be able to have Alienware control center open with alienfan-gui and alienfan-gui would work as normal.
I did this so that I could use the G-mode from alienware control center to boost my GPU's tgp to 130 watts instead of 115 watts.
Unfortunately it seems that having Alienware control center open with alienfan-gui breaks alienfan-gui now with a new update, so I have uninstalled alienware control center and alienfan-gui is back to working as normal.

**Describe the solution you'd like**
Would it be possible to add the GPU boost to alienfan-gui when I click the G-mode button or if I change the power mode or GPU limit in alienfan-gui

Model: Dell G5 15 5515 (Ryzen Edition)

Thanks

Hellohi3654, 2022-03-20:
Oh nevermind the bug still seems to be having even with alienware control center uninstalled
This is the bug
My fans are stuck at that RPM even though they should be at boost 100
[image]

T-Troll, 2022-03-20:
Yes, AWCC can override fan/power settings, this is the result.
Also, this can be a result of so high overboost (i see you have 190(!), it's quite high).

About G-key... I check, but can't find how AWCC/BIOS switch it, and, as you understand, haven't hardware to check,
So i recommend you to continue study your ACPI BIOS (especially EC and around AWCC block), i can support you with test app and future implementation into main ones.

Hellohi3654, 2022-03-20:
RW Everything is no longer working I get this error everytime I try to open it
I have tried reinstalling
[image]

Hellohi3654, 2022-03-20:
[image]

T-Troll, 2022-03-21:
Check or disable antivirus. RWE, like me, use its own driver, and it can be blocked/removed by it.
By the way, seems like recent W11 update change something into driver subsystem, we are discussing it into different issue.

Hellohi3654, 2022-03-21:
I have my antivirus always disabled I think it might be something to do
with that

T-Troll, 2022-03-21:
I don't sure you have, check please - sometimes Windows can enable it back after update "for your safety".
RWE driver is signed, so it should not have any issue to load other way.

T-Troll, 2022-05-11:
I have some news about G-key.
Seems like it's status in EC, offset 0x220 (ACPI name GMOD).
But it only read from some ACPI functions, not wrote.
Can you please check - `alienfan-cli direct=25,2` - it should pring current G-Mode state (on/off) and `alienfan-cli direct=25,1` - this MAY switch it on and off?

Hellohi3654, 2022-05-11:
Oh cool thanks
I'll try that out

T-Troll, 2022-05-11:
> I have also been playing around with rweverything and it seems that it not longer works on my device
Disable "Memory Integrity" at security settings (see details into project readme, we meet this issue already).

Hellohi3654, 2022-05-12:
both of them seem to do nothing
Also I have tried disabling Memory Intergrity

[image]

T-Troll, 2022-05-12:
Try to run first command then g-mode ON by AWCC. It shows we are on the right way or not.

Hellohi3654, 2022-05-12:
yeah must be on the right track
[image]

T-Troll, 2022-05-13:
Great! Now we need to find how to set it up and check. Let me think.

Hellohi3654, 2022-05-13:
> Great! Now we need to find how to set it up and check. Let me think.

Cool thanks

T-Troll, 2022-05-14:
Hold on!

Thermal mode is BIOS profile, already supported (you have 2 of them, remember?).
Yes, it should change PLs (not GPU, but CPU too in most cases).
G-mode alter PLs and fix fans to max. Like "Performance/Full speed" thermal mode at Alienware (not G-series).

Anyway, i check all G-series dumps i have. Function 25 SHOULD switch g-mode, question is arguments. I have only one BIOS which different. Most interesting, it presents at Alienware as well, but do nothing ^_^.
So... Try `alienfan-cli direct=25,1,1` (last parameter 0 or 1). If this not help, try last parameter 99 and 98 (it SHOULD switch to G-mode, the question what number to be set...)

Hellohi3654, 2022-05-15:
I tried it out
the only time I got the result to change was when I did this
[image]

direct=25,0,any number would result in -1 and direct=25,1,any number would result in 0

I also tried doing these commands while g mode was on to see if they turned it off but it made no difference
although when g mode was on and I did direct=25,1,0 I got result 1 and when I did direct=25,0,0 I got -1 even when g mode was on

T-Troll, 2022-05-15:
Thanks. Seems like you have that BIOS that doesn't support it.
Ok, we'll try more.

BTW, 25,0 is wrong command, only 25,1 and 25,2 valid - that's why it's always -1.
Do you try 25,1,98 and 99? Other values?

T-Troll, 2022-05-15:
BTW, i configure out i didn't have you ACPI BIOS dump. Can you share it with me?
(to doing so, install [rweverything](http://rweverything.com/), start, open "ACPI" (wait a moment, it took time), press "save all" button).

Hellohi3654, 2022-05-15:
> Thanks. Seems like you have that BIOS that doesn't support it.
> Ok, we'll try more.
> 
> BTW, 25,0 is wrong command, only 25,1 and 25,2 valid - that's why it's always -1.
> Do you try 25,1,98 and 99? Other values?
> Don't need to check 25,2 - it's G-mode status reading.

Yeah I tried 25,1,99 and 98

Hellohi3654, 2022-05-15:
> BTW, i configure out i didn't have you ACPI BIOS dump. Can you share it with me? (to doing so, install [rweverything](http://rweverything.com/), start, open "ACPI" (wait a moment, it took time), press "save all" button).

I can't get rweverything to work on my PC so I used a friend's laptop to get it
He has the exact same model as me so there shouldn't be any difference

[AcpiTbls.rw.txt](https://github.com/T-Troll/alienfx-tools/files/8696281/AcpiTbls.rw.txt)

T-Troll, 2022-05-16:
> I can't get rweverything to work on my PC

Hey! It's THE SAME reason fans stop working. Check security and try to disable AV, now kernel driver loading blocked for your gear completely!

I check BIOS - yes, it's the only one function 25,1 does not enable G-mode.
For now, i found function _Q14 (you can search for `Method(_Q14)` into file - it's switch the same value as 25,1. But i still don't sure.
So next time i drop you cli to check.

Hellohi3654, 2022-05-16:
I told you before I have no AV I have completely disabled it on my computer
Alien fan control works sometimes but rweverything never works
I'm also using the portable version of rweverything and I have
re-downloaded it a couple of times

T-Troll, 2022-06-04:
Hey.
Here are a special cli app for you - 
[alienfan-cli.zip](https://github.com/T-Troll/alienfx-tools/files/8836583/alienfan-cli.zip)

Try `alienfan-cli test=0` (or 1) - did this switch g-mode?

Hellohi3654, 2022-06-05:
> 

Cool thanks I have tried it out
[image]

`alienfan-cli test=0` switches on g-mode
the fans are set to their max RPMs and my GPU's tdp is increased from 115 watts to 130 watts 

running `alienfan-cli test=1` doesn't seem to do anything but running `alienfan-cli test=-1` switches g-mode back off again where the fans start running at a lower rpm and the GPU's tdp is reduced to 115 watts

Other thing I noticed when I use AWCC g-mode the fans are only increased to 5000 on fan 1 and 4700 on fan 2 but with your g-mode they run at what I assume to be their actual maximum rpm

Anyway thanks for making that cli how are you planning to add it to alienfan-gui?
Just by pressing the g-mode button or maybe a customisable button
Thanks

Hellohi3654, 2022-06-05:
I also have one another question
is it possible to set overboost values manually
I can't find an option anywhere and I want to set them by myself as the overboost sometimes doesn't correctly get my maximum rpm
I don't really care if boost 90 is the same as 100 my fan curves are very aggressive anyway

T-Troll, 2022-06-05:
Cool! I'll add it at next release (it will be BIG...).
Yes, you can set overboost manually - `alienfx-cli setover=<fan id>,<boost>`

T-Troll, 2022-06-05:
> alienfan-cli test=0 switches on g-mode
> running alienfan-cli test=1 doesn't seem to do anything but running alienfan-cli test=-1 switches g-mode back off

I check source code. -1 = 0 into it. So, seems like 1 - turn on, 0 - turn off. Can you check, please?

Hellohi3654, 2022-06-05:
> > alienfan-cli test=0 switches on g-mode
> > running alienfan-cli test=1 doesn't seem to do anything but running alienfan-cli test=-1 switches g-mode back off
> 
> I check source code. -1 = 0 into it. So, seems like 1 - turn on, 0 - turn off. Can you check, please? Or maybe 0 both on/off switch.

Yeah 0 is a toggle on/off
1 doesn't do anything

Hellohi3654, 2022-06-05:
> Cool! I'll add it at next release (it will be BIG...).
> Yes, you can set overboost manually - `alienfx-cli setover=<fan id>,<boost>`

Cool Thanks

T-Troll, 2022-06-05:
Here are small gift - [AlienFan.zip](https://github.com/T-Troll/alienfx-tools/files/8839577/AlienFan.zip)
It's pre-release of the next version.
Command `togglegmode` added to CLI, and G-key shortcut (F17) in GUI. Check.
Later i'll add it into GUI interface as well.

Hellohi3654, 2022-06-06:
thanks been working great
I think we can close this now

T-Troll, 2022-06-06:
Hold on! Now only Ryzen is supported, and it's a lot of job even for it.
I need to integrate in into GUI (do you remember - we have current status), as well as support other G-series gear.

T-Troll, 2022-06-07:
Can you check this version, please - [AlienFan.zip](https://github.com/T-Troll/alienfx-tools/files/8849143/AlienFan.zip)

Changes:
- G-mode avaliability check
- G-mode switch state now into profile and in UI
- CLI commands changed - `gmode` for read state, `setgmode=x` to set it on/off (x=1/0)

Did i break something (like UI change, shortcut change, etc.)?

Hellohi3654, 2022-06-07:
Oh ok
I'll check

Hellohi3654, 2022-06-07:
ok I tested 
Found a few problems
When I click the g mode button in the gui it turns on but when I uncheck it, it doesn't turn off
When I press the g mode button on my keyboard the gui button doesn't become ticked it also doesn't become unticked when I disable g mode via the keyboard button

T-Troll, 2022-06-07:
> When I click the g mode button in the gui it turns on but when I uncheck it, it doesn't turn off

Should not be like this. Ok, i check.
Did it still disable then you use keyboard shortcut?

> When I press the g mode button on my keyboard the gui button doesn't become ticked it also doesn't become unticked

It's normal, as it should be. It's trying to reflect current state as well.

T-Troll, 2022-06-07:
Ok, can you check CLI for me, please?
Call `alienfan-cli gmode setgmode=1 gmode setgmode=0 gmode` and share the log (hint: you can copy text, not image - use "mark" from cmd window menu).

Hellohi3654, 2022-06-08:
```
C:\Alienfan\alienfan-cli>alienfan-cli setgmode=0
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0

C:\Alienfan\alienfan-cli>alienfan-cli gmode
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode state is Off

C:\Alienfan\alienfan-cli>alienfan-cli gmode
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode state is Off

C:\Alienfan\alienfan-cli>alienfan-cli setgmode=0
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0

C:\Alienfan\alienfan-cli>alienfan-cli gmode
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode state is Off

C:\Alienfan\alienfan-cli>alienfan-cli gmode
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode state is Off
```

Still not showing that gmode is on

```
C:\Alienfan\alienfan-cli>alienfan-cli setgmode=0
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0

C:\Alienfan\alienfan-cli>alienfan-cli direct=25,2
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Direct call result: 0

C:\Alienfan\alienfan-cli>alienfan-cli direct=25,2
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Direct call result: 0

C:\Alienfan\alienfan-cli>alienfan-cli setgmode=0
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0

C:\Alienfan\alienfan-cli>alienfan-cli direct=25,2
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Direct call result: 0

C:\Alienfan\alienfan-cli>alienfan-cli direct=25,2
AlienFan-cli v6.0.3
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Direct call result: 0
```

Also doesn't change from 0

Hellohi3654, 2022-06-08:
Btw everytime I boot gmode turns on when alienfan-gui opens 
Happens with 6.0.2 and 6.0.2.1

T-Troll, 2022-06-08:
A mystery! It works some posts ago into this thread.
I know why it's always on at start now, will fix soon.

I need another check.
Can you, please, run `alienfan-cli setgmode=0 direct=25,1 direct=25,2 gmode` two times?
Is it some difference in output?

Pedro-Thales, 2022-06-08:
> Btw everytime I boot gmode turns on when alienfan-gui opens
> Happens with 6.0.2 and 6.0.2.1

Same with me

T-Troll, 2022-06-08:
Guys, this bug connected to 25,1 always return 0, so, @Pedro-Thales, joint the testing.
I'll fix it into next release one way or another, but better if we make status working.

T-Troll, 2022-06-08:
As a hotfix, please use this [alienfx-tools.zip](https://github.com/T-Troll/alienfx-tools/files/8865449/alienfx-tools.zip)
It will work a bit better, but will miss the point if G-Mode already started before app start.

T-Troll, 2022-06-08:
Some words about what happened and what to do.

How it works now (in release above, not 2.1):
At startup, app get g-mode status using 25,2. It should be 3 values - unsupported (like on my gear), on, off. Last 2 compared with checkbox state - and toggle it if not the same.
Some BIOS uses 25,1 function for this (it stores state), but for SE/Ryzen i use other function - Q14 - 25,1 have a bug and don't work. Q14 has no parameters or reply, only toggle to other state.

The issue is 25,2 stops working (but works before) at Ryzen/SE - so i don't know initial G-mode status. Always off.

So i need your help to configure out why it stops working or how to get current g-mode state.

Hellohi3654, 2022-06-10:
> 

Easy way to see if g-mode is on is too detect what boost the fans are running at and the temperature of the sensors

If the fans are running at max boost but the fan curve says they shouldn't be then g-mode must be on

You could also try detecting the GPU power limit as turning on g-mode increases the tgp from 115 watts to 130 watts so their must be some way to detect that

Hellohi3654, 2022-06-10:
Cool found a way to detect g-mode status

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: 0

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: -1

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: -1

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: -1

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: -1

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: 0

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: 0

C:\Alienfan\alienfx-tools-6.1.0>

When power mode was -1 g-mode was on and 0 was when it was off

Hellohi3654, 2022-06-10:
Here's a better representation

```
C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli setgmode=1
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: -1

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: -1

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli setgmode=1
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli getpower
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
Current power mode: 0

C:\Alienfan\alienfx-tools-6.1.0>
```

T-Troll, 2022-06-10:
Boost is always reported as 0 at this mode (and Aleinware's "Performance" as well).
GPU power limit different for different GPUs and systems. And modes as well.
Power mode is a hint for your system, yes. I need to check what will happen to others... BTW -1 means "error".

PS: Did you run commands from [here](https://github.com/T-Troll/alienfx-tools/issues/123#issuecomment-1149546191)?

T-Troll, 2022-06-10:
Ok, try this cli/gui - [fan_control.zip](https://github.com/T-Troll/alienfx-tools/files/8875790/fan_control.zip)
I use your hint about getpower (and check other way if not fail).

Hellohi3654, 2022-06-10:
> 

```
C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli setgmode=0 direct=25,1 direct=25,2 gmode
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0
Direct call result: 0
Direct call result: 0
G-mode state is Off

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli setgmode=0 direct=25,1 direct=25,2 gmode
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0
Direct call result: 0
Direct call result: 0
G-mode state is Off

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli setgmode=0 direct=25,1 direct=25,2 gmode
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0
Direct call result: 0
Direct call result: 0
G-mode state is Off

C:\Alienfan\alienfx-tools-6.1.0>alienfan-cli setgmode=0 direct=25,1 direct=25,2 gmode
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0
Direct call result: 0
Direct call result: 0
G-mode state is Off

C:\Alienfan\alienfx-tools-6.1.0>
```

Sorry didn't run that command earlier as I was fairly certain this would be the case

Hellohi3654, 2022-06-10:
> Ok, try this cli/gui - [fan_control.zip](https://github.com/T-Troll/alienfx-tools/files/8875790/fan_control.zip) I use your hint about getpower (and check other way if not fail).

```
C:\Alienfan\alienfx-tools-6.1.0.1-getpower>alienfan-cli setgmode=1
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 1

C:\Alienfan\alienfx-tools-6.1.0.1-getpower>alienfan-cli gmode
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode state is On

C:\Alienfan\alienfx-tools-6.1.0.1-getpower>alienfan-cli setgmode=1
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode set result 0

C:\Alienfan\alienfx-tools-6.1.0.1-getpower>alienfan-cli gmode
AlienFan-cli v6.1.0
Supported hardware v2 detected, 2 fans, 2 sensors, 2 power states. Light control disabled.
G-mode state is Off

C:\Alienfan\alienfx-tools-6.1.0.1-getpower>
```

Seems to work fine for me at least

Btw the Gmode tick box thing in the GUI does not become ticked when I press the Gmode button on my keyboard
[image]
but clicking on the GUI button and ticking it and unticking it turns Gmode on and off

T-Troll, 2022-06-10:
> Btw the Gmode tick box thing in the GUI does not become ticked

Noted.

Ok, thanks, i fix this then release. Close this bug after that. Thank you, i'll add you to credits at next release!

T-Troll, 2022-06-10:
Fixed in [6.1.1](https://github.com/T-Troll/alienfx-tools/releases/tag/6.1.1)

closed 2022-06-10 by T-Troll

Hellohi3654, 2022-06-10:
the g mode button still doesn't become ticked after using the g mode button on the keyboard
[image]

T-Troll, 2022-06-10:
Got it.
Yes, this happened in Fan-GUI if it was minimized. Real state are correct, it's only a glitch (until restart).

Note to self: Never update app then little drunk (it's Friday evening here).

Hellohi3654, 2022-06-10:
lol ok

Hellohi3654, 2022-10-11:
Cool thanks I will try this

If you mean switching from manual to level 1 in alienfan-gui
Switching to level 1 just stops the fans from working