T-Troll/alienfx-tools #125: Support for Alienware M15 R6

rodneypantonial, 2022-03-28:
Alienware M15 R6 is currently not supported.

I am currently using it and I'd like to make a pull request for it.
Where do I start to code the support for it for other developers to use?

T-Troll, 2022-03-28:
m15R6 100% supported, but not fully tested and have no entry into device database.
Or do you mean older M15? In this case, only lights supported, not fans/power.

So in the first case, we just need to check and add light mappings into database (and maybe other minor changes). In the second case, you need to commit into AlienFan-SDK (see M13R2 as an example).

T-Troll, 2022-03-28:
PS: My light code is quite universal, but device mappings different from model to model. So even fresh R7 and x15R2 supported, with manual setup.

rodneypantonial, 2022-03-28:
I mean the m15R6. Yes, I meant the entry into device database.
I'd like to add that entry since I have the device :)

T-Troll, 2022-03-28:
Great! So you need to manually add lights (both tron and keyboard) - here is [wiki page](https://github.com/T-Troll/alienfx-tools/wiki/How-to-start-(Beginner's-guide-and-tips)) how to do it.

Then save your light mappings (`alienfx-gui` have a special button for it), and attach it here - or you can add it into device database [here](https://github.com/T-Troll/alienfx-tools/tree/master/alienfx-gui/Mappings).

It will be included into next release.

ASchneiderBR, 2022-04-26:
Hello guys!

First of all, thank you @T-Troll for this great software :)

@rodneypantonial, I also own this notebook. I was able to probe all lights, except the power button. Here's my exported CSV (power button does not work):

[Alienware M15 R6.csv](https://github.com/T-Troll/alienfx-tools/files/8564449/Alienware.M15.R6.csv)

Device 01 has 135 lights, I could not figure if one of them is the power button...

Have you found the power button entry?

Thanks!

T-Troll, 2022-04-26:
Thank you,
But CSV is not correct, it's for 4 zones, not per-key. Can you please provide full one for share (as well as name devices correctly)?

The power button should be around Tron device, but it can be tricky to see - it blinks on test, not change color.

T-Troll, 2022-06-27:
4-zone model added to device database at release #156 (both lights and gird data).
Per-key model share same tron lights, but need other keyboard map.

ASchneiderBR, 2022-06-27:
Working great! Thanks!

closed 2022-06-27 by T-Troll