T-Troll/alienfx-tools #138: Fan control not working

Hellohi3654, 2022-05-15:
**Describe the bug**
Alien fan control has stopped working

[image]

I just restarted my computer and now it's decided not to work
My laptop didn't update or anything

I have tried 5.9.1, 5.9.2 and 5.9.3 
They all have the same result

My friend who also has the same laptop is experiencing this problem as well

T-Troll, 2022-05-15:
As described [here](https://github.com/T-Troll/alienfx-tools/issues/114), it can be resulted of Microsoft antivirus update - it broke some files at installation.
Try not to use installer, but download and unpack .zip file.
If this didn't help, try 5.5.5 (from zip).

Hellohi3654, 2022-05-15:
> As described [here](https://github.com/T-Troll/alienfx-tools/issues/114), it can be resulted of Microsoft antivirus update - it broke some files at installation. Try not to use installer, but download and unpack .zip file. If this didn't help, try 5.5.5 (from zip).

I always use the zip version

version 5.5.5.1 and 5.5.4 doesn't work either

Hellohi3654, 2022-05-15:
The bug is weird
My fan control has decided to work now on 5.9.3 not really sure why

T-Troll, 2022-05-16:
Ok, i close this bug, but just for remark.

Fan control can not start by, mainly, two reasons:
1. Fail to load hardware driver. I use hack-way to do so (i don't want to pay every year for signature), so some system settings and antivirus can interfere. First check it (what do you have in Update->Security, disable antivirus).
2. Driver loaded, but gear not detected (so i don't know how to control it). This can happen if i change something into fan SDK, so it only depends on new release. And, of course, i check it every release at all gear i have at home.

If Microsoft does something weird in update (both OS and AV) - it can trigger reason 1.
If Dell releases new BIOS - it can trigger reason 2.

closed 2022-05-16 by T-Troll

Hellohi3654, 2022-05-16:
The fan control has been working fine today but yesterday it wasn't working
at all
Seems very random when it decides not to work