T-Troll/alienfx-tools #146: G-Mode state toggles when opening/closing alienfan-gui

vinikmv, 2022-06-11:
*Describe the bug*
G-Mode state is toggled when opening/closing alienfan-gui. If G-Mode state is on, it toggles to off when opening/closing alienfan and vice-versa.

*To Reproduce*
Here's a video showing the behavior when closing the application. Same behavior when opening:
https://www.dropbox.com/s/19voar69j4hymnn/WindowsTerminal_hExo5mCsih.mp4?dl=0

https://www.dropbox.com/s/jyp0vve5o2ekbek/WindowsTerminal_ocwaFk7QbA.mp4?dl=0

*Expected behavior*
G-mode state doesn't change when opening/closing the application

*Desktop (please complete the following information):*
    Dell G15 5515 Ryzen Edition
    Windows 10 21H2
    AlienFX-Tools Version 6.1.1.0

T-Troll, 2022-06-11:
Interesting bug.
I can't check it myself because my zoo have no gear with g-mode support, but i have BIOS from your model, so let me try to configure out.

Hellohi3654, 2022-06-11:
Yeah can confirm this is happening

T-Troll, 2022-06-12:
Already fix it. Wait for the new release.

T-Troll, 2022-06-12:
Check [6.1.2](https://github.com/T-Troll/alienfx-tools/releases/tag/6.1.2).
Please close this issue if fixed.

closed 2022-06-12 by vinikmv