T-Troll/alienfx-tools #149: AlienFX Control does not work ( X15 R1)

SuppliedOrange, 2022-06-13:
**Describe the bug**
The AlienFX Fan Control and AlienFX Monitor seem to work fine. However, the AlienFX Control (Shortcut) does not start or seems to silently crash when executed.

**To Reproduce**
Steps to reproduce the behavior:
1. Attempting to start AlienFX Control through the shortcut.

**Expected behavior**
AlienFX Control should start when the shortcut is executed

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 6.1.3
 - System: Alienware X15 R1

**Additional context**
I would like to help by providing logs or such but I have no knowledge of C++ or H. Can you help me help you?

T-Troll, 2022-06-13:
Ok, let me check, looks strange (alienfan-gui should not work as well in this case).
I don't need logs for now, it's clean, thank you.

T-Troll, 2022-06-13:
Thanks. You reveal a serious bug. I'll fix it ASAP.

T-Troll, 2022-06-13:
Check this - https://github.com/T-Troll/alienfx-tools/releases/tag/6.1.3.1

SuppliedOrange, 2022-06-13:
The program works now. I'm having a hard time figuring out how to get it to detect my per-key lighting keys though. I'll be downgrading my system to Windows 10 tomorrow in hopes that AWCC will finally start working again. Thank you for the quick fix!

closed 2022-06-13 by SuppliedOrange

T-Troll, 2022-06-13:
You are welcome.
This app package is for enthusiast, so yes - not so easy to start with lights.
You can help the community by share your device config via "Save Lights and Grids" - so all x15 owners don't need to do it again.