T-Troll/alienfx-tools #150: AlienFan Control Does Not Open

personhoodie, 2022-06-18:
When I open the Fans and Power tab, the software just crashes.

It worked the first time I installed the software, and it was checking the overboost, but my laptop ran out of battery before finishing and now the tab doesn't work.

This is on 6.1.3.1

T-Troll, 2022-06-19:
Interesting bug...
First, can you please export registry data from `HKCU/Software/Alienfan` and `HKCU/Software/Alienfxgui` (it's general fan control and GUI configuration) for me?
Second, try to run `alienfan-gui`. Did it start correctly?

Seems like you have emergency shutdown during data update, so broken data into registry.

Anyway, let us pinpoint it!

personhoodie, 2022-06-19:
Here are the files, they don't seem to have any problems. I have tried deleted the reg data, and it now allows me to do another overboost check, but crashes after finishing the overboost check. Probably not a problem with my pc turning off? Alienfan-gui has the same problem, just opens briefly then closes.

[reg files.zip](https://github.com/T-Troll/alienfx-tools/files/8935343/reg.files.zip)

Using the old separate fan software I can still use the software to monitor RPM but it won't adjust the fans.

T-Troll, 2022-06-19:
Thanks for details, i'll investigate it.

T-Troll, 2022-06-20:
Being honest, i can't reproduce this bug - apps start correctly at my gear with your config.
But i discover and fix some bugs into overboost process, as well as add some checks into the places CAN provide exception.

Can you please test this version (alienfan-gui only)- [AlienFan.zip](https://github.com/T-Troll/alienfx-tools/files/8936692/AlienFan.zip)?
(please delete registry config to check correct overboost calculation for all fans)

In case it still crashes, i'll make a debug one for you to pinpoint it.

personhoodie, 2022-06-20:
This one also closes as soon as it is open as well, even after clearing reg.

Super weird: I just downloaded the 6.2.0 alienfx-tools.zip and when go on the fan and power tab, it prompts me for overboost. If I start overboost then stop it immediately, the software closes. If I let overboost finish, it closes after finishing. If I stop it just as something appears on the graph, the app stays open?

T-Troll, 2022-06-21:
It's hard to pinpoint - it works well on many systems from my testers, including mine.
I assume there is some issue into system data, so it crashes, then switch back to common mode from overboost one.

Can you please experiment with `alienfan-cli` (it should be run from Administrator CMD from app folder) ?
- `alienfan-cli rpm`
- `alienfan-cli temp`
- `alienfan-cli percent`
- `alienfan-cli getpower`

And then share logs for me?
Seems like one of these operations provide weird data or crash at your gear.

T-Troll, 2022-06-25:
Should be fixed into [6.2.1.1](https://github.com/T-Troll/alienfx-tools/releases/tag/6.2.1.1), please check!

closed 2022-06-27 by T-Troll