T-Troll/alienfx-tools #153: Functions to allow changing power options.

SiddharthManthan, 2022-06-21:
**Is your feature request related to a problem? Please describe.**
Currently, In order to change battery charging and power settings we need to use My Dell App, which starts multiple background services. Plus it has less features than it's predecessor Dell Power Manager. There is not function to enable peak shift or allow charging to 100% at specific time. 

Tools like Dell Command Configure is not available on Alienware and G series gaming laptops. 

**Describe the solution you'd like**
Implementing charging controls to AlienFX tools or creating new app will allow more control over charging controls without needing My Dell App. As far as I understand, alienfx-tools already uses proprietary function calls so it would not be very difficult to implement the functionalities.

T-Troll, 2022-06-22:
Good idea, i wish to have charge limiter at mine Alienware as well...

... But i am not sure that it's easy to implement.

Yes, i use proprietary Alienware ACPI BIOS methods. So fan/power controls is universal as a result and works with any Dell/AW hardware.
There are some calls inside it, i don't configure out yet, so MAYBE some of them are about battery control.

I only have Alienware gear, not Dell G, so i don't have this (or other) tool. So i need your help.
Can you, please:
a. Share some app screenshots to understand what kind of data it CAN send/read?
b. Dump your ACPI BIOS to check calls chain?
c. Be ready for test (a lot of BSODs or system issues will be into process, but non-fatal)?

SiddharthManthan, 2022-06-22:
> a. Share some app screenshots to understand what kind of data it CAN send/read?

[Dell Command Power Manager](https://www.dell.com/support/home/en-in/drivers/driversdetails?driverid=8mjfn) : 
Surprisingly this old app from 2017 works with laptop from 2021.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

My Dell : 
Available on app store.
[image]
[image]
[image]
[image]

> b. Dump your ACPI BIOS to check calls chain?  

What tool should I use to dump BIOS ?

> c. Be ready for test (a lot of BSODs or system issues will be into process, but non-fatal)?

Okay, I'll help with testing. I have a Dell G15 5515 Ryzen Edition.

SiddharthManthan, 2022-06-22:
> a. Thank you. I think most of the "features" are useless and can set to default. But i like min-max charge idea! b. [RWEverything](http://rweverything.com/). But DO NOT share it in public (it has some private data), drop me a link in private conversation like Discord. c. I have some BIOS dumps from it, but there are 4(!) known variations (completely different).

Can you share your discord profile ?

T-Troll, 2022-06-27:
Hmm... I dig deeper into this task, so that's what i find up to date:
- Seems like no proprietary function or method for it, feature is BIOS-related (so not universal).
- For my BIOS, i found some methods to control charge and read battery stats. I check how universal is it, and maybe add it for some AW notebooks.
- For your BIOS still looking (they don't use common naming, even in EC).

I also want to try to check different approaches (f.e. trough Windows WMI).

T-Troll, 2022-07-01:
Status update.

I dig into ACPI specs, and yes, there is a common way to control battery charge and discharge state.
But there are two issues:
- This method is optional and... it's not implemented both in you and mine BIOS.
- Using this violates IBM patent (f...king copyright!) - i think what's why Dell nerf their software.

But we have a dirty trick.

Now i can read full battery state directly from ACPI. I can just disable battery device if battery charged enough and enable it back if it below charge limit.
Pluses: it will work.
Minuses: Windows battery indicator and a lot of software using Windows data will be broken and don't reflect state correctly.

If this is OK for you, i can provide you a test version.

T-Troll, 2022-08-30:
Update: As i configure out, there is some functions about power control, but they don't work for my gear (do nothing).
Are you ready to help me test it?

SiddharthManthan, 2022-08-30:
Yes. I am ready.

T-Troll, 2023-01-09:
Latest [7.10.3](https://github.com/T-Troll/alienfx-tools/releases/tag/7.10.3) release has experimental charge control support.
There are still some limitations by both models and behavior, but i'll continue working on it.