T-Troll/alienfx-tools #159: Fan control start failure, disabling fan contol!

LazyServer, 2022-07-02:
Previously able to configure Fan curve, but unable to save the new setup. However, today, I'm unable to start Fan Control. 
Installed via MSI, version  6.2.3.2

**To Reproduce**
Steps to reproduce the behavior:
1. Start AFXControl
2. Click on Fan and Power
3. Received Windows Taskbar notification,  "Fan control start failure, disabling fan control!"
4. Go to Setting, Enable Fan Control cannot be selected.

Edition	Windows 11 Home
Version	21H2
Installed on	‎6/‎30/‎2022
OS build	22000.739
Experience	Windows Feature Experience Pack 1000.22000.739.0

System: Alienware R9, Intel i9

T-Troll, 2022-07-02:
Did we start to discuss it at Discord?

First, better experiment on `alienfan-gui` (fan control), instead of `alienfx-gui` (FX control) - it's lighter, but they share the same system profile.

Anyway, we have 2 issues in one:

1. Fan control can't start.
It's happened from time to time at W11, but the reason not completely clear.
First, check "Fast startup" is disabled into Windows Power Scheme (see [readme](https://github.com/T-Troll/alienfx-tools/blob/master/README.md)). In case it's enabled, this will happen if the system restarted, but not happened after full shutdown.
If it still continues, read #114 - it has some possible solutions.
Please write back here about result.

2. Fan configuration didn't store.
It can be some possible reasons, so i need to check your configuration. Can you please export 2 registry tree - `HKCU/Software/Alienfan` and `HKCU/Software/Alienfxgui` and attach files here (no personal data inside)?

LazyServer, 2022-07-03:
Did try the lighting, and it responded, but didn't investigate further, since I just need to control the fans.
I will go thru the rest of the notes.

Attached is the requested registry.
[Alienfan.zip](https://github.com/T-Troll/alienfx-tools/files/9034102/Alienfan.zip)

T-Troll, 2022-07-04:
I check your registry.

Please keep in mind - you enable "Fan settings" into Default profile. This means system profile will not be used if profile active - and so, you have 2 copy of different (not so different) fan control data. Maybe the bug was about the switch between system and profile-related settings. Anyway, i add some fixes about this into the latest release (#160).

According to fan control issue, you are not the only one. Seems like MS broke something again into the latest Windows/AV update. Most annoying this is random issue - so it starts and stop working for no reason after restart.

We are debugging it remotely with some guys, so if you can add something to our knowledge it will be helpful!

> Did try the lighting, and it responded,

Thanks, noted!

T-Troll, 2022-08-15:
Release v7 utilize different access method, so this issue is not relevant anymore.

closed 2022-08-15 by T-Troll