T-Troll/alienfx-tools #166: Is the "Low battery" condition reversed?

icestraw, 2022-07-26:
**Describe the bug**
Theoretically the "Low battery" event should have "on" state when below battery level, and "off" state when beyond.
Currently it's "off" when below and "on" when beyond

**To Reproduce**
Steps to reproduce the behavior:
1. Set effect mode to "monotoring"
2. Use "event monitoring" and set "low battery" event
3. You'll see the bug described above.

**Expected behavior**
As described

**Screenshots**
Not a UI bug, no screenshots

**System (please complete the following information):**
 - Model: Alienware m15 R6
 - Model details: 4 zone RGB
 - OS: Windows 11 RP (22621.232)

**Additional context**
Nope

T-Troll, 2022-07-27:
Let me check. It was reversed before, but maybe i lost the point, then rework it.

T-Troll, 2022-07-27:
Yes, it was a bug there. I fix it, check next release.

T-Troll, 2022-07-28:
Please check [6.4.3.1](https://github.com/T-Troll/alienfx-tools/releases/tag/6.4.3.1) and close this issue if it fixed.

icestraw, 2022-07-28:
Seems fixed

closed 2022-07-28 by icestraw