T-Troll/alienfx-tools #167: Events should overwrite other monitor types only on its "on" state

icestraw, 2022-07-28:
**Describe the bug**
As manual described, "Event colour always override Performance one then triggered, as well as both override Power state."
However currently it's not "triggered", it's "off" state overrides other monitor types as well, makes "mix monitor types" meaningless.

**To Reproduce**
Steps to reproduce the behavior:
1.Set multiple monitor types
2.The color will always match "off" state of "Event" events, event it's condition not satisfied ("off" state)

**Expected behavior**
The "On" state of "Event" should have the highest priority, but the "Off" state shouldn't.

**Screenshots**
Nope

**System (please complete the following information):**
 - Model: Alienware m15 R6
 - Model details: 4 zone RGB
 - OS: Windows 11 RP

**Additional context**
There're other strange behaviors in the "Event Monitoring" section, however I cannot distinguish whether they're bugs or by design.

T-Troll, 2022-07-28:
Thank you for testing, let's clarify it.
In fact, i use Montoring+Event mix all the time.

How it should work:
First, Monitoring color calculated.
Second, if event happened, it switches color to event one (or blink between Monitoring and Event).

And... It should work as designed... For gauge.
But you right, in case gauge mode off - it starts always activate event.

So let me check, is it configuration or code issue.

icestraw, 2022-07-28:
Since I'm using 4 zone RGB keyboard (not perkey) only, the gauge effect doesn't make much difference to me. So I just use single color mode and turned other effects off...

T-Troll, 2022-07-28:
I use gauge at 4-zone as well - here is my settings for whole 4-zone keyboard:
[image]

So... I found a bug, but it's not about trigger. Monitoring mode just set 2 colors instead of one if triggered for non-gauge zone.

Please test this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9205140/alienfx-gui.zip) - it should fix this.

icestraw, 2022-07-28:
I cannot download this package since Windows Defender treated as virus and auto-deleted

T-Troll, 2022-07-28:
Wow! WD starting crazy. It not complains original .exe, but complain if i put it into .zip...
Anyway, let me check to be sure, but, of course, it's not a virus.

T-Troll, 2022-07-28:
https://www.virustotal.com/gui/file/23635df024e64dbe3f4d6d2bd57990c3e2129e0d8b5284e7b365a08459366228?nocache=1
Of course, it's not. 

MS just acting as usual.

It can complain about KDL.dll (and it's right, i use hack for driver loading), but exe is clean.

icestraw, 2022-07-28:
[image]

I've replaced the exe directly and I don't know if I'm right.
My settings are just presented at the image above, I just want to reproduce what the system originally does: Distinguish AC/DC and Low battery. However with settings above my power button stays magenta whatever AC/DC mode. I don't know how to accomplish this setting. Did I just misunderstand some of the components?

T-Troll, 2022-07-28:
Ah, you are using Power state as a basis... Rear stuff, usually power button is enough.
Ok, let me check (power state calculated other way, because, instead of battery level, it not to be checked every time, but by system event).

T-Troll, 2022-07-28:
Ok... I forced to rework power monitoring - Windows now return different values, as well as some logic was not right about it.
So check this version - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9214482/alienfx-gui.zip).

The only limit it should have - in case other monitoring event in action for power light, it will stop animation (for charging or battery low).
Colors should be correct anyway.

icestraw, 2022-07-28:
Yes it's correct using this verision 👍

closed 2022-07-28 by icestraw

T-Troll, 2022-07-28:
Thank you for testing, it's hard for me to test all in solo...

icestraw, 2022-07-28:
Yes it's hard work since there're infinite conditions. Nevertheless your work is far much better than DELL.