T-Troll/alienfx-tools #168: Ambient mode should match "Desktop resolution" not "Active resolution"

icestraw, 2022-07-30:
**Describe the bug**
As title
In ambient mode, light zones can follow screen's color & lightness. However, sometimes resolutions of screenshots does not always match user's resolution, leading to "offsets" between keyboard light & screen zones. A technology named "Virtual super resolution" is available, and in this mode there're two resolutions: "Desktop resolution" (Logical) and "Active resolution" (Physical)
This app use "Physical resolution" but in fact there should be a map between "Physical" and "Logical" resolutions.

**To Reproduce**
Steps to reproduce the behavior:
1. Set multiple ambient zones between "keyboard" and "screen"
2. Set a higher resolution than physical (more different, more offsets)

**Expected behavior**
Keybaord lights should follow screen exactly.

**Screenshots**
My 4 zone keyboard doesn't have accurate bug representation on phone recording

**System (please complete the following information):**
 - Model: Alienware m15 R6
 - Model details: 4 zone RGB
 - OS: Windows 11 RP 22H2

**Additional context**
Nope

T-Troll, 2022-07-31:
It's not this.

I use data directly from active GPU screen buffer via DirectX.
So, it always reflects current screen (not desktop) resolution. Desktop one can vary for multi-screens, but i have a switch to select one of them.

What CAN be broken - i monitor screen configuration changes and adopt buffer to it. Here can be a bug (mine or Windows) after resolution change. I'll check.

PS: Then this happened, try to push "Reset" button at Ambient tab. Did it help?

T-Troll, 2022-07-31:
Yes, it was a bug, but the source is different.
Windows report virtual resolution (divided by scale level) for non-DPI-Aware apps, even from DirectX!

Quite strange, but easy to fix.

T-Troll, 2022-07-31:
Try this one - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9227902/alienfx-gui.zip)

Warning: It can provide incorrect app UI scaling then moving app window between different scale monitors! Can you check this?

icestraw, 2022-07-31:
[image]
Hmm maybe yes?
Plus, the ambient effect seems not watching monitor change

T-Troll, 2022-07-31:
I see it a bit unstable now.
I can share debug version with data trace, but you will need to have dbgview.

For now, just restart app (it's a result of how i fight incorrect window scaling).

T-Troll, 2022-08-01:
Try this one - [alienfx-gui.zip](https://github.com/T-Troll/alienfx-tools/files/9230201/alienfx-gui.zip)

This should always get resolution correctly.... Hope i not broke multi-monitor resolution, i can't check it now.

closed 2022-08-10 by T-Troll