T-Troll/alienfx-tools #170: Fan control start failure, disabling fan control!

mario6714, 2022-08-01:
Fan control don't work.

To Reproduce
Steps to reproduce the behavior:

    Start AFXControl
    Click on Fan and Power
    Received Windows Taskbar notification, "Fan control start failure, disabling fan control!"
    Go to Setting, Enable Fan Control cannot be selected.

Edition Windows 11 Pro
Version 21H2
OS build 22000.832

System: Dell G15 5510

T-Troll, 2022-08-01:
Duplicate of #114. Temporary override also available there.

T-Troll, 2022-08-01:
(Being honest, it's some ugly W11 "security" setting we didn't catch yet).

closed 2022-08-01 by T-Troll

mario6714, 2022-08-01:
i already tried a few things from that post and nothing work

T-Troll, 2022-08-01:
Try to load driver via other provider using KDU.

mario6714, 2022-08-01:
Can you tell me how? thanks

mario6714, 2022-08-01:
Like this:

C:\Users\mario\Downloads\KDU-1.2.0\Bin>kdu.exe -prv 20 -scv 3 -drvn HwAcc -map HwAcc.sys
[#] Kernel Driver Utility v1.2.0 (build 2202) started, (c)2020 - 2022 KDU Project
[#] Build at Thu Feb 17 01:33:48 2022, header checksum 0x4D588
[#] Supported x64 OS : Windows 7 and above
[*] Windows version: 10.0 build 22000
[*] SecureBoot is enabled on this machine
[+] Selected provider: 20
[*] Driver mapping using shellcode version: 3
[+] Driver object name: "HwAcc"
[+] No driver registry key name specified, driver object name will be used instead
[+] Input driver file "HwAcc.sys" loaded at 0x00007FF700B00000
[+] Provider: "CVE-2021-36276", Name "DBUtilDrv2"
[+] Drivers database "drv64.dll" loaded at 0x000002126DEC0002
[+] Extracting vulnerable driver as "C:\Users\mario\Downloads\KDU-1.2.0\Bin\DBUtilDrv2.sys"
[+] Return value: 0. Bye-bye!

T-Troll, 2022-08-01:
Yes. But maybe -drvn HwAcc0 - check.
After you load driver, all apps will work until reboot.

mario6714, 2022-08-01:
Other question, i am using the zip version, i need do something like move the dll to other place or other thing?

T-Troll, 2022-08-01:
No. .msi is just for guys like it installed.

mario6714, 2022-08-01:
I have this:

C:\Users\mario\Downloads\KDU-1.2.0\Bin>kdu.exe -prv 20 -scv 3 -drvn HwAcc0 -map HwAcc.sys
[#] Kernel Driver Utility v1.2.0 (build 2202) started, (c)2020 - 2022 KDU Project
[#] Build at Thu Feb 17 01:33:48 2022, header checksum 0x4D588
[#] Supported x64 OS : Windows 7 and above
[*] Windows version: 10.0 build 22000
[*] SecureBoot is enabled on this machine
[+] Selected provider: 20
[*] Driver mapping using shellcode version: 3
[+] Driver object name: "HwAcc0"
[+] No driver registry key name specified, driver object name will be used instead
[+] Input driver file "HwAcc.sys" loaded at 0x00007FF647E90000
[+] Provider: "CVE-2021-36276", Name "DBUtilDrv2"
[+] Drivers database "drv64.dll" loaded at 0x0000017F7B140002
[+] Extracting vulnerable driver as "C:\Users\mario\Downloads\KDU-1.2.0\Bin\DBUtilDrv2.sys"
[+] Return value: 0. Bye-bye!

now what i need to do?

mario6714, 2022-08-01:
I have the same error when i tried to use alienfan-gui.exe

mario6714, 2022-08-01:
this: Fan control start failure, terminating

mario6714, 2022-08-01:
Still nothing:

C:\Users\mario\Downloads\alienfx-tools(3)>kdu.exe -prv 20 -scv 3 -drvn HwAcc0 -map HwAcc.sys
[#] Kernel Driver Utility v1.2.0 (build 2202) started, (c)2020 - 2022 KDU Project
[#] Build at Thu Feb 17 01:33:48 2022, header checksum 0x4D588
[#] Supported x64 OS : Windows 7 and above
[*] Windows version: 10.0 build 22000
[*] SecureBoot is enabled on this machine
[+] Selected provider: 20
[*] Driver mapping using shellcode version: 3
[+] Driver object name: "HwAcc0"
[+] No driver registry key name specified, driver object name will be used instead
[+] Input driver file "HwAcc.sys" loaded at 0x00007FF72DE10000
[+] Provider: "CVE-2021-36276", Name "DBUtilDrv2"
[+] Drivers database "drv64.dll" loaded at 0x000002866C3A0002
[+] Extracting vulnerable driver as "C:\Users\mario\Downloads\alienfx-tools(3)\DBUtilDrv2.sys"
[+] Return value: 0. Bye-bye!

T-Troll, 2022-08-01:
1. You forget to copy drv64.dll from KDU.
2. Try to use different provides (`-prv`) - use `-list` to check - some of them didn't work for some systems.

mario6714, 2022-08-01:
Ok.

1: reboot.
2: run this command "kdu.exe -prv 20 -scv 3 -drvn HwAcc0 -map HwAcc.sys"
3: copy the dll "drv64.dll" in alienfx-tools(3)  folder
4 run alienfan-gui.exe

if this work i need repeat always when i turn on my laptop?

T-Troll, 2022-08-01:
3 before 2.
Just add driver loading at system start, before running any tool.

mario6714, 2022-08-01:
This dll "drv64.dll" is in the same folder for your app

mario6714, 2022-08-01:
in this folder i have your app and the app KDU

mario6714, 2022-08-01:
Not working:

PS: Here is link to Discord support server - guy write a batch file - https://discord.com/channels/906343597987758151/906343769782226944/997472865974767678

i created a bat file and create a task but the task just say is running and still have the same error

mario6714, 2022-08-01:
Not working:

1: reboot.
2: copy the dll "drv64.dll" in alienfx-tools(3) folder (KDU and alienfx-tools is i  the same folder)
3: run this command "kdu.exe -prv 20 -scv 3 -drvn HwAcc0 -map HwAcc.sys" (the bat in the task only say is running but nothing happen and still have the same error)
4 run alienfan-gui.exe

Same error " Fan control start failure, terminating" in alienfan-gui.exe

mario6714, 2022-08-01:
Any other solution?

maybe another command for KDU?