T-Troll/alienfx-tools #171: Crash Dumps

TheSQLGuru, 2022-08-11:
I have 4 crash dumps for alienfx-gui.exe in my user/appdata/local/crashdumps folder. 2 are from 7/12 and 7/16 and I am not sure what build of the app they are from. The other two are from 8/6 and they would be from the 6.4.3.2 build. I would be happy to get them too you, in some secure manner, if they might be of use. However, I have no idea what I might have been doing when they were created.

T-Troll, 2022-08-11:
No, dumps doesn't help me, i strip all debug info from release apps.
But thank you to reveal the fact.
Do you have any ideas, then it happened?
Do you have "Auto profile switch" enabled?

TheSQLGuru, 2022-08-12:
I didn't set up any profiles using AlienFX, so everything is still (to my
recollection) to their defaults. I did verify that I only have the Default
profile and Auto-Switch is not enabled.

I am not a person that concerns myself much with "bling", so I am currently
just using AlienFan Control to set fan curves and settings.

I just opened AlienFX to check the above. Sure enough when I use WhoCrashed
to check for a new dump there was one. I did nothing but open the app. Why
don't you cut me a debug-enabled build and I will run it and get you
whatever you need to assist with troubleshooting. I am technically savvy.

BTW, I realized that you only expose Save and Minimize buttons for the FX
GUI (maybe the Fan GUI too? didn't check). You absolutely need a Cancel or
Revert Changes button. You may also want to provide the ability to reset
all settings back to default if you do not have that.

See attached for WhoCrashed info.

Kevin G. Boles
SQL Server Consultant
SQL Server MVP 2007 - 2012
Indicium Resources, Inc.
@TheSQLGuru

TheSQLGuru, 2022-08-14:
I don't know if you got this, but I hope you did.

I also hope you have been notified that Github.com has flagged your site as
being malicious!!

"
This site has been reported as unsafeHosted by github.com

Microsoft recommends you don't continue to this site. It has been reported
to Microsoft for containing harmful programs that may try to steal personal
or financial information.
Go back
More information
Malicious sites, that contain links to malware, can damage your device and
disrupt your ability to use it. If you visit a malicious site, malware
could infect your device automatically. You might also be putting your
sensitive information—like passwords, credit card numbers, contact info, or
software activation keys—at risk. Learn more

Malicious sites often use spam emails, advertisements, or redirections from
other sites to trick you into downloading malicious software. Sometimes, a
trusted site may be hacked by bad actors and used to link to malicious
content. If in doubt, go back.

   -  Report that this site doesn’t contain malware threats
   -  Continue to the unsafe site (not recommended)

Microsoft Defender SmartScreen

Kevin G. Boles
SQL Server Consultant
SQL Server MVP 2007 - 2012
Indicium Resources, Inc.
@TheSQLGuru

T-Troll, 2022-08-15:
Read [this wiki page](https://github.com/T-Troll/alienfx-tools/wiki/Why-antivirus-complain-about-some-alienfx-tools-components%3F) - Windows Defender/Smart Screen not about security in general, but about to keep Microsoft monopoly.

It's a good example.
Instead of smart check for driver, it complains about driver loader - because they don't have $200/year from me, which they want.

Anyway, new release v7 should solve it.

TheSQLGuru, 2022-08-22:
OOPS!! I forgot that I had mentioned the Microsoft UNSAFE flag, and posted a new ticket for it! Sorry! I will see if I can delete/close that other ticket.

closed 2022-08-22 by TheSQLGuru