T-Troll/alienfx-tools #174: 7.0.0 version AlienFan-GUI cannot start with Windows

SDM855, 2022-08-15:
I was on the 6.4.3.2 version, and it starts with Windows fine. However, I updated the 7.0.0 version today, and it seems it can no longer run at startup.
I tried adding it to startup listings / disabling UAC / fully uninstalling and reinstalling. Unfortunately, it still cannot run at startup.
As I reverted to the 6.4.3.2 version, it works fine again.
Do you have any ideas or suggestions about this? Thank you very much!

**System**
 - Model: m15 r7
 - OS: Windows 11 stable

T-Troll, 2022-08-15:
Well... I found a bug preventing add start entry, but it should not interfere if already added.

First, what tool not start?
Second, open Task Scheduler and check entry with the name of the tool. Does it exist?

SDM855, 2022-08-15:
> Well... I found a bug preventing add start entry, but it should not interfere if already added.
> 
> First, what tool not start? Second, open Task Scheduler and check entry with the name of the tool. Does it exist?

The Alienware Fan Control. It is in the Task Scheduler but still won't boot on the 7.0.0 version.

T-Troll, 2022-08-16:
Kind of a mystery - it starts well for me.
Hmm... Do you install 6 and 7 into different folders?

SDM855, 2022-08-16:
It is the same folder. A mystery to me as well.

T-Troll, 2022-08-16:
Can you, please:
1. Check Task Scheduler entry (Alienfan-gui) - what in "Actions"?
2. Check Windows event logs about something related (app crash, task scheduler fail...)?
3. Start v7, then disable "Start with Windows" and enable it back?

mario6714, 2022-08-16:
I have the same issue, i check in settings "Start with windows" and i think is not working, "alienfan-gui
[image]
"

T-Troll, 2022-08-16:
Yes, it was a bug which prevent enable this functions (it can't create TS entry, fix will be today). But topic starter tells he has entry in TS, but app still not start.

mario6714, 2022-08-16:
Yes, i created the task but just ran a system level (the app is open but i cant open the gui to modify the options) i think, and if you open you will have two exe open in the same time

T-Troll, 2022-08-17:
Check [7.0.1](https://github.com/T-Troll/alienfx-tools/releases/tag/7.0.1) - it should fix it. But first disable "Start with Windows" then enable it back.

mario6714, 2022-08-17:
My entry is already removed

T-Troll, 2022-08-18:
Did it start now correctly?

mario6714, 2022-08-18:
Sorry for the late answer, Yes

T-Troll, 2022-08-18:
Great.

closed 2022-08-18 by T-Troll