T-Troll/alienfx-tools #181: Is the overboost option safe?

t3dium, 2022-08-21:
Does it overclock the fans? My cpu cooler used to get around 4800 rpm without it and now its able to reach 6700rpm which is a very significant increase.

T-Troll, 2022-08-21:
Yes. "Overboost" about it.
Also, without app, AWCC can set different thermal profile with other fan settings (lower, than manual mode).

t3dium, 2022-08-22:
Is it possible to reset fan curves back to normal or remove the overboost setting
[image]

I'm assuming if i make a flat-line at temp 0, boost 0, that'll use the stock fan curve?

T-Troll, 2022-08-22:
Use "X" button on top of the fan list.

PS: Stock is 0-0 curve.

t3dium, 2022-08-22:
Perfect, thank you. And if i were to keep the overboost, this won't blow the fans quicker in the long run?

T-Troll, 2022-08-22:
No. Performance mode do the same.

closed 2022-08-22 by t3dium