T-Troll/alienfx-tools #184: v7.0.3 Does Not Respect Existing Install Directory

TheSQLGuru, 2022-08-22:
SETUP:
I have v6.4.302 installed in D:\Program Files\Utilities\AlienFXTools\, and showing as available for uninstall in Programs and Features control panel. 

ACTION:
When I attempt to install v7.0.3 using MSI file, it defaults to installing in C:\Program Files\AlienFX Tools\.

EXPECTED:
Installing via MSI file should always check for an existing installation, and when finding one offer to install in that same directory.

T-Troll, 2022-08-22:
Uff... I use quite simple installer pack generator, i'll check is it possible for it...

TheSQLGuru, 2022-08-23:
Obviously this isn't a very high priority, as you are still given the chance to change the location to where you have it installed. 

I do wonder what will happen though if someone installs version X to a non-default folder and then newer version Y to the default location. Presumably all registry and other requisite system entries will be updated, and the older version will just be abandonware. This is probably something to verify though. It could be problematic if two sets of drivers, et al, are still active.

T-Troll, 2022-08-23:
Ah... No, it will be OK mostly.

Installer do not change any registry settings, just copy files and add start menu icons. Being honest, you don't need to run in after first install (zip is smaller and faster to deploy).

So BOTH versions with work (in case no changes in config, but it's rare).

I'll check this case at my gear (maybe old will be removed by installer, not sure - it has update process so check it).

TheSQLGuru, 2022-08-23:
I had wondered if I could just use the zip file to refresh my build. I will start doing that.

T-Troll, 2022-08-23:
Yes, you can do it for minor releases without any issue.
Major have changes preventing it in most cases.

closed 2023-06-16 by T-Troll