T-Troll/alienfx-tools #186: Fan Control Issues

TheSQLGuru, 2022-08-22:
v7.0.3

Findings Regarding Fan Power modes. I do not think I have done anything to affect these settings (I do have manual fan curves set though). Laptop was running at 2-5% with 11980HK/3080 throughout the testing. Windows 10 x64 21H2 fully patched.

GUI Settings: Overboost not greyed out, CPU Boost AC and Battery Efficient.

Level 1 Speeds:
Fan 1 - 2400
Fan 2 - 2200
Fan 3 - 2300
Fan 4 - 2000

Level 2 Speeds:
Fan 1 - 2200
Fan 2 - 2000
Fan 3 - 0
Fan 4 - 0

Level 3 Speeds:
Fan 1 - 2400
Fan 2 - 2200
Fan 3 - 2300
Fan 4 - 2000

Level 4 Speeds:
Fan 1 - 5400 --> 5000
Fan 2 - 5300 --> 5000
Fan 3 - 4600 --> 3000
Fan 4 - 4300 --> 3000

Level 5 Speeds (bug?) sometimes stayed at lower level (say 1) when going to 5. But the final values are still WAY less than both 4 and 6!
Fan 1 - 0 --> 2400
Fan 2 - 0 --> 2200
Fan 3 - 0 --> 2300
Fan 4 - 0 --> 2000

Level 6 Speeds:
Fan 1 - 5400 --> 5000
Fan 2 - 5300 --> 5000
Fan 3 - 4600 --> 3000
Fan 4 - 4300 --> 3000

ISSUES:

1) I changed back and forth a LOT for testing. 3 times my entire laptop "hung", with the display static and accepting no inputs from mouse or keyboard. This happened for about 8 seconds or so. Then all was fine. The setting DID change immediately, but the GUI did not represent the change until after the Hang was over.

2) I would never expect a HIGHER level to have a LOWER fan speed for ANY fan. This is clearly the case for several transitions. 

3) Care must be taken to NEVER allow a fan to instantaneously drop to ZERO speed on ANY transition, even if the final fan speed is zero. This happens on a number of transitions, and it opens the potential for (very large) instantaneous temperature spikes if the system is under load. Keep the fan speeds all the same from the just-left level until the app has acquired the new values and sets them (or better, per #4 below, gradually shifts to the new speed). 

4) Sometimes, but not always, a reduction in fan speed (say from  level 6 to 2 or manual) the fans take a LONG time to spin down to their new speeds (10-15 seconds?). Other times this is insatantaneous. I suggest something in between for ALL transitions. Never go from 0 to max fan speed instantanously when a higher level is chosen for example. Take the new/old fan speed delta and divide by some small time increment (2-3 secs??) and ramp up or down evenly over that time span - increaseing or decreasing some fixed amount per tenth/twentieth of a second? 

5) In the 7.0.2 release notes:
"Fan-control GUI apps:

New: Current fan state visualization clanged. Now big yellow/green dots follow the boost proposed by the sensor (from its curve). Red dot present real hardware situation."

This is not the case at all on my GUI. The vertical axis is Fan Speed and the horizontal axis is Temperature. 

a) When in Manual Mode the Red dot tracks along my curve for both CPU and GPU. 

b) In ANY Power Level 1-6, the Red dot is ALWAYS located at the very bottom of the graph, i.e. ZERO fan speed. I see a small Green dot where the actual temp is. I would expect the Red dot to be horizontally at the current temperature but well up the vertical scale to where ever the current fan speed actually is. 

c) Also, I see NO Yellow dot when in any numbered Power Level, but it IS present when I am in Manual mode - but only on the non-active graph. 

6) Lastly, I have found no documentation regarding the functionality of the 5 CPU Boost levels. I also note NO change in ANY fan speed while on AC and cycling between each of the 5 CPU Boost AC options. What are they supposed to do?

T-Troll, 2022-08-22:
In general, it works as designed.

Power modes are not about fans only. Yes, they have different fan settings (stop point, curve), but also alter a lot of power setting into the system - PLs, watchdogs, etc.
In fact, it's what AWCC names "Thermal profiles".

Fan control only available at Manual mode. System always return 0 for boost in other modes (but returns 100 all the time for some system if g-mode enabled).

We need to check the source of delay in UI, i'll try to reproduce, but it can be you BIOS-specific.

BTW, you can use ALT+CTRL+0...6 shortcut to change it any time.

TheSQLGuru, 2022-08-23:
REALLY?!? I am quite surprised to hear that a module named FAN CONTROL is messing with POWER settings! At first blush I am definitely not on board with that either. I use a combination of ThrottleStop, Process Lasso and MSI Afterburner to handle all things power related. IMHO, FAN CONTROL should do just that - control the fans. I do not want it overriding or monkeying with my extensively-tested settings. 

Questions:

A) Is there a way I can adjust the various fan/power settings/curves for each of the different levels?

B) Is is possible to disable all Power adjustments in the current edition of the application?

T-Troll, 2022-08-23:
> FAN CONTROL is messing with POWER

It named fans&powers in main app. But it's a point, i'll rename it.

> A) Is there a way I can adjust the various fan/power settings/curves for each of the different levels?

No. Fans can only be controlled at Manual mode.

> B) Is is possible to disable all Power adjustments in the current edition of the application?

In case you enable Manual but have other mode at app start - no. Else, just don't switch modes.

But you need to dig deeper inside what it really does. TS can't set or alter these platform (not CPU) related settings, it's blocked for it. So all TS adjustments is just in place.
I use TS as well (and lucky, -0.145 on CPU), so i sure about it.

TheSQLGuru, 2022-08-23:
-.145?!? Sheesh. I can only get -.06 on my 11980HK X17 CPU, and that is only outside high utilization. I was able to get -.135 on my previous laptop, a 7th gen Dell XPS 9550. Good thing too, as that thing had totally suck @ss cooling!

I REALLY wish I could find more details about all the nooks and crannies of TS. I sadly don't have time to read the Intel technical manuals to work things out on my own. And all of the online coverage of TS only hit the high points. 

I am going to copy the repository. Maybe some day I will have the time and drive to brush off my C/C++ skills and have a look under the covers to see what it is doing and how it is doing it. That may allow me to take better advantage of it. I doubt that day will ever come though. :-(

I do wish some enterprising user would take it upon themselves to flesh out the documentation a bit. That could make a big difference in our ability to use the program optimially! I would love it if you would publish some of the technical specs on exactly what the different setttings actually do. That too would be very helpful!!

TheSQLGuru, 2022-08-23:
Oh, I think there are some things from my OP that have not been addressed:

i) Why does power level 5 drop way down and then spool back up?

ii) Is there not a way you can keep the fans from dropping to 0 speed during any transition (assuming they are at a non-zero speed at the start of the trainsition)?? That just seems like a REALLY important thing to do, and if you are controlling the inputs to the fan control system it would seem like it should be doable. 

iii) Likewise can you throttle the ramp up like you obviously do on ramp down? 0 rpm to 5400 rpm instantaneously can't be doing the fans or control electronics any favors. 

iv) Hmm, I wonder if this statement in your initial reply means you may NOT be actually controlling the fan speed/changes outside of Manual mode??

"Fan control only available at Manual mode. System always return 0 for boost in other modes (but returns 100 all the time for some system if g-mode enabled)."

Could you provide a bit of detail about what you mean with that?

v) All of my points/questions about the graph responses in various modes seem to have slipped through the cracks. 

vi) Same for the functionality of the 5 CPU Boost levels.

T-Troll, 2022-08-23:
I don't sure Dell will publish some platform implementation details one time...

What my app do is not always (and mostly) into the app, but into ACPI. So you can dump ACPI (as you can see, i ask about it for undetected systems), and dig inside to configure out what and how is implemented inside.
For example, power mode switch is into ASL method named WMAX, function 15 subfunction 1. Exact method, modes and other settings can vary from platform to platform, so results different too.
In general, it set platform CPU and GPU power limit, alter fan PASV/HOT/CRIT values and provide different BIOS fan curves.

So...

1. Have no idea for your platform, they all different. Commonly, the last level is most power-saving.
2. It's platform implementation as well. Please keep in mind - i don't control fans directly (it's even impossible at some systems), i just communicate with BIOS about additional spin-up. Final control on BIOS - and i like it - it's quite safe.
3. According to 2 - i CAN NOT damage system any way (in case BIOS have no bugs into it). I even can not stop fans in case temp higher than PASV. And spin up should be ASAP - see my answer into other issue.
4. Even in manual mode. I only give BIOS some hints - see 2.
5. Yes. Irrelevant, sorry.
6. See above what it about.

TheSQLGuru, 2022-08-24:
Thanks for the details - it is very interesting!! It makes some of the operations and choices more understandable. 

I DEFINITELY agree with you that it is safer and thus better for you to be providing "guidance" and not having explicit control!

I will see about dumping the ACPI table to learn what it can do and how it works. 

HAHAHAHAHA!! On a lark I looked up the ACPI manual to see if I could find ASL or WMAX in it. That document (ACPI_Spec_6_4_Jan22.pdf) is ONE THOUSAND AND EIGHTY SEVEN PAGES LONG!!! Holy f*ck! I think that will likely end my pursuit of sorting out how this app works its magic! Extra to you for sorting this stuff out. Oh, and I was unable to find either of those terms in the manual, so I guess they may be program/platform/bios/driver specific.

As a likely final question, I still don't understand what the CPU Boost levels to (off/enabled/aggressive/efficient/efficient aggressive), nor why they don't seem to do anything on my laptop. I did "See above what is about", but didn't note anything specific. Do they just set varying parameters in the ACPI? And if so, could you at least provide some general guidance in the Wiki about what each level is supposed to do and when we might want to pick different values depending on what we are doing on our computers and whether or not we are plugged in or on battery?  Thanks in advance, and thank you again for all the time you have taken with my issues and questions! It is too bad that at least some of these posts can't be seen in the Wiki, as they are definitely helpful.

T-Troll, 2022-08-24:
> HAHAHAHAHA!! On a lark I looked up the ACPI manual to see if I could find ASL or WMAX in it. That document >> (ACPI_Spec_6_4_Jan22.pdf) is ONE THOUSAND AND EIGHTY SEVEN PAGES LONG!!! Holy f*ck! I [...]

Spec about other, you don't need to read it all (only to clear some common devices/methods). WMAX is a proprietary method added by Dell (and tied to Windows WMI calls).
So all you need is something like RWEverythig, then open ACPI and browse/dump it for easy view.

CPU boost levels was an idea of the one guy from Reddit (he and initial repo is listed into credits). it's about Windows Power Plan (hidden setting there). It's effective for Ryzen mostly, but provide a little juice (about 1-3%) even for my 9th gen Intel.

TheSQLGuru, 2022-08-26:
Ahh, WMAX is a Dell-specific command. I guess maybe that is why I couldn't find it. 

I have played around with RWEverything already. There are a couple of BIOS hacks available for the X17 R1 that I want to try out eventually. 

So for CPU boost levels - can I take it that Efficient Aggressive will result in the LOWEST power levels/clocks and Aggressive will result in the HIGHEST power levels/clocks? Those are the only two values I care about.

T-Troll, 2022-08-26:
It's not about PLs, AFAIK. But about some turboboost slopes and core parking algorithm.
As i said, it adds a little for Intel, but significant for Ryzen.
Top SHOULD BE (but who knows?) Aggressive, bottom should be Effective.

I check this using TS Bench, difference exist for 9th gen (above common variation), but small one.

WMAX is not exactly Dell's name, i just point to it for simplicity. More precise, i need to say "Look for device with HID "PNP0C14", UID "AWCC" and decode _WDG/WQMF blocks". ^_^
But this name is stable for the most of the hardware. Main functions inside too.

closed 2022-08-29 by T-Troll